raspberrypi/linux #1725: "Hardware: BCM2835" for kernel breaks wiringPi with rpi-4.8.y kernel

clivem, 2016-11-18:
Just noticed on Pi2B with 4.8 kernel, that /proc/cpuinfo "Hardware: BCM2835" breaks wiringPi. Suspect this is to do with recent changes to rpi-4.8.y..... ISTR, Arch used to be either BCM2708 or BCM2709, not BCM2835.

pelwell, 2016-11-18:
You'd be better off telling Gordon - he's got a while to make the necessary changes.

clivem, 2016-11-18:
Yeah, probably would. Just sticking it here for the moment so I don't forget it. Obviously something has changed on the kernel side of things...... I also wonder if this will affect other software that for years has depended on "Hardware" from /proc/cpuinfo to determine whether dealing with original ARMv6 hardware or ARMv7. I've not checked what is in the "Hardware" field on a BCM2708 build, but I wonder if that will now use BCM2835 as well. WHich if people are using that field to differentiate hardware.......

Ruffio, 2016-12-30:
@clivem has this been resolved (by email) or what is the status of this issue?

clivem, 2016-12-30:
I made Gordon aware of the issue via email. Whether he has a workaround in a new version of wiringPi, to differentiate the actual hardware in use some other way, I do not know. Anyway, I'll close the issue, based on the fact that @pelwell thought it was down to user-space to accommodate the change.

closed 2016-12-30 by clivem