raspberrypi/linux #1732: UART0 is broken on Raspberry Pi 2

mirzak, 2016-11-27:
Hi.

In commit https://github.com/raspberrypi/firmware/commit/d0bc6ce8e2ae7850959fed4edb0695f3cddfb96a, the default clock rate for UART0 changed from 3 MHz to 48 MHz. 

The update in firmware has not yet been aligned in the Linux kernel. I noticed this because the console port on my RPi 2 stopped working when I updated to latest firmware files.

I currently have below patch as a workaround, even though during boot I just get "scrambled" prints but when I get to the login prompt it works normally. So probably need to change a value somewhere else.

```patch
From e1fa015c619f880abd7c1c483c2e468e13f05641 Mon Sep 17 00:00:00 2001
From: Mirza Krak <mirza.krak@gmail.com>
Date: Sun, 27 Nov 2016 01:10:50 +0100
Subject: [PATCH 1/1] BCM270X: change clk_uart0 to 48 MHz

Recently the default values for uart_clock changed from 3 MHz to 48 MHz
on all pi`s. This is setup up by the boot firmware files.

Above actually breaks the console port (at least on RPi 2).

Align configuration to match boot firmware.

---
 arch/arm/boot/dts/bcm2708_common.dtsi | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/arch/arm/boot/dts/bcm2708_common.dtsi b/arch/arm/boot/dts/bcm2708_common.dtsi
index c449493..a9fd97e 100644
--- a/arch/arm/boot/dts/bcm2708_common.dtsi
+++ b/arch/arm/boot/dts/bcm2708_common.dtsi
@@ -456,7 +456,7 @@
 			reg = <2>;
 			#clock-cells = <0>;
 			clock-output-names = "uart0_pclk";
-			clock-frequency = <3000000>;
+			clock-frequency = <48000000>;
 		};
 
 		clk_apb_p: clock@3 {
-- 
2.1.4
```

pelwell, 2016-11-27:
You are correct that the frequency in the base .dtb should now be changed, but in practise it doesn't matter because the firmware patches the property at boot time. Try this:
```
dtc -I fs /proc/device-tree | grep -5 uart0_pclk
```
The clock_frequency will be in hex, with 0x2dc6c0 being 3MHz and 0x2dc6c00 being 48MHz.

If for some reason this reports 3MHz without your patch, can you provide the following information:

1) The output of `uname -a`

2) Add "dtdebug=1" to your config.txt, then after booting record the output of:
```
sudo vcdbg log msg |& grep -v -E "(HDMI|gpioman)"
```

mirzak, 2016-11-27:
> You are correct that the frequency in the base .dtb should now be changed, but in practise it doesn't matter because the firmware patches the property at boot time

How does this "patching" work? Cause I have a somewhat special case. I run u-boot first, and my kernel image and dtb live in another partition. So this is my problem then I guess, the "patching" does not happen.

If I run Linux instead u-boot first then it all works just fine.

pelwell, 2016-11-27:
The firmware loads a kernel first. Then it loads a DTB, applies any patches and overlays, and starts the kernel. The default stubs (see https://github.com/raspberrypi/firmware/issues/579) pass the kernel (or U-Boot) the address of the DTB in register r2 (ARMv7). Alternatively you can use the device_tree_address config setting to specify an alternative, fixed location.

The patches and overlays only modify the in-memory DTB, so unless U-Boot can forward it to the kernel, instead of loading a fresh one, the modifications are lost.

I will, of course, correct the default in the .dtb files.

pelwell, 2016-11-27:
Future builds will include the correction.

pelwell, 2016-12-05:
This corrected DTBs are available via rpi-update. Can you test and close if happy?

mirzak, 2016-12-05:
Gave this a test run I and get same result as my initial comment when I tried out my patch.

This is the output:
```
U-Boot 2016.11 (Dec 05 2016 - 22:50:55 +0100)                                                
                                                                                             
DRAM:  944 MiB                                                                               
RPI 2 Model B (0x2a01041)                                                                    
MMC:   bcm2835_sdhci: 0                                                                      
reading uboot.env                                                                            
*** Warning - bad CRC, using default environment                                             
                                                                                             
In:    serial                                                                                
Out:   lcd                                                                                   
Err:   lcd                                                                                   
Net:   Net Initialization Skipped                                                            
No ethernet found.                                                                           
starting USB...                                                                              
USB0:   Core Release: 2.80a                                                                  
scanning bus 0 for devices... 4 USB Device(s) found                                          
       scanning usb for storage devices... 0 Storage Device(s) found                         
       scanning usb for ethernet devices... 1 Ethernet Device(s) found                       
Hit any key to stop autoboot:  0                                                             
switch to partitions #0, OK                                                                  
mmc0 is current device                                                                       
Scanning mmc 0:1...                                                                          
Found U-Boot script /boot.scr                                                                
reading /boot.scr                                                                            
448 bytes read in 9 ms (47.9 KiB/s)                                                          
## Executing script at 02000000                                                              
switch to partitions #0, OK                                                                  
mmc0 is current device                                                                       
4221904 bytes read in 996 ms (4 MiB/s)                                                       
15356 bytes read in 429 ms (34.2 KiB/s)                                                      
## Booting kernel from Legacy Image at 01000000 ...                                          
   Image Name:   Linux-4.4.36                                                                
   Image Type:   ARM Linux Kernel Image (uncompressed)                                       
   Data Size:    4221840 Bytes = 4 MiB                                                       
   Load Address: 00008000                                                                    
   Entry Point:  00008000                                                                    
   Verifying Checksum ... OK                                                                 
## Flattened Device Tree blob at 00000100                                                    
   Booting using the fdt blob at 0x000100                                                    
   Loading Kernel Image ... OK                                                               
   Using Device Tree in place at 00000100, end 00006cfb                                      
                                                                                             
Starting kernel ...                                                                          
                                                                                             
B�V��2B���������B�������z���ZP�f�zB�w�����B�^Ҏ���gB�'��2��BN"Z�����B���ZJ���ς�g��BnK   ��J�

                                                                                            J
                                                                                            [
                                                                                            [
                                                                                            �
Poky (Yocto Project Reference Distro) 2.2 raspberrypi2 ttyAMA0                               
                                                                                             
raspberrypi2 login: [    8.651122] wlan0: authenticate with 9c:97:26:73:f5:83                
[    8.739098] wlan0: send auth to 9c:97:26:73:f5:83 (try 1/3)                               
[    8.746730] wlan0: authenticated                                                          
[    8.760613] wlan0: associate with 9c:97:26:73:f5:83 (try 1/3)                             
[    8.766716] wlan0: RX AssocResp from 9c:97:26:73:f5:83 (capab=0x1411 status=0 aid=5)      
[    8.783911] wlan0: associated                                                             
[... cut ...]
root@raspberrypi2:~#                                                                         
root@raspberrypi2:~# cat /uboot/config.txt | grep -i init_uart_clock                         
## init_uart_clock                                                                           
#init_uart_clock=3000000
```

The "scrambled" part should be the Linux boot log. But strangely it works when I get to the login prompt. Is there a second location that needs to be updated? Maybe something to do with early printk?

pelwell, 2016-12-06:
What does the kernel command line say (`cat /proc/cmdline`)?

mirzak, 2016-12-06:
The kernel command line said:
```
earlyprintk console=tty0 console=ttyAMA0 root=${mender_kernel_root} rootfstype=ext4 rootwait noinitrd
```
changed to
```
earlyprintk console=tty0 console=ttyAMA0,115200 root=${mender_kernel_root} rootfstype=ext4 rootwait noinitrd
```
which resolved the problem with "scrambled" output. 

I am OK with closing this issue, since this was only a problem on my end.

closed 2016-12-06 by mirzak