raspberrypi/linux #1733: raspberrypi-ui-mods update assumes the existence of user "pi", unable to update package

SarenT, 2016-11-28:
Hi,

I have recently updates some packages via aptitude and I have got the following error during the update process:

> /home/pi/.config/gtk-3.0/gtk.css
> chown: invalid user: ‘pi:pi’
> dpkg: error processing package raspberrypi-ui-mods (--configure):
>  subprocess installed post-installation script returned error exit status 1
> Setting up raspi-config (20161114) ...
> Processing triggers for libc-bin (2.19-18+deb8u6) ...
> Errors were encountered while processing:
>  raspberrypi-ui-mods
> E: Sub-process /usr/bin/dpkg returned an error code (1)
> Failed to perform requested operation on package.  Trying to recover:
> Setting up raspberrypi-ui-mods (1.20161115) ...
> Backing up old config files...

I don't have a user called pi anymore (due to security reasons). And I guess the package should not assume the existence of such a user anyway.
Please note that, I have unlocked aptitude and started again, it doesn't show any issues or updates related to the package.

pelwell, 2016-11-28:
This is the kernel repo - try the [raspberrypi-ui-mods](https://github.com/RPi-Distro/raspberrypi-ui-mods) repo instead.

But first, ask yourself why the absence of the userid pi is any safer? If you don't want anyone to log in as pi, make it so they can't - `sudo passwd -l pi` ought to lock it.

SarenT, 2016-11-28:
Hi, pelwell sorry for that I came through the website. And I guess you are right. But I have read about it on a website (removing the user pi). Form the point of simplicity, it makes a bit sense. If I don't need that user, than I simply remove it and forget about it. But, I didn't thoroughly about it.

closed 2016-11-28 by SarenT