raspberrypi/linux #1734: bcm2835-camera is missing the V4L2_CAP_VIDEO_CAPTURE_MPLANE capability

gcampax, 2016-11-29:
The PiCamera V4L2 driver is exposing planar pixel formats such as YUV420 (YV12 fourcc), but it's not including the MPLANE capability, which means the v4l2 core code rejects any planar API call with EINVAL and those formats cannot be used.

6by9, 2016-11-30:
Based on my understanding it is correct.

In the case of YUV420, it is supporting V4L2_PIX_FMT_YUV420 http://hverkuil.home.xs4all.nl/spec/uapi/v4l/pixfmt-yuv420.html which requires that the planes are contiguous in memory.

It is NOT supporting V4L2_PIX_FMT_YUV420M http://hverkuil.home.xs4all.nl/spec/uapi/v4l/pixfmt-yuv420m.html where the planes are discontiguous and would need to use the MPLANE capability, as specified in http://hverkuil.home.xs4all.nl/spec/uapi/v4l/vidioc-querycap.html#vidioc-querycap for V4L2_CAP_VIDEO_CAPTURE_MPLANE.

Can you give a specific example (ie code) of what you are trying to do and the result you're expecting?

6by9, 2016-11-30:
(That does remind me that I ought to run v4l2-conformance on the driver again to see if they've made any further changes to it that we no longer comply with. We were passing with no errors or warnings).

6by9, 2016-11-30:
v4l2-compliance run. 1 error, 1 warning which were related to one field not being filled in on VIDIOC_S_PARM. PR created to fix it and that is then a clear test - https://github.com/raspberrypi/linux/pull/1736

I still don't believe it is handling multi planar buffer capabilities incorrectly, but I could be convinced otherwise.

gcampax, 2016-11-30:
Ok, I have to admit I did not know the distinction between V4L2_PIX_FMT_YUV420 and V4L2_PIX_FMT_YUV420M.

I'm basing myself on https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/pixfmt-yuv420.html which describes YUV420 as planar not packed. I assumed that would mean I need to use the multi-plane API.

If that's not the case, please close this issue, and sorry for the noise.

6by9, 2016-12-01:
Yes YUV420 is a planar format, but V4L2_PIX_FMT_YUV420 predates the multi-plane API and has fixed expectations of how those planes are laid out within a single buffer. (Y plane first with bytesperline bytes per line, then U with bytesperline/2 bytes per line, and then V with same stride as U)

V4L2_PIX_FMT_YUV420M (not supported on Pi) is the new multi-planar format which allows the planes to be stored in multiple buffers and/or have different bytesperline values per plane.

Due to the way the hardware works, there is no gain in having multiple buffers on Pi, so there is little point in looking to support it.

gcampax, 2016-12-01:
Oh, ok. My code was wrong then.
Thanks for the info.

closed 2016-12-01 by gcampax