raspberrypi/linux #1735: ENC28J60 module doesn't work on SPI2 interface for compute module

yidingalan, 2016-11-30:
I am testing to see if ENC28J60(Ethernet module) works on SPI2 interface, and I am having some issues. Basically the wire connections will be: 

INT - GPIO39
MISO - GPIO40
MOSI - GPIO41
SCLK - GPIO42
CE - GPIO43

I have it working with Pi zero, and SPI0 on compute module, but didn't work on SPI2 for compute module. It is always trying to access SPI0 during the bootup and therefore it wouldn't pick up the ethernet. The current enc28j60 overlay file is for SPI0 apparently, and I am wondering if I need to create my own overlay .dts file specifically tailored to SPI2, or does it already exist? If not, are there any alternatives or shortcut that allows me to access SPI2 right away instead of creating a new dts file(for example one line of change in /boot/config.txt)?

[image]
This is the boot time message and you can tell it's accessing SPI0 although I already connected my module to SPI2

Any help and insights will be appreciated

pelwell, 2016-11-30:
Yes, you will need a new overlay for that. Are you up for having go yourself?

I'm happy to guide you through it, but start by copying the existing one and changing all spi0 references to say spi2. Then you will probably need to merge in some of the spi2 overlay, but you can omit all references to spidev2 - those that add it and those that disable it.

yidingalan, 2016-11-30:
@pelwell That would be great and thanks for the help. Do you have the ENC28J60.dts file for SPI0 by any chance? I could not find it in my /boot/ directory or online

pelwell, 2016-11-30:
The source is in the kernel tree, [here](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/enc28j60-overlay.dts).

yidingalan, 2016-11-30:
@pelwell I am assuming I need to include `dtoverlay=spi2-1cs` in my config.txt file right

pelwell, 2016-11-30:
Yes you do, but add ",cs0_spidev=no" to prevent the spidev nicre from being enabled.

pelwell, 2016-11-30:
Sorry, that should day node - Swype error.

yidingalan, 2016-11-30:
OK so I modified the original .dts file and changed all the reference to spi2, all the interrupt pin from 25 to 39. I also added what you told me to spi2-1cs.

It's still not working and when I run `dmesg | grep spi`, this is what showed up:
`[   13.751248] enc28j60 spi2.0: could not find pctldev for node /soc/interrupt-controller@7e00b200, deferring probe 
[   14.339506] enc28j60 spi2.0: could not find pctldev for node /soc/interrupt-controller@7e00b200, deferring probe`

Any idea of what's causing the issue here?

pelwell, 2016-11-30:
I'm going to have to continue this in the morning, but can you post the output from raspi-gpio? Sometimes those deferring probe messages are just a transient hiccup in the dependencies that gets resolved when another driver is loaded.

yidingalan, 2016-11-30:
@pelwell sure no rush. Appreciate your help
```
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=0 alt=  func=INPUT
  GPIO 03: level=1 fsel=0 alt=  func=INPUT
  GPIO 04: level=1 fsel=0 alt=  func=INPUT
  GPIO 05: level=1 fsel=0 alt=  func=INPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 08: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 09: level=0 fsel=4 alt=0 func=SPI0_MISO
  GPIO 10: level=0 fsel=4 alt=0 func=SPI0_MOSI
  GPIO 11: level=0 fsel=4 alt=0 func=SPI0_SCLK
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=0 fsel=0 alt=  func=INPUT
  GPIO 15: level=0 fsel=0 alt=  func=INPUT
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=4 alt=0 func=PCM_CLK
  GPIO 19: level=0 fsel=4 alt=0 func=PCM_FS
  GPIO 20: level=0 fsel=4 alt=0 func=PCM_DIN
  GPIO 21: level=0 fsel=4 alt=0 func=PCM_DOUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=0 fsel=0 alt=  func=INPUT
  GPIO 29: level=0 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=0 fsel=0 alt=  func=INPUT
  GPIO 33: level=0 fsel=0 alt=  func=INPUT
  GPIO 34: level=1 fsel=0 alt=  func=INPUT
  GPIO 35: level=1 fsel=0 alt=  func=INPUT
  GPIO 36: level=1 fsel=0 alt=  func=INPUT
  GPIO 37: level=0 fsel=0 alt=  func=INPUT
  GPIO 38: level=0 fsel=0 alt=  func=INPUT
  GPIO 39: level=1 fsel=0 alt=  func=INPUT
  GPIO 40: level=0 fsel=3 alt=4 func=SPI2_MISO
  GPIO 41: level=0 fsel=3 alt=4 func=SPI2_MOSI
  GPIO 42: level=0 fsel=3 alt=4 func=SPI2_SCLK
  GPIO 43: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 44: level=0 fsel=0 alt=  func=INPUT
  GPIO 45: level=0 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=0 fsel=0 alt=  func=INPUT
  GPIO 47: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 48: level=1 fsel=7 alt=3 func=SD1_CLK
  GPIO 49: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 50: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 51: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 52: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 53: level=1 fsel=7 alt=3 func=SD1_DAT3
```

pelwell, 2016-11-30:
One more: `sudo vcdbg log msg | grep -v -E "(HDMI|gpioman)"`

yidingalan, 2016-11-30:
```
001562.058: HDMI:EDID version 1.3, 1 extensions, screen size 52x29 cm
001562.122: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
001562.183: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
001562.223: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
001562.261: HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
001562.299: HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
001562.338: HDMI:EDID found DMT format: code 8, 800x600p @ 56 Hz in established timing I/II
001562.374: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
001562.413: HDMI:EDID found DMT format: code 10, 800x600p @ 72 Hz in established timing I/II
001562.450: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
001562.489: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
001562.527: HDMI:EDID found DMT format: code 17, 1024x768p @ 70 Hz in established timing I/II
001562.566: HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
001562.606: HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
001562.694: HDMI:EDID standard timings block x 8: 0x714F 81C0 8100 8180 9500 A9C0 B300 0101
001562.747: HDMI:EDID found DMT format: code 21, 1152x864p @ 75 Hz (4:3) in standard timing 0
001562.809: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 1
001562.858: HDMI:EDID found DMT format: code 28, 1280x800p @ 60 Hz (16:10) in standard timing 2
001562.908: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 3
001562.961: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 4
001563.015: HDMI:EDID found DMT format: code 83, 1600x900p @ 60 Hz (16:9) in standard timing 5
001563.066: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 6
001576.069: HDMI:EDID parsing v3 CEA extension 0
001576.108: HDMI:EDID monitor support - underscan IT formats:yes, basic audio:no, yuv444:yes, yuv422:yes, #native DTD:1
001576.150: HDMI:EDID found CEA detail timing format: 1280x720p @ 50 Hz (19)
001576.194: HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
001576.236: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
001576.275: HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz (native)
001576.307: HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz
001576.341: HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz
001576.373: HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz
001576.406: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz
001576.438: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz
001576.456: HDMI:EDID found HDMI VSDB length 5
001576.481: HDMI:EDID HDMI VSDB has physical address 1.0.0.0
001576.495: HDMI:EDID HDMI VSDB has no extension fields
001576.537: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
001576.749: HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
001580.562: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.581: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.612: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.630: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.660: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.678: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.707: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.726: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.755: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.773: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.802: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.821: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.850: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.868: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.897: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.915: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.945: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001580.963: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001580.992: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.010: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.040: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.058: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.088: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.106: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.136: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.155: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.184: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.203: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.232: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
001581.251: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
001581.280: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
[... 188 lines cut ...]
002038.660: Kernel trailer DDTK property says yes
[... cut ...]
002270.879: Device tree loaded to 0x1bfec500 (size 0x3afd)
002271.420: gpioman: gpioman_get_pin_num: pin SMPS_SDA not defined
002271.439: gpioman: gpioman_get_pin_num: pin SMPS_SCL not defined
002272.161: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
004804.009: vchiq_core: vchiq_init_state: slot_zero = 0x5b880000, is_master = 1
004808.424: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004813.748: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
```

pelwell, 2016-12-01:
Using a fresh kernel and a hacked enc28j60-spi2 overlay it gets as far as probing the enc28j60 and not finding it - not surprising since I don't have one.

My overlay has only two differences to the original - <&spi0> is now <&spi2>, and I deleted fragment 1 that disabled spidev0, renaming the following fragment to be the new fragment 1.

I wonder if you are having problems because your kernel and DTBs are too old - what does `uname -a` report?

To save time you can upload your overlay and config.txt and I can try with them.

By the way, what is your reason for using the mmc overlay? Once you have updated to a current kernel it should no longer be necessary - I found the reason for the unreliability using the sdhost driver with eMMC, and now it is solid (and quicker).

yidingalan, 2016-12-01:
what `uname -a` gave me: `Linux is-024 4.4.11+ #888 Mon May 23 20:02:58 BST 2016 armv6l GNU/Linux`

I am not sure why we are using mmc overlay since I didn't configure that, so are you suggesting I could comment that line out if I update to a current kernal?

my config.txt:
```
# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
dtoverlay=enc28j60-spi2
dtoverlay=spi2-1cs,cs0_spidev=no

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi
dtoverlay=mmc
#dtoverlay=remap-i2c0,sda0=28,scl0=29

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
#dtparam=i2c_arm=on
#dtparam=i2c_vc=on
#dtparam=i2c=on_vc=on
#dtparam=i2c=on0=on
#dtparam=i2c=on1=on
#dtparam=i2c0=on
#dtparam=i2c1=on

#device_tree=

max_usb_current=1
#start_x=0
#dtparam=i2c_arm=on
#dtparam=i2c0=on
#dtparam=i2c1=on
#dtparam=i2c_vc=on
dtparam=spi=on
```

and my enc28j60-spi2.dts (I included your change plus changed the int pin):
```
// Overlay for the Microchip ENC28J60 Ethernet Controller - SPI2
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2708";

    fragment@0 {
        target = <&spi2>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;

            status = "okay";

            eth1: enc28j60@0{
                compatible = "microchip,enc28j60";
                reg = <0>; /* CE0 */
                pinctrl-names = "default";
                pinctrl-0 = <&eth1_pins>;
                interrupt-parent = <&gpio>;
                interrupts = <39 0x2>; /* falling edge */
                spi-max-frequency = <12000000>;
                status = "okay";
            };
        };
    };

    fragment@1 {
        target = <&gpio>;
        __overlay__ {
            eth1_pins: eth1_pins {
                brcm,pins = <39>;
                brcm,function = <0>; /* in */
                brcm,pull = <0>; /* none */
            };
        };
    };

    __overrides__ {
        int_pin = <&eth1>, "interrupts:0",
                  <&eth1_pins>, "brcm,pins:0";
        speed   = <&eth1>, "spi-max-frequency:0";
    };
};
```

Also, when i compiled the .dts to .dtbo, I got this warning not sure if it's something wrong: 
```
Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
```

yidingalan, 2016-12-01:
I think it's working now. After deleting the old fragment 1 and changed the old 2 to 1, the Ethernet module starts working now. Thanks a lot

With regards to the mmc overlay that you mentioned earlier, how do I change that? @pelwell

pelwell, 2016-12-01:
> I think it's working now.

Cool - good job.

The warnings are currently a fact of life - the overlay syntax doesn't obey some of the newer rules the DT compiler is trying to enforce. Overlay support in dtc is currently going through the upstreaming process, so I'm hoping that will change.

The sdhost driver was made standard for CM back in kernel 4.4.19, but that is newer than the one you have (which doesn't look like an RPF build). You can either run rpi-update to get a new firmware and kernel - there are no known problems with the current version - or you can go back to the supplier of your kernel and see what they have.

yidingalan, 2016-12-01:
ok cool I will do that, btw is there a need for a PR for this spi2 overlay file?

pelwell, 2016-12-01:
It's unfortunate that we are accumulating an n by m matrix of SPI devices and their buses, but there isn't an alternative at the moment. Feel free to create a PR.

pelwell, 2016-12-01:
Actually there is an alternative, but it's ugly: see https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/mcp23s17-overlay.dts

yidingalan, 2016-12-02:
@pelwell do you happen to know if there is an existing .dts file for decawave DWM1000 module on SPI0?

pelwell, 2016-12-02:
All the overlays are listed in /boot/overlays/README.

closed 2016-12-02 by pelwell