raspberrypi/linux #1737: RPI3 and usb touch screen fb

gandji, 2016-12-01:
Hi,
I have been struggling for some time with this, need help.

I have a RPI3 with jessie based raspbian distro. I use it with a MIMO usb touch screen (only screen device I have). kernel version is now 4.1.18

MIMO screen used to work with old kernel 3.x.x, and relevant drivers installed. But I don't know how it had been installed.

I upgraded some packages to compile and install OpenCPN (an app for sailing). Screen acceleration/refresh does not work anymore on the screen, 
neither do touch functionalities on the screen.

So I try to compile a kernel with USB frame buffer support.

After some tries, I focused on the rpi-3.8.y branch of this repository (following [http://elinux.org/Raspberry_Pi_Kernel_Compilation#1._On_the_Raspberry_Pi]).

But compilation of modules (sudo make modules) fails with:
  Building modules, stage 2.
  MODPOST 1221 modules
ERROR: "bcm_dma_wait_idle" [drivers/video/bcm2708_fb.ko] undefined!
/home/pi/raspbian/rpi-3.8.y/scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[1]: *** [__modpost] Error 1
Makefile:942: recipe for target 'modules' failed
make: *** [modules] Error 2

I added an export for bcm_dma_wait_idle in dma.c:
EXPORT_SYMBOL_GPL(bcm_dma_wait_idle);

to no avail.

I am at a loss. If anyone has a solution or hint? I can provide more info if needed.

Thanks in advance,
Gandji

popcornmix, 2016-12-01:
This isn't really the right place for building advice ([forum](https://www.raspberrypi.org/forums) would be better). But start by following these instructions:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

JamesH65, 2017-05-18:
Closing as this issue has been deemed unnecessary.

closed 2017-05-18 by JamesH65