raspberrypi/linux #1738: rpi-ft5406 64 bit

Gerharddc, 2016-12-01:
I am trying to get the RPI touch screen working on a 64 bit image and have already got VC4 and DSI working, I only need the touch driver. This is proving difficult because the "writel(99, &ts->regs->num_points);" line is causing a crash related to paging to an invalid address. I have concluded that this is caused by some invalid address translation because the driver was hard-coded for 32 bit addresses.

I have already tried replacing all the direct pointer dereferencing in the code with the explicit io functions but am still facing the same issue.

I have decided that the problem lies in:
```
rpi_firmware_property(fw, RPI_FIRMWARE_FRAMEBUFFER_GET_TOUCHBUF,
&touchbuf, sizeof(touchbuf));
...
touchbuf &= ~0xc0000000;
ts->ts_base = ioremap(touchbuf, sizeof(*ts->regs));
```
With touchbuf being a u32. 

I am more than willing to fix this driver myself, I just can't find any information on the firmware interface and therefore have no idea what type of address translation needs to take place. For instance, does it still provide a 32 bit address when running a 64 bit kernel and what is the reason for "touchbuf &= ~0xc0000000"? Basically I just need know how I can map the RPI_FIRMWARE_FRAMEBUFFER_GET_TOUCHBUF to a 64 bit virtual address.

I guess one also needs to do the right translation when suggesting to RPI_FIRMWARE_FRAMEBUFFER_SET_TOUCHBUF.

pelwell, 2016-12-01:
The firmware deals in bus addresses. It has no knowledge of whether the OS is 32- or 64-bit, and it doesn't care.

The VPU address space is limited to 1GB, with the top two bits indicating certain addressing modes. The ARM's kernel space is configured to start at bus address 0, so the mask is to remove those upper bits and leave you with a bus address ready to map to a suitable kernel virtual address.

Gerharddc, 2016-12-01:
@pelwell Ok so the pointer at ts->ts_base from ioremap at the moment should be perfectly valid for use in for example iowrite32?

pelwell, 2016-12-01:
I don't know off the top of my head, and I'm not about to start digging into 64-bit kernel internals until we decide to support it.

Gerharddc, 2016-12-01:
@pelwell Thanks for the support, I don't want to waste too much of your time. The last thing though, my error is as follow:

```
[    2.300780] Unable to handle kernel paging request at virtual address ffffff8008b32002
[    2.300799] pgd = ffffffc034ba9000
[    2.300808] [ffffff8008b32002] *pgd=00000000344a3003, *pud=00000000344a3003, *pmd=0000000000000000
[    2.300831] Internal error: Oops: 96000061 [#1] SMP
[    2.300842] Modules linked in: uio_pdrv_genirq uio rpi_ft5406
[    2.300869] CPU: 2 PID: 151 Comm: ft5406 Not tainted 4.8.11-v8+ #1
[    2.300877] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[    2.300886] task: ffffffc03459e800 task.stack: ffffffc034724000
[    2.300904] PC is at ft5406_thread+0x60/0x1e0 [rpi_ft5406]
[    2.300915] LR is at ft5406_thread+0x50/0x1e0 [rpi_ft5406]
[    2.300925] pc : [<ffffff8000680148>] lr : [<ffffff8000680138>] pstate: 60000145
[    2.300932] sp : ffffffc034727d90
[    2.300939] x29: ffffffc034727d90 x28: 0000000000000000 
[    2.300951] x27: 0000000000000000 x26: 0000000000000000 
[    2.300963] x25: 0000000000000000 x24: 0000000000000000 
[    2.300974] x23: 0000000000000001 x22: ffffffc034686218 
[    2.300986] x21: ffffffc034686218 x20: ffffff800899a718 
[    2.300997] x19: ffffffc0345d5300 x18: 0000000000000000 
[    2.301009] x17: 0000007f8c855ab8 x16: 0000000000000000 
[    2.301020] x15: ffffffffffffffff x14: 0000000000000011 
[    2.301032] x13: 6b636f6c623a636d x12: 6d3d5341494c4144 
[    2.301043] x11: 0000000000000000 x10: 00000000000008e0 
[    2.301055] x9 : ffffffc034724000 x8 : ffffffc03459f140 
[    2.301066] x7 : 0000000001903fe6 x6 : 0000000000000000 
[    2.301077] x5 : 0000000000000000 x4 : 0000000000000037 
[    2.301089] x3 : ffffffc034727e18 x2 : 0000000000000000 
[    2.301100] x1 : 0000000000000063 x0 : ffffff8008b32002 
[    2.301110] 
[    2.301119] Process ft5406 (pid: 151, stack limit = 0xffffffc034724020)
[    2.301129] Stack: (0xffffffc034727d90 to 0xffffffc034728000)
[    2.301140] 7d80:                                   ffffffc034727e20 ffffff80080bd0d8
[    2.301152] 7da0: ffffffc0345d5300 ffffff800899a718 ffffff8008766c48 ffffffc034686218
[    2.301164] 7dc0: ffffff80006800e8 0000000000000000 0000000000000000 0000000000000000
[    2.301175] 7de0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301186] 7e00: 0000000000000000 0000000000000000 0000000000000000 ff00000000000000
[    2.301198] 7e20: 0000000000000000 ffffff8008083680 ffffff80080bd000 ffffffc0345d5300
[    2.301209] 7e40: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301221] 7e60: ffffffc034727ea0 0000000000000000 ffffffc034435b00 ffffffc034686218
[    2.301232] 7e80: 0000000000000000 0000000000000000 ffffffc034727e90 ffffffc034727e90
[    2.301244] 7ea0: 0000000000000000 ffffff8000000000 ffffffc034727eb0 ffffffc034727eb0
[    2.301256] 7ec0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301267] 7ee0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301277] 7f00: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301288] 7f20: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301300] 7f40: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301311] 7f60: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301322] 7f80: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301333] 7fa0: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[    2.301344] 7fc0: 0000000000000000 0000000000000005 0000000000000000 0000000000000000
[    2.301355] 7fe0: 0000000000000000 0000000000000000 75f155517775df57 5555555775555575
[    2.301363] Call trace:
[    2.301373] Exception stack(0xffffffc034727bc0 to 0xffffffc034727cf0)
[    2.301385] 7bc0: ffffffc0345d5300 0000008000000000 ffffffc034727d90 ffffff8000680148
[    2.301397] 7be0: ffffffc034727d28 ffffff8008897d00 ffffff80088e8010 ffffffc034727ca8
[    2.301409] 7c00: ffffff8008898f80 0000000000000001 ffffffc034727c60 ffffff8008671a60
[    2.301421] 7c20: 0000000000000140 ffffffc03679dd00 ffffffc03679dd00 ffffffc03679dd00
[    2.301432] 7c40: ffffff80088e5000 00000000ffffffff 0000000000000000 0000000000000001
[    2.301444] 7c60: ffffff8008b32002 0000000000000063 0000000000000000 ffffffc034727e18
[    2.301455] 7c80: 0000000000000037 0000000000000000 0000000000000000 0000000001903fe6
[    2.301466] 7ca0: ffffffc03459f140 ffffffc034724000 00000000000008e0 0000000000000000
[    2.301478] 7cc0: 6d3d5341494c4144 6b636f6c623a636d 0000000000000011 ffffffffffffffff
[    2.301488] 7ce0: 0000000000000000 0000007f8c855ab8
[    2.301503] [<ffffff8000680148>] ft5406_thread+0x60/0x1e0 [rpi_ft5406]
[    2.301521] [<ffffff80080bd0d8>] kthread+0xd8/0xf0
[    2.301536] [<ffffff8008083680>] ret_from_fork+0x10/0x50
[    2.301549] Code: f9400aa0 91000800 d5033e9f 52800c61 (b9000001) 
[    2.301561] ---[ end trace bf0c79f56a4b849d ]---
```
Maybe this can tell you more than that the address is invalid?

Also, is there any device tree configs (dma maybe) in the normal 32bit tree related to the address translations for this buffer that I can try in my 64bit one?

pelwell, 2016-12-01:
It's not obvious to me what is going wrong. I suggest you use pr_err() to display some key values in ft5406_thread. If that doesn't help, disassembler ft5406_thread and post it along with the error and I'll see what I can do.

Although we aren't using it in the 4.4 tree, the dma-ranges property does the translation from physical addresses to bus addresses. On a Pi 3 it should be:
```
dma-ranges = <0xc0000000 0x00000000 0x3f000000>;
```
which says that virtual addresses from 0x00000000 to 0x3effffff map to bus addresses 0xc0000000 to 0xfeffffff.

Gerharddc, 2016-12-01:
@pelwell It seems dma-ranges was already there and set correctly.

Thanks for the help, I finally got it working. I went through all the code that works with shared memory and reimplemented it to be more platform agnostic and I also changed from writel (u32) to wiritng 99 in 8bit because it is just an 8bit number so now everything works.

My fork is a bit of mess with all of the other work that I have done (and on 4.8 ...) so I will have to see what I can do to get a pull request for this driver when I get the time. I'm guessing I'll have to send a pull request to the 4.9 branch?

pelwell, 2016-12-01:
Nice one. I see no problem with you submitting a PR to the 4.8 branch, since that's where the bcmrpi3_defconfig is.

Gerharddc, 2016-12-01:
Ok I'll see if I can do that tomorrow. 4.9 also has bcmrpi3_defconfig but it's pretty broken there (another issue I might deal with later). The only problem with this driver (and the rest) on 4.8 at the moment is that I haven't been able to get dt overlays to load on 64bit so I had to add everything to the normal trees. I guess I can submit only the code for the drivers so long and then try to solve that seperately (maybe open another issue) if anyone wants to be able to use it.

pelwell, 2016-12-01:
When you say overlays don't work, do you mean using the dtoverlay command line utility or via the bootloader (config.txt)? If you can give me a simple concrete example, ideally not requiring external hardware, I'll add it to the list of things to look at.

Gerharddc, 2016-12-01:
@pelwell I'm pretty beat right now and will try to send you a complete example first thing tomorrow morning. I'm talking about the config.txt, I haven't tried the utility yet but that's not what I need anyway. From my experience though you should be able to try any overlay in the overlays folder and see that it has no effect on a 64 build, that is after you manually copy the dts though because at the moment the arm64 folder does not have an overlays directory nor a symlink which is also a problem.

pelwell, 2016-12-01:
OK - I'll give it a try. Thanks for the heads up.

pelwell, 2016-12-01:
That wasn't too hard - there's a patch in the rpi-4.8.y tree. It still doesn't build the overlays though - I didn't find a satisfactory solution.

Gerharddc, 2016-12-02:
@pelwell Could you get overlays to load with your patch? I have the code needed to build the overlays at https://github.com/raspberrypi/linux/pull/1739 .

pelwell, 2016-12-02:
I believe the overlay mechanism is working as it should - I tested it using the dtmerge utility which contains the same code, and just now on a Pi3. Having said that, the 64-bit system is not behaving as I would expect it to. For example, I2C isn't showing up even when enabled, which is because bcmrpi3_defconfig is enabling the wrong I2C driver (bcm2708-i2c instead of bcm2835-i2c). With a corrected bcmrpi3_defconfig it works.

Note that if you use the dtbs_install make target then the base dtbs will be installed into the "broadcom" subdirectory - not ideal.

It will take time to bring the arm64 configuration in line with the existing arm configurations, and some drivers still need attention (no, vcdbg doesn't work yet).

Gerharddc, 2016-12-02:
Getting dtmerge built and running on my machine looks like it could be a lot of effort. I doubt if the issue is with the kernel though, I think the problem has to be with the firmware and config.txt . If it's not to much effort I would appreciate if you could check to see if config.txt works for you.

I need my machine to load the overlay at boot so if this doesn't work I'll have to work with custom main trees for now. I'm not sure what procedure the firmware uses to communicate the overlay loading, is there any possibility that this could be broken with the different architecture?

I'll work on sorting out bcmrpi3_defconfig as my needs dictate. Lucklily having the dtbs in "broadcom" isn't much of a problem for me.

pelwell, 2016-12-02:
The entire device tree is visible in /proc/device-tree - what you see there is what was written by the firmware, but with directories for nodes and files for properties.

`dtc -I fs /proc/device-tree` generates a textual version of the DT. If you run that with and without your overlay applied and diff the two you will see the changes.

If you want you can also post a link to your overlay (or email me if you prefer - I won't share it) and I'll check it over.

Gerharddc, 2016-12-02:
Ok I'll check the proc. I don't have a fancy overlay yet I am simply talking about the standard stuff like the vc4 or touchscreen overlays. The simplest one I'm working with at the moment is rpi-ft5406-overlay.dts. When I make the dtbo, put it in the overlay folder and ask config.txt to load it then nothing happends but when I instead add

```
rpi_ft5406: rpi_ft5406 {
				compatible = "rpi,rpi-ft5406"; 
				firmware = <&firmware>;
				status = "okay";
			};
```

directly to "/" in bcm2710-rpi-3-b.dts then it works perfectly. Obviously it just isn't loading but I don't know if there is an error or if the firmware just isn't trying to load it.

pelwell, 2016-12-02:
The overlay loads for me:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.8.11-v8+ #15 SMP Fri Dec 2 11:18:10 GMT 2016 aarch64 GNU/Linux
pi@raspberrypi:~$ dmesg | grep 5406
[    8.822578] rpi-ft5406 rpi_ft5406: Probing device
[    8.824329] input: FT5406 memory based driver as /devices/virtual/input/input4
```
Make sure you are spelling the overlay name correctly, and underscores are different from hyphens. config.txt needs to contain:
```
dtoverlay=rpi-ft5406
```

Gerharddc, 2016-12-02:
Ok I'll double check. My fix is now at https://github.com/raspberrypi/linux/pull/1740 . I am also now trying to see if perhaps there is an issue with the DDTK flag on my kernel.

pelwell, 2016-12-02:
I recommend that you skip the mkknlimg step now - the defaults are DDTK on, and it will try both new (.dtbo) and old (-overlay.dtb) overlays - just cp Image to kernel8.img.

Gerharddc, 2016-12-02:
I really don't get it, that just doesn't want to work for me. Could userspace make a difference? I am currently on a yocto build and guess I could try raspbian because the 64bit should run it (I hope).

PS. I do have the very latest firmware files.

pelwell, 2016-12-02:
Should I assume that your device tree with the dtoverlay line is the same as without?

1) What is in your config.txt? You can strip comments and blank lines.

2) Can I see an "ls -lR /boot"?

Gerharddc, 2016-12-02:
Yes I'm running exactly the same device tree.

My config.txt after cleaning the million comments:

```
gpu_mem=256
dtoverlay=rpi-ft5406
```

And it seems that /boot is actually not being mounted automatically. Is this usually done by fstab? Could that be my problem or is loading done directly with regards to the partition?

The contents after manual mounting are:

```
total 21462
-rw-r--r-- 1 gerhard users   13485 Dec  2 14:18 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 gerhard users       0 Dec  2 10:51 bcm2835-bootfiles-20161020.stamp
-rw-r--r-- 1 gerhard users   17932 Nov 26 15:10 bootcode.bin
-rw-r--r-- 1 gerhard users     105 Dec  2 10:56 cmdline.txt
-rw-r--r-- 1 gerhard users   36223 Dec  2 14:59 config.txt
-rw-r--r-- 1 gerhard users   18693 Nov 26 15:10 COPYING.linux
-rw-r--r-- 1 gerhard users    2527 Nov 26 15:10 fixup_cd.dat
-rw-r--r-- 1 gerhard users    6620 Nov 26 15:10 fixup.dat
-rw-r--r-- 1 gerhard users    9751 Nov 26 15:10 fixup_db.dat
-rw-r--r-- 1 gerhard users    9749 Nov 26 15:10 fixup_x.dat
-rw-r--r-- 1 gerhard users      51 Dec  2 10:51 image-version-info
-rw-r--r-- 1 gerhard users 9525760 Dec  2 14:48 kernel8.img
-rw-r--r-- 1 gerhard users    1494 Nov 26 15:10 LICENCE.broadcom
drwxr-xr-x 2 gerhard users    2048 Dec  2 14:19 overlays
-rw-r--r-- 1 gerhard users  633636 Nov 26 15:10 start_cd.elf
-rw-r--r-- 1 gerhard users 4954692 Nov 26 15:10 start_db.elf
-rw-r--r-- 1 gerhard users 2821540 Nov 26 15:10 start.elf
-rw-r--r-- 1 gerhard users 3904260 Nov 26 15:10 start_x.elf

./overlays:
total 2
-rw-r--r-- 1 gerhard users 434 Dec  2 15:13 rpi-ft5406.dtbo
```

pelwell, 2016-12-02:
Both your base DTB and your overlay are smaller than mine. Your kernel looks about right though. Can you upload them somewhere so I can a) compare them to mine, and b) try the dtmerge ?

Gerharddc, 2016-12-02:
It seems I was messing with the "compatible" value on the overlay and messed it up a bit. After restoring it it is 462B.

I have posted both on https://www.dropbox.com/sh/6068fga35idjtvj/AAAa-0MgNHVniV2rlAWI4W_8a?dl=0 and I'll see if I can get dtmerge going.

pelwell, 2016-12-02:
They both look OK to me, and they merge OK. I can see that you've folded in the firmwarekms, dma and dsi nodes, but otherwise they look the same as mine.

Gerharddc, 2016-12-02:
Yes I had to fold them in to get them running. I'm compiling a completely clean kernel from git now and will try to see if any of my other changes did something but I doubt it.

Are you using normal raspbian as a rootfs? I'm trying figure out an ideal clean setup to just test overlay functionaility with...

pelwell, 2016-12-02:
I have an NFS-mounted rootfs which has accumulated some cruft, but it's essentially regular Raspbian.

Gerharddc, 2016-12-02:
Ok for some reason both my dirty and a clean kernel load the overlay just fine with Raspbian. I don't know if this is because it is a 32bit userspace or if Raspbian is doing some other magic.

Strangely enough though the old touch driver also works with the Raspbian userpace and an arm64 kernel. It's also immediately getting the suggested dma buffer accepted by the firmware instead of having to request one from the firmware. I'll build my userspace in 32bit as well to try and see if that's the difference.

I do feel my changes to the touch driver are still relevant though because from what I have read dealing directly with iomem pointers as normal pointers is not entirely legal and very dangerous. I still don't get the effect of userspace though.

Gerharddc, 2016-12-02:
The point of maximum confusion has now been reached.

It seems that when using a 32bit rootfs (my Yocto or Raspbian) even with the 64bit kernel then the overlay loads perfectly. The strangest things though is that it loads by simply being in the overlay folder, I don't even have to pass anything to config.txt I mean wtf?

When dealing with Raspbian I found that when I originally started getting the ft5406 errors, I was dealing with slightly oldish firmware and that the latest firmware is fine with the old code so that's why it works on Raspbian.

I'm still puzzled with the overlay loading. I'll double check to see that everyhting in my 64 rootfs is the same as the 32 one I just built but it seems there is definitely something userspace involved in overlay loading in some way.

pelwell, 2016-12-02:
Because the driver is used by the official RPi display it gets special handling by the firmware in the form of loading the overlay.  Off the top of my head I can't remember if there are predicates or if it is unconditional, but it isn't magic.

Gerharddc, 2016-12-02:
@pelwell Thanks, that actually helps a lot :)

pelwell, 2016-12-02:
I checked, and when the firmware detects one of the components in the display it will attempt to load the rpi-ft5406 and rpi-backlight overlays.

Gerharddc, 2016-12-03:
@pelwell Ahhh that makes sense.

It seems I have found the problem after messing around with Raspbian etc. There might have been a lot of other things wrong but it seems the main thing is with the `DTC_FLAGS` you added in your patch. It seems for a clean kernel to build the dtbs properly that have to be added to the makefile in dts and not the one in broadcom. Otherwise it seems that "__symbols__" never makes it into the dtb.

Gerharddc, 2016-12-03:
@pelwell I've been playing around and it seems that the flag has to be in the dts makefile if you want to build the file directly but in your location to be built correctly as part of a full build. Should I add it to the dts folder as well or is there a better way to do it?

pelwell, 2016-12-03:
Submit a PR explaining why you think it is necessary and I'll give it a try.

Electron752, 2016-12-05:
Hi,

I just wanted to add that fixing bcmrpi3_defconfig is very easy.  Just copy over the 4.8 version and run it though "make savedefconfig".  That usually fixes it.

I may submit a PR in the future to fix it on 4.9 once 4.9 is released and declared to be a LTS release.    Although I think having a different config for 64 bit and 32 bit doesn't make sense and is a lot of work to maintain.    I think it would be relatively easy to share the configuration between the two.

Electron752, 2017-01-15:
@Gerharddc 

I've confirmed that the touchscreen overlay is automatically being loaded by the firmware and the touchscreen is working fine for ARM64 in 4.9 and 4.10.   

If things are working for you, perhaps it would make sense to close out the issue?

closed 2017-01-15 by Gerharddc