raspberrypi/linux #1741: Limits on number of connections for the BCM43438

halindrome, 2016-12-02:
Sorry if this is a FAQ but I could not find the information.  Is there a documented limit to the number of WiFi connections the BCM43438 can support with it is used as an Access Point?  And is this a tunable parameter? At runtime or build time?

pelwell, 2016-12-02:
No, it's the first time I've heard the question, and I'm not aware of any documented limits. The brcmfmac driver does have a constant - BRCMF_MAX_IFS, described as the maximum number of interfaces, although it isn't immediately obvious what constitutes an interface - which is hardcoded to 16. Does that sound about right to you?

halindrome, 2016-12-02:
I actually think that is the number of separate wifi boards that it is configured to support. "MAX INTERFACES" is the mnemonic.  It is used that way in core.c in brcmf_ifname. I do see BRCMF_MAX_ASSOCLIST of 128.  I will troll through the code a little but that feels like the right concept... MAX ASSOCIATED LIST.

Ruffio, 2016-12-30:
@halindrome how did your trolling through the code go? Can the issue be closed?

halindrome, 2016-12-31:
I didn't really resolve the question, but yes I am happy to close this.  Thanks!

closed 2016-12-31 by halindrome