raspberrypi/linux #1742: Kernel trace in brcmfmac wifi driver when entering ad-hoc mode

mildmojo, 2016-12-02:
I'm trying to set up a batman-adv mesh between a RasPi B+ v1.2 and a RasPi 2 B v1.1. Each board has wifi through a Redbear IoT HAT which uses the `brcmfmac` driver (like the RasPi 3's wifi chip). I'm running Raspbian Jessie Lite on both, I've updated to latest packages, and I've updated kernel/firmware with `rpi-update`.

On attempting ad-hoc mode, I see an immediate `brcmfmac` kernel trace on the RasPi B+ with kernel 4.4.35 (latest available from running rpi-update). 

The RasPi 2 doesn't exhibit this problem on the same kernel version (4.4.35-v7+). And I haven't thoroughly tested this with whatever kernel is currently shipping through `apt`.

```
# Relevant part of /etc/network/interfaces, from memory (sorry)
allow-hotplug wlan0
iface wlan0 inet6 manual
```

```
$ uname -a
Linux raspberrypi 4.4.35+ #931 Sat Nov 26 13:00:50 GMT 2016 armv6l GNU/Linux
```

```
$ iwconfig wlan0 mode ad-hoc essid my-mesh ap 02:12:34:56:78:9A channel 1
$ dmesg
... <snip> ...
[   62.429425] ------------[ cut here ]------------
[   62.430644] WARNING: CPU: 0 PID: 61 at net/wireless/sme.c:850 cfg80211_roamed+0x88/0x98 [cfg80211]()
[   62.430862] Modules linked in: bnep hci_uart btbcm bluetooth joydev evdev hid_logitech_hidpp 8192cu hid_logitech_dj brcmfmac brcmutil snd_bcm2835 snd_pcm snd_timer cfg80211 snd rfkill bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio batman_adv ipv6
[   62.430995] CPU: 0 PID: 61 Comm: kworker/0:2 Not tainted 4.4.35+ #931
[   62.431012] Hardware name: BCM2708
[   62.431391] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[   62.431586] [<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[   62.431696] [<c0013c20>] (show_stack) from [<c02e43bc>] (dump_stack+0x20/0x28)
[   62.431743] [<c02e43bc>] (dump_stack) from [<c0021ef4>] (warn_slowpath_common+0x8c/0xc4)
[   62.431779] [<c0021ef4>] (warn_slowpath_common) from [<c0021fe8>] (warn_slowpath_null+0x2c/0x34)
[   62.432936] [<c0021fe8>] (warn_slowpath_null) from [<bf10fcfc>] (cfg80211_roamed+0x88/0x98 [cfg80211])
[   62.434429] [<bf10fcfc>] (cfg80211_roamed [cfg80211]) from [<bf2bbf38>] (brcmf_notify_roaming_status+0x134/0x1c0 [brcmfmac])
[   62.435089] [<bf2bbf38>] (brcmf_notify_roaming_status [brcmfmac]) from [<bf2c3b38>] (brcmf_fweh_call_event_handler+0x44/0x9c [brcmfmac])
[   62.435561] [<bf2c3b38>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<bf2c3d10>] (brcmf_fweh_event_worker+0x180/0x358 [brcmfmac])
[   62.435917] [<bf2c3d10>] (brcmf_fweh_event_worker [brcmfmac]) from [<c00376a4>] (process_one_work+0x11c/0x39c)
[   62.436009] [<c00376a4>] (process_one_work) from [<c0037964>] (worker_thread+0x40/0x4d0)
[   62.436058] [<c0037964>] (worker_thread) from [<c003d2e8>] (kthread+0xdc/0xf8)
[   62.436106] [<c003d2e8>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
[   62.436127] ---[ end trace 8b5ff760734b4080 ]---
$
```

(The 8192cu module is loaded because wlan1 is a USB rtl8188cus chip; the trace reproduces with or without the second adapter connected.)

pelwell, 2016-12-02:
You are in luck - we have a couple of the IoT pHATs. It would improve the chances we look at this issue if you give us step be step instructions (or point us at some other step by step instructions to follow).

mildmojo, 2016-12-02:
Sure. I'll try to work out minimal reproduction steps starting from a bare Raspbian Jessie Lite image. Thanks!

mildmojo, 2016-12-04:
It looks like it blows up when connecting to an ad-hoc network with a mismatched MTU, and it's not specific to the IoT pHAT. I've been able to duplicate it on a RasPi 3's built-in wifi under the OSMC distro (kernel 4.4.16-6-osmc).

## Steps to reproduce

0. Create an ad-hoc wifi network with MTU 1532 on a Linux PC or RasPi nearby. Set it aside and leave it running.
  `sudo iwconfig wlan0 mode ad-hoc essid KernelBug ap 02:12:34:56:78:9A channel 1 && sudo ifconfig wlan0 mtu 1532`
1. Install Raspbian Jessie Lite 2016-09-23 on a flash card and install it in a Pi with the Broadcom wifi chip (any Pi with IoT pHAT attached, or just a naked RasPi 3).
2. Boot the Pi.
3. Update it to the latest kernel via `apt`, preferably via ethernet cable so you don't have to modify the stock wifi setup: `sudo apt update && sudo apt install raspberrypi-kernel`
4. Reboot.
5. Connect to the ad-hoc network (MTU defaults to 1500):
`sudo iwconfig wlan0 mode ad-hoc essid KernelBug ap 02:12:34:56:78:9A channel 1`
6. Confirm kernel trace in system log: `dmesg | tail -20`

Stock Raspbian Jessie Lite 2016-09-23 does **_not_** emit a kernel trace. Seems to work normally.

```
$ uname -a
Linux raspberrypi 4.4.21+ #911 Thu Sep 15 14:17:52 BST 2016 armv6l GNU/Linux
```

Other kernels emit the kernel trace on ad-hoc association when the MTU is mismatched:

```
RPi3 + OSMC: Linux raspberrypi 4.4.16-6-osmc #1 SMP PREEMPT Sun Oct 2 13:04:26 UTC 2016 armv7l GNU/Linux
RpiB+ apt upgrade: Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux
RpiB+ rpi-update: Linux raspberrypi 4.4.36+ #933 Fri Dec 2 21:57:25 GMT 2016 armv6l GNU/Linux
```

mildmojo, 2016-12-05:
I think the problem is actually linking two `brcmfmac` devices in ad-hoc mode. I think I left a Pi running in another room while I conducted the first set of tests, throwing my results off.

I've updated the reproduction steps above. Pretty simple; boot Raspbian on two Pis and ad-hoc them together. The second one logs a kernel trace.

Aneke, 2017-02-02:
Was there any follow up on this problem? I am using two RPi 3 as well and cannot associate an ad-hoc network. Both are using the built in brcmfmac drivers as well.

mildmojo, 2017-05-19:
@JamesH65 Sure. I'll try to do this in the next week or two; I haven't had that configuration up and running for a while.

JamesH65, 2017-12-04:
@mildmojo Did you have a chance to try the latest kernel and wireless drivers? I'd like to close this issue if it is fixed.

mildmojo, 2017-12-04:
@JamesH65 I'm sorry for dropping the ball on this. I can't test the new kernel in a timely manner, so I'm going to close the issue. If I ever get a chance to come back to this project and it's still broken, I'll open a new issue. Thanks for your patience.

closed 2017-12-04 by mildmojo

jesswren, 2018-06-17:
I'm still experiencing this issue, using kernel version 4.14.34-v7+

When I try to associate with an AP (created by a host using RTL8188CE, not another `brcmfmac` as @mildmojo experienced), I get the following error:

```
[  411.067206] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  411.214796] ------------[ cut here ]------------
[  411.215037] WARNING: CPU: 0 PID: 1025 at net/wireless/sme.c:942 cfg80211_roamed+0x154/0x158 [cfg80211]
[  411.215043] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[  411.215149] CPU: 0 PID: 1025 Comm: kworker/0:1 Tainted: G         C      4.14.34-v7+ #1110
[  411.215154] Hardware name: BCM2835
[  411.215331] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[  411.215365] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[  411.215384] [<8010c240>] (show_stack) from [<807840a4>] (dump_stack+0xd4/0x118)
[  411.215403] [<807840a4>] (dump_stack) from [<8011da54>] (__warn+0xf8/0x110)
[  411.215417] [<8011da54>] (__warn) from [<8011db3c>] (warn_slowpath_null+0x30/0x38)
[  411.215625] [<8011db3c>] (warn_slowpath_null) from [<7f2254d8>] (cfg80211_roamed+0x154/0x158 [cfg80211])
[  411.215980] [<7f2254d8>] (cfg80211_roamed [cfg80211]) from [<7f3b89b4>] (brcmf_notify_roaming_status+0x178/0x208 [brcmfmac])
[  411.216301] [<7f3b89b4>] (brcmf_notify_roaming_status [brcmfmac]) from [<7f3c208c>] (brcmf_fweh_call_event_handler+0x48/0x88 [brcmfmac])
[  411.216619] [<7f3c208c>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<7f3c22d0>] (brcmf_fweh_event_worker+0x204/0x464 [brcmfmac])
[  411.216789] [<7f3c22d0>] (brcmf_fweh_event_worker [brcmfmac]) from [<80137490>] (process_one_work+0x158/0x454)
[  411.216808] [<80137490>] (process_one_work) from [<801377f0>] (worker_thread+0x64/0x5b8)
[  411.216825] [<801377f0>] (worker_thread) from [<8013d860>] (kthread+0x13c/0x16c)
[  411.216842] [<8013d860>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  411.216868] ---[ end trace e5d38be1e3572d11 ]---
```

JamesH65, 2018-06-18:
Just to confirm, you have an AP on a Pi using a RTL8188CE, and you are attempting to connect to this using another Pi using the onboard wireless, and it's this second Pi that produces the stack trace above?

JamesH65, 2018-06-18:
Just a quick note to myself (or anyone else who fancies taking a look).

Appears that the brcmf_bss_roaming_done function must be calling the cfg80211_roamed function with a incorrect/bad bssid value in the roam_info variable (see drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c!5508). No idea how this could happen., needs debugging.

jesswren, 2018-06-18:
@JamesH65 "Just to confirm, you have an AP on a Pi using a RTL8188CE" ... Actually, it's a laptop using the RTL8188CE which I am connecting to the with the Raspberry Pi (which is using onboard wireless). It is the Pi that is producing the stack trace.

Here is the script I'm running on the Pi if it makes any difference:

```
WIFI_IFACE=wlan0
MESH_AP=02:12:34:56:78:9a
MESH_IFACE=bat0
MESH_IFACE_IP=172.27.0.1/16
CHANNEL=6
ESSID=olymesh

echo "Bringing down wireless interface & killing wpa_supplicant ..."
sudo pkill wpa_supplicant

echo "Configuring wireless interface $WIFI_IFACE as a mesh node ..."
sudo ip link set $WIFI_IFACE down
sleep 1s
sudo iwconfig $WIFI_IFACE mode ad-hoc essid olymesh ap $MESH_AP channel $CHANNEL
sudo ip link set $WIFI_IFACE up
sleep 3s

echo "Activating batman-adv module, and bringing up the mesh interface"
sudo modprobe batman_adv
sudo batctl if add $WIFI_IFACE
sudo ip link set $MESH_IFACE up
```

JamesH65, 2018-06-20:
It would be useful to have a debug log when it goes wrong. Can you try adding

brcmfmac.debug=0x9416 

on the kernel command line (/boot/cmdline.txt). Then posting the log here when it fails.