raspberrypi/linux #1748: No firmware support for linux kernel address randomization

Electron752, 2016-12-05:
I noticed that some of the more recent linux kernels have support for randomizing the address of the linux kernel.   Since people are sort of considering the RPI for server applications, I wonder if this would be a good feature to add to the firmware.

Here is the text from menuconfig.   All it takes is passing a random values in the DT which could be as simple as combining the board serial number with a few power voltage samples.

CONFIG_RANDOMIZE_BASE:                                                                                                                    

Randomizes the virtual address at which the kernel image is 
loaded, as a security feature that deters exploit attempts
relying on knowledge of the location of kernel internals.

It is the bootloader's job to provide entropy, by passing a  
random u64 value in /chosen/kaslr-seed at kernel entry.

mi-hol, 2016-12-05:
@Electron752 firmware is mainted in repository https://github.com/raspberrypi/firmware/issues 
Maybe raising it there would increase likelihood of implementation?

Electron752, 2016-12-06:
OK, I'll repost it on the firmware page.

closed 2016-12-06 by Electron752