raspberrypi/linux #1749: Using a JTAG debugger on Linux Kernels for the RPI

Electron752, 2016-12-05:
I'm not sure if this is the best place to discuss an issue like this.  I know the best place on the regular RPI forum would be the "bare metal" section, but lately they have been kicking out people that discuss debugging the Linux kernel.

I bought the Segger J-Link JTAG debugger so that I can get more enhanced kernel debugging and be able to debug kernel builds that fail very early in the boot sequence.

I found a few articles/papers on the web that explain how to do it.  The main issue is that it appears that the firmware doesn't route the JTAG signals to the GPIO header by default.   It looks like I can write some ASM code in the kernel startup code to enable the JTAG pins, but before I go through the pain of that I'm wondering if anybody knows of an easier way to do it.

For example, does the config.txt have a setting to enable the JTAG pins?

pelwell, 2016-12-05:
Try `enable_jtag_gpio=1`.

Electron752, 2016-12-07:
Yep, that worked. I'm able to debug the RPI 3 with openocd.

Just so others know, the enable_jtag_gpio option enabled the JTAG pins in the "Alt 4" setting.  That's a bit different then some of the examples floating around the web that use a mix of both "Alt 4" and "Alt 5".

closed 2016-12-07 by Electron752

pelwell, 2016-12-07:
Yes, the option selects Alt4 for GPIOs 22-27:

| GPIO | Signal |
|:----:|:------:|
| 22 | TRST |
| 23 | RTCK |
| 24 | TDO |
| 25 | TCK |
| 26 | TDI |
| 27 | TMS |