raspberrypi/linux #1751: Raspberry PI 3 slipped time

ppal-prophecy, 2016-12-07:
It happened with 2 of the PIs. Their time was thrown 2 weeks back (it was showing 23 Nov on 6 Dec).
NTP is configured with default servers (<n>.debian.pool.ntp.org)
NTP daemon was running, ports were open, FW was allowing udp/123.
Internet connectivity was fine as we were receiving control plane and data plane messages from those devices in our cloud servers.

First, at some point of time, they automatically slipped back.
Second, even after reboot, NTP didn't catch up.

We are on 23 Sep 2016 Jessie Lite distro with our custom code that has nothing to do with time
management. Have anyone seen similar issue ?

popcornmix, 2016-12-07:
If ntp is unavailable then use the [fake-hwclock](https://packages.debian.org/sid/main/fake-hwclock) driver which is usually a timestamp from a previous boot which would explain the 2 weeks back date.

Why ntp failed is a more difficult question.

If the time is wrong can you try:
```sudo /etc/init.d/ntp stop && sudo ntpd -gq && sudo /etc/init.d/ntp start```

But really this isn't a linux (kernel) issue. I suspect the problem is somewhere much higher level.

Electron752, 2016-12-08:
Hi @ppal-prophecy:

I just stumbled on this while looking at something else.

I've been hit by similar issues myself.  I think what may be happening is I'm rather certain NTP has a sanity check on the current date/time vs. the time received from the network.    If the times are too far apart but not so far apart that the machine's current time is 1970, then the NTP daemon assumes that it received bad data from the network clock.  In which case it just refuses to set the time.  I know the NTP daemon in most Linux distros have this issue and so does the version of NTP on most versions of Windows.

Two reasons why the initial set of the clock could have failed:
1. You are using DHCP/BOOTP and it takes a long time for the DHCP/BOOTP server to allocate a IP address.   Some implementations can be very slow.
2. Your network switch takes awhile to turn the port on after the RPI boots.  Many newer switches either have an energy detection feature for power saving or they have STP enabled, both of which can cause the port to only be enabled after a long delay.

Popcornmix's "-g" parameter will ignore the sanity check but only for the first time the clock is set.   So you might be able to add it to your NTP daemon's startup parameters. Other then that though, this may well be a limitation of current software.

ppal-prophecy, 2016-12-08:
Hi @Electron752, 

1) Are you saying that NTP runs only once during start-up - so if PI has not got the IP address by then or network switch's spanning tree has not converged by then, NTP request will fail and it will never recover until rebooted or NTP restarted ?
2) I will set the sanity check disable option. Is the "allowable time lag" configurable somewhere in /etc/ntp.conf or somewhere else ?

Electron752, 2016-12-08:
1. No, I'm not saying that.   I'm saying that as popcornmix says if the network isn't available at boot time(no IP, STP not converged, etc), then the system falls backs to fake_hwclock to set the time.  Once that happens, the time is going to be skewed from network time.  Then when the network does kick it, ntp will refuse to update the time because the local skewed time is too different from network time.

2.  Here is the man page for the -g option to ntpd on my debian sid install.  It appears the max lag is 1000s which is hard coded.

> 
       -g     Normally, ntpd exits with a message to the  system  log  if  the
              offset  exceeds the panic threshold, which is 1000 s by default.
              This option allows the time to  be  set  to  any  value  without
              restriction; however, this can happen only once.  If the thresh‐
              old is exceeded after that, ntpd will exit with a message to the
              system log.  This option can be used with the -q and -x options.

> 

Also, as popcornmix says, this really isn't a kernel issue.  It's more of an ntp issue.

Ruffio, 2016-12-30:
@ppal-prophecy has this issue been resolved?

renewagner, 2017-01-15:
This may partially be an instance of [Debian Bug 766838 "ntpdate runs before network is up"](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=766838). I've attached a patch there to fix the if-up.d script.

Ruffio, 2017-02-05:
@ppal-prophecy can this be closed?

Novatec1, 2017-02-06:
Yes, please close it.

closed 2017-02-06 by pelwell