raspberrypi/linux #1754: motion stops with: bcm2835-v4l2: error 0 waiting for frame completion

fhaverkamp, 2016-12-08:
Hi,

I try to use motion with my RaspberryPi model A and a NoIR camera module.
I can open the webinterface and it appears working for a while, but after a few
frames I find the following output in dmesg:

[ 4028.377732] bcm2835-v4l2: error 0 waiting for frame completion

Now the motion webinterface is not working anymore. I tried to restart motion using:

sudo /etc/init.d/motion restart

That somehow reanimates the camera.

Here my version information as far as I gathered it:
pi@raspberrypi4:~ $ uname -a
Linux raspberrypi4 4.4.36+ #933 Fri Dec 2 21:57:25 GMT 2016 armv6l GNU/Linux

pi@raspberrypi4:~ $ v4l2-ctl -V
Format Video Capture:
        Width/Height  : 1024/768
        Pixel Format  : 'YU12'
        Field         : None
        Bytes per Line: 1024
        Size Image    : 1179648
        Colorspace    : Broadcast NTSC/PAL (SMPTE170M/ITU601)
        Flags         :
pi@raspberrypi4:~ $  /opt/vc/bin/vcgencmd version
Dec  3 2016 18:56:05
Copyright (c) 2012 Broadcom
version bc845578aa7adbf453bbda241faadbcde2e670bf (clean) (release)
pi@raspberrypi4:~ $

pi@raspberrypi4:~ $ modinfo bcm2835-v4l2
filename:       /lib/modules/4.4.36+/kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko
version:        0.0.2
license:        GPL
author:         Vincent Sanders
description:    Broadcom 2835 MMAL video capture
srcversion:     A7D6DBEB2D67BE83A6FC776
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-core,v4l2-common
intree:         Y
vermagic:       4.4.36+ mod_unload modversions ARMv6
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           video_nr:videoX start numbers, -1 is autodetect (array of int)
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)

Can you recommend anything stabelizing the camera support?

Regards

Frank

6by9, 2016-12-09:
```[ 3.986852] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720```
```Width/Height : 1024/768```
768 > 720, so it switches to stills mode. Stills mode has the downside of locking AE/AGC, which isn't great for a general surveillance camera.
```sudo mod-probe bcm-2835-v4l2 max_video_width=1024 max_video_height=768``` will increase that threshold.

I thought I'd pushed a patch to count the number of pixels in the frame instead of absolute width/height limits, but it appears not.
I'm sorely tempted to ditch using stills mode from V4L2, but it will take significantly more GPU memory in the process.

fhaverkamp, 2016-12-10:
Hi 6by9,

thanks for the hint. I'll change my /etc/modules to set the max_ parameters and will try if I still get the error.
Regards

Frank

fhaverkamp, 2016-12-11:
Hi 6by9,

it appears that the module parameters have changed the resolution nicely, but it seems that it stays in "stills mode". I also see the error popping up once in a while, sucht that is is not yet stable.

pi@raspberrypi4:~ $ dmesg |grep bcm2835-v
[    4.023688] bcm2835-v4l2: scene mode selected 0, was 0
[    4.024375] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1024x768
[    4.029895] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[   33.561065] bcm2835-v4l2: error 0 waiting for frame completion

Any idea what I should try next? I am not tied to the resolution, so if using a different one works better, it is fine.

Regards

Frank

6by9, 2016-12-11:
No immediate thoughts. motion has a watchdog that kicks in if it fails to receive a buffer for a certain amount of time, and it is the shutdown on that event that normally triggers the issue with an error waiting on frame completion (ie it doesn't get a buffer back when it thought it should).
I'd suggest you try running raspivid for an extended period and confirm that your camera is streaming correctly. The other thing to try is adding ```start_debug=1``` to /boot/config.txt to enable GPU side asserts. You can read those after the event using ```sudo vcdbg log assert```

fhaverkamp, 2016-12-11:
Hi 6by9,

here the 1st results from running into the problem:

001136.117: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
001645.092: assert( card not initialised ) failed; ../../../../../filesystem/media/sdcard/sdcard.c::sdhost_close line 1045 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
006338.513: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1116 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
025752.398: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
032867.437: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
034671.079: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
041086.252: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed

I hope it might help you to figure out what is wrong with my system/setup.
Regards

Frank

fhaverkamp, 2016-12-12:
Using different resolutions does not help. I tried 1280x1020 and saw the same effect.

fhaverkamp, 2016-12-12:
Hi,

I tried as follows:
pi@raspberrypi4:~ $ raspivid -o vid.h264 -t 10000 -h 1020 -w 1280
pi@raspberrypi4:~ $ raspivid -o vid.h264 -t 10000 -h 1020 -w 1280
pi@raspberrypi4:~ $ raspivid -o vid.h264 -t 100000 -h 1020 -w 1280
pi@raspberrypi4:~ $ raspivid -o /dev/null -t 100000 -h 1020 -w 1280
pi@raspberrypi4:~ $ raspivid -o /dev/null -t 10000000 -h 1020 -w 1280

worked ok so far and I got a video which showed something reasonable.
Regards

Frank

fhaverkamp, 2016-12-12:
Just in case, here my motion.conf, now trying a different resolution, but still does not work:

```
# Rename this distribution example file to motion.conf
#
# This config file was generated by motion 3.2.12+git20140228

############################################################
# Daemon
############################################################

# Start in daemon (background) mode and release terminal (default: off)
daemon on

# File to store the process ID, also called pid file. (default: not defined)
process_id_file /var/run/motion/motion.pid

############################################################
# Basic Setup Mode
############################################################

# Start in Setup-Mode, daemon disabled. (default: off)
setup_mode off

# Use a file to save logs messages, if not defined stderr and syslog is used. (default: not defined)
;logfile /tmp/motion.log

# Level of log messages [1..9] (EMR, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). (default: 6 / NTC)
log_level 6

# Filter to log messages by type (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). (default: ALL)
log_type all

###########################################################
# Capture device options
############################################################

# Videodevice to be used for capturing  (default /dev/video0)
# for FreeBSD default is /dev/bktr0
videodevice /dev/video0

# v4l2_palette allows to choose preferable palette to be use by motion
# to capture from those supported by your videodevice. (default: 17)
# E.g. if your videodevice supports both V4L2_PIX_FMT_SBGGR8 and
# V4L2_PIX_FMT_MJPEG then motion will by default use V4L2_PIX_FMT_MJPEG.
# Setting v4l2_palette to 2 forces motion to use V4L2_PIX_FMT_SBGGR8
# instead.
#
# Values :
# V4L2_PIX_FMT_SN9C10X : 0  'S910'
# V4L2_PIX_FMT_SBGGR16 : 1  'BYR2'
# V4L2_PIX_FMT_SBGGR8  : 2  'BA81'
# V4L2_PIX_FMT_SPCA561 : 3  'S561'
# V4L2_PIX_FMT_SGBRG8  : 4  'GBRG'
# V4L2_PIX_FMT_SGRBG8  : 5  'GRBG'
# V4L2_PIX_FMT_PAC207  : 6  'P207'
# V4L2_PIX_FMT_PJPG    : 7  'PJPG'
# V4L2_PIX_FMT_MJPEG   : 8  'MJPEG'
# V4L2_PIX_FMT_JPEG    : 9  'JPEG'
# V4L2_PIX_FMT_RGB24   : 10 'RGB3'
# V4L2_PIX_FMT_SPCA501 : 11 'S501'
# V4L2_PIX_FMT_SPCA505 : 12 'S505'
# V4L2_PIX_FMT_SPCA508 : 13 'S508'
# V4L2_PIX_FMT_UYVY    : 14 'UYVY'
# V4L2_PIX_FMT_YUYV    : 15 'YUYV'
# V4L2_PIX_FMT_YUV422P : 16 '422P'
# V4L2_PIX_FMT_YUV420  : 17 'YU12'
#
v4l2_palette 17

[... 628 lines cut ...]
# Output images to a video4linux loopback device
# The value '-' means next available (default: not defined)
; video_pipe value

# Output motion images to a video4linux loopback device
# The value '-' means next available (default: not defined)
; motion_video_pipe value

##############################################################
# Thread config files - One for each camera.
# Except if only one camera - You only need this config file.
# If you have more than one camera you MUST define one thread
# config file for each camera in addition to this config file.
##############################################################

# Remember: If you have more than one camera you must have one
# thread file for each camera. E.g. 2 cameras requires 3 files:
# This motion.conf file AND thread1.conf and thread2.conf.
# Only put the options that are unique to each camera in the
# thread config files.
; thread /etc/motion/thread1.conf
; thread /etc/motion/thread2.conf
; thread /etc/motion/thread3.conf
; thread /etc/motion/thread4.conf
```

6by9, 2016-12-13:
Nothing in those asserts out of the ordinary.

You can enable more logging from bcm2835-v4l2 by setting module parameter bcm2835_v4l2_debug to 2. There may be something more useful in there.

fhaverkamp, 2016-12-13:
To rule out that the camera might be defective, I plugged it now into a RaspberryPi3 B+ instead of the Raspberry A+ I was using in the failing scenario. That runs now stable using the same configuration for 15min. I will leave that running for some time to see if it stays working before switching back to debug mode 2 on the A+ model.

Is there anything which I might have been missing to configure on the A+ model, which cannot be verified by the data I attached so far? E.g. I have the 1wire bus configured with a DS1820 to sense the temperature, some GPIOs for switches and a LCD 2x16 via GPIOs too? At least I thought, that should not make a difference, since it always works for some minutes before it fails. So something should be right at least. But obviously something it wrong too, which makes it time out and stop working.

6by9, 2016-12-13:
Just checking the simple things - you have got a decent power supply I assume (no lightning bolt on the HDMI output)

fhaverkamp, 2016-12-13:
HDMI is not connected, the power-supply has 5V 2.4A. I used it for the Raspi3 before. It still runs using the Raspi3. I guess I connect it back to the A+ and see what happens with debug=2 as you recommended.

6by9, 2016-12-13:
Check ```vcgencmd get_throttled``` then. 
See https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&start=50#p972790 for full bit meanings, but basically if not coming back as 0 then your PSU isn't up to scratch despite ratings.

fhaverkamp, 2016-12-13:
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50000
```
3322.2222.2222.1111.1111.11
1098.7654.3210.9876.5432.1098.7654.3210
---------------------------------------
0000.0000.0000.0101.0000.0000.0000.0000
```
```
0: under-voltage
1: arm frequency capped
2: currently throttled
16: under-voltage has occurred
17: arm frequency capped has occurred
18: throttling has occurred
```

16 and 18 if I have the bit-order right. Well. Under-voltage sounds strange regarding the 2.4A power-supply. Maybe my switch GPIOs? Or the LCD?

fhaverkamp, 2016-12-13:
I assume the camera is using the 3.3V and not the 5V?

fhaverkamp, 2016-12-13:
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50000
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ 
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50000
pi@raspberrypi4:~ $ 
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ vcgencmd get_throttled
throttled=0x50005
pi@raspberrypi4:~ $ 

Now "under-voltage" and "currently throttled" too. Still, the picture is transmitted. I have turned the python script off, which should normally use GPIOs and the 1-wire port. Let me see how long it survives. The Raspberry3 had no GPIOs connected and no 1-wire port populated.

6by9, 2016-12-14:
I'll agree that it is backward that a Pi3 (which has higher power requirements) works, but an A+ doesn't. Power problems do tend to be a little random in nature though. The USB lead can have as much impact to supply voltage as the PSU itself.

It is 3.3V fed up the ribbon to the camera, which then has regulators to 2.8V and 1.8V on board IIRC. Most of the SoC itself will run with a lower supply voltage than specified, but USB devices (including LAN) tend to fail early, and you then have no headroom for momentary supply voltage sags.

fhaverkamp, 2016-12-14:
I think I could try to detach my GPIO connections to LCD, switches, 1-wire, just in case I did something wrong or if they are too many (well the switches are input and I would hope that the LCD ones are not serious too, but anyways).

I wonder if motion itself produces huge CPU load and therefore power-consumption once it looks at the pictures to detect the "motion"?

That you mention USB and the WLAN connected into it makes me wonder, if I like to try operating the WLAN-stick with an externally powered USB hub, before I rip the thing entirely apart. I mean there is just one USB-connect or for the A+ Raspberry.

6by9, 2016-12-14:
Certainly last time I looked at motion (probably >2 years ago) it was quite processor intensive. It was worse when it was trying to software decode JPEG/MJPEG from the camera, analyse it, and the JPEG the result if it saw motion. It now supports YU12 and various other formats, so isn't quite as crazy (but still does software JPEG for encoding).
My other recollection was that the frame rate parameter only controlled how many buffers it processed, and did not alter the frame rate requested of V4L2. I don't recall looking in to whether that had been fixed.

fhaverkamp, 2016-12-14:
Ok, the experiment with the self-powered USB-Hub for the WLAN card, failed. It timed out again.
The following should be the dump with debug option set to 2:
```
pi@raspberrypi4:~ $ sudo vcdbg log assert
001142.247: assert( source ) failed; ../../../../../middleware/confzilla/cp_front_fdt.c::cp_front_fdt_load_builtin line 115 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
001651.205: assert( card not initialised ) failed; ../../../../../filesystem/media/sdcard/sdcard.c::sdhost_close line 1045 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
006162.354: assert( camera_subsystem_state.power_enable_ref_count[port] > 0 ) failed; ../../../../../vcfw/drivers/device/camera_subsystem/camera_subsystem.c::camera_disable_ctrl line 1116 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
036270.981: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
039576.109: assert( limit_turbo == last_limit_turbo && arm_freq_cap == last_arm_freq_cap ) failed; ../../../../../helpers/arm_loader/arm_loader.c::tempcheck_task line 3181 rev bc84557
vcdbg_ctx_get_dump_stack: dump_stack failed
````

fhaverkamp, 2016-12-14:
I disconnected all GPIOs and the problem still occurs.
I have now set the bcm2835_v4l2 module parameter debug=2 (I had this wrong before).
Results once the error occurs looks as follows:
```
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: buffer_queue: dev:c1ac0800 buf:c4b45200
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: buffer_cb: status:0, buf:c4b45200, length:1382400, flags 4, pts 767779239
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: Convert start time 739.370391 and 741635175 with offset 767779239 to 765.514455
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: buffer_prepare: dev:c1ac0800
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: buffer_queue: dev:c1ac0800 buf:c6952a00
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: buffer_cb: status:0, buf:c6952a00, length:1382400, flags 4, pts 768278979
[Mi Dez 14 22:28:50 2016] bcm2835-v4l2: Convert start time 739.370391 and 741635175 with offset 768278979 to 766.014195
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: buffer_prepare: dev:c1ac0800
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: buffer_queue: dev:c1ac0800 buf:c6abb800
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: buffer_cb: status:0, buf:c6abb800, length:1382400, flags 4, pts 768778721
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: Convert start time 739.370391 and 741635175 with offset 768778721 to 766.513937
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: stop_streaming: dev:c1ac0800
[Mi Dez 14 22:28:51 2016] bcm2835-v4l2: stopping capturing
[Mi Dez 14 22:28:52 2016] bcm2835-v4l2: error 0 waiting for frame completion
[Mi Dez 14 22:28:52 2016] bcm2835-v4l2: disabling connection
[Mi Dez 14 22:28:52 2016] bcm2835-v4l2: Disabling camera
[Mi Dez 14 22:28:52 2016] bcm2835-v4l2: Camera refcount now 0
```

fhaverkamp, 2016-12-14:
Is this just a problem only I have? Or is this camera somehow not really working well with the RaspberryPi A+ model? Or could my RaspberryPi A+ be an unlucky Monday version?

6by9, 2016-12-14:
That log appears to be showing the driver producing frames all the way up to 22:28:51 when stop_streaming is called. That is not indicative of any error on the part of the driver, just that motion isn't wanting more buffers.
In that context the message "error 0 waiting for frame completion" is nothing to worry about, it just means that there wasn't a buffer in flight at the time streaming was stopped.

When you swapped from the A+ to the Pi3, did you use the same uSD card? It sounds like you've got something going wrong in that install of motion, particularly as raspivid is quite happy. Try again from a clean install.

fhaverkamp, 2016-12-15:
In the Pi3 I had a different uSD card. It is one based on RetroPi, but I made sure that in both cases I did a `sudo apt-get update && apt-get dist-upgrade -y' to try the latest version available.
So you suggest to start from scratch and install the whole thing new to outrule and potential installation issue? Whow, that sounds like brute force ;-).

Let me try to turn motions logging to the highest level and see what that says.

JamesH65, 2016-12-15:
You might call it brute force, but could still be quicker than trying to diagnose some tiny version difference in the multitude of packages used by motion (and the rest of the system). Fortunately, quickly building a new SD card is very quick for the Pi, so it's a valid approach.

6by9, 2016-12-15:
Take uSD card out of Pi3 and use that in the A+. Does it work? If so then something is wrong on the uSD originally in the A+. 

Download the latest image from https://www.raspberrypi.org/downloads/raspbian/, install on a uSD card. "sudo apt-get update". "sudo apt-get install motion". Update V4L2 driver params. Drop your motion config file in. Does that work? Sounds like about 30mins work to me.

fhaverkamp, 2016-12-15:
I tried to increase the motion loglevel somewhat more and have set it to 9. I find now the following in my /tmp/motion.log:

```
[1] [DBG] [VID] [Dec 15 19:34:17] v4l2_next: the_buffer index 1 Address (b395c000)
[1] [DBG] [ALL] [Dec 15 19:34:18] alg_labeling: Label: 2 (0) Size: 49889 (253,0)
[1] [DBG] [ALL] [Dec 15 19:34:18] alg_labeling: Label: 3 (1) Size: 750 (189,321)
[1] [DBG] [ALL] [Dec 15 19:34:18] alg_labeling: 2 Labels found. Largest connected Area: 49889 Pixel(s). Largest Label: 2
[1] [DBG] [ALL] [Dec 15 19:34:18] motion_loop: Setup post capture 0
[1] [NTC] [EVT] [Dec 15 19:34:18] event_new_video FPS 2
[1] [INF] [ENC] [Dec 15 19:34:18] ffmpeg_open FPS 2
[1] [ERR] [ENC] [Dec 15 19:34:18] ffmpeg_open: url_fopen - error opening file /var/lib/motion/01-20161215193417.avi ... check access rights to target directory: 
[1] [ERR] [EVT] [Dec 15 19:34:18] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motion/01-20161215193417.avi]: 
[1] [NTC] [ALL] [Dec 15 19:34:18] motion_detected: Motion detected - starting event 1
[1] [ERR] [ALL] [Dec 15 19:34:18] myfopen: Error opening file /var/lib/motion/01-20161215193417-01.jpg with mode w: 
[1] [ERR] [ALL] [Dec 15 19:34:18] put_picture: Can't write picture to file /var/lib/motion/01-20161215193417-01.jpg - check access rights to target directory
Thread is going to finish due to this fatal error: 
[1] [NTC] [ALL] [Dec 15 19:34:18] motion_loop: Thread exiting
[1] [NTC] [STR] [Dec 15 19:34:18] stream_stop: Closing motion-stream listen socket & active motion-stream sockets
[1] [NTC] [STR] [Dec 15 19:34:18] stream_stop: Closed motion-stream listen socket & active motion-stream sockets
[1] [INF] [ALL] [Dec 15 19:34:18] motion_cleanup: Calling vid_close() from motion_cleanup
[1] [NTC] [VID] [Dec 15 19:34:18] vid_close: Closing video device /dev/video0
[0] [NTC] [STR] [Dec 15 19:34:18] httpd_run: motion-httpd - Finishing
[0] [NTC] [STR] [Dec 15 19:34:18] httpd_run: motion-httpd Closing
[0] [NTC] [STR] [Dec 15 19:34:18] motion_web_control: motion-httpd thread exit
[0] [NTC] [ALL] [Dec 15 19:34:20] main: Threads finished
[0] [NTC] [ALL] [Dec 15 19:34:21] main: Motion terminating
[0] [NTC] [ALL] [Dec 15 19:34:21] motion_remove_pid: Removed process id file (pid file).
[0] [NTC] [ALL] [Dec 15 19:34:21] motion_remove_pid: Closing logfile (/tmp/motion.log).
```
I tried now to change the access rights to the /var/lib/motion directory to 777:
```
pi@raspberrypi4:~ $ sudo chmod 777 /var/lib/motion/
```
It ran ok for some minutes now. Could it be that easy? I continue to test it and will reduce all the logging now step by step.

fhaverkamp, 2016-12-15:
I was still running when I left the room.

6by9, 2016-12-15:
Based on the kernel logs, that would fit the symptoms. No actual error, the client (motion) just bailed.

fhaverkamp, 2016-12-16:
Since it seems working now, let me close this bug or was it user-error ;-)? 6by9, thanks for helping to figure out what was wrong!

closed 2016-12-16 by fhaverkamp

andreh7, 2017-07-30:
The `sudo chmod 777 /var/lib/motion/` made `motion` work for more than one second (i.e. it works stably now for me) on a Raspberry Pi3.

crus1313, 2019-01-08:
I logged in via ssh (as admin/pass set in motioneye) but I cannot apply the command. "ls" command shows no folders. It is like I am logged as admin in the "Pictures" :) folder or something, cannot even "cd" to another folders because there aren't any.

wisehackermonkey, 2021-04-08:
### Update: 20210408
# this 
```
sudo chmod 777 /var/lib/motion/
``` 
### worked for me!,
I had to hit disable and enable "Automatic Brightness" motioneye for it to work. 
### Hardware:
```
Raspberry Pi 2
```