raspberrypi/linux #1755: DragonRise USB Driver Issue

DancingPigeon, 2016-12-09:
I've posted about this [here](https://retropie.org.uk/forum/topic/5962/dragonrise-usb-driver-issue/2), but it's an issue more broad than RetroPie:

https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/DragonRise_Inc.___Generic___USB__Joystick__.cfg

> # Note that the linux kernel driver for this controller was broken from
> # kernel 4.4 until 4.9. With the broken driver the joystick axes
> # numbers will be off. This can be fixed by reverting commit
> # 18339f59c3a6698ee17d32970c9e1e450b16e7c3 and rebuilding hid-dr.ko.

pelwell, 2016-12-09:
Since the reversion is already upstreamed I've backported it to rpi-4.4.y and rpi-4.8.y.

DancingPigeon, 2016-12-09:
Fantastic- thank you.

closed 2016-12-09 by DancingPigeon

jangeador, 2016-12-20:
How can I apply this fix to a system?

popcornmix, 2016-12-20:
You can use [rpi-update](https://github.com/Hexxeh/rpi-update) to get the latest testing firmware/kernel which does include this fix.
But be aware that rpi-update firmware/kernel has latest changes in and could have regressions.
The apt-get kernel will be updated at some point (typically every month or two).

DancingPigeon, 2016-12-20:
That explains why apt-get update didn't fix it for me :)

I will run rpi-update sometime in the next few days (after backing up) and report back.1

DancingPigeon, 2016-12-21:
The axes are now fixed, unfortunately Y axis movement is still binary.

popcornmix, 2016-12-21:
@DancingPigeon if there are upstream commits that improve this, then we can consider cherry-picking. If not then it sounds like a problem best reported upstream.

DancingPigeon, 2016-12-21:
@popcornmix I'll look into it. Cheers.