raspberrypi/linux #1757: Debian directory for building Debian packages

stefanberger, 2016-12-09:
I would like to be able to build a recent Debian package. Do you maintain a public package for the 4.8 or 4.9 kernels that contains the debian directory for building debian packages via 'debuild -us -uc'?

popcornmix, 2016-12-09:
No. This git repo is the place to get kernel source.
We don't provide debian packages for the Pi specific kernel source.

stefanberger, 2016-12-09:
I think I found it: mirrordirector.raspbian.org/raspbian/pool/main/l/linux/linux_4.8.7-1.debian.tar.xz

closed 2016-12-09 by stefanberger

dlech, 2016-12-09:
Creating a debian package is built into the kernel source code makefiles. You can do something like this:

    make ARCH=arm CROSS_COMPILE=/path/to/toolchain deb-pkg KDEB_PKGVERSION=1