raspberrypi/linux #1759: Alacarte new item ok button permanently greyed

StuartIanNaylor, 2016-12-09:
Linux raspberrypi 4.4.36-v7+ #933

Happened with the last upgrade as was able to enter menu items,

Run as root or pi same and unable to add new items.

StuartIanNaylor, 2016-12-09:
doh apologies permissions wasn't executable.

closed 2016-12-09 by StuartIanNaylor

pelwell, 2016-12-09:
In case you do find a real issue with Alacarte, you would be better off reporting it in the [raspberrypi/alacarte repo](https://github.com/raspberrypi/alacarte).

StuartIanNaylor, 2016-12-09:
Cool apols as getting to grips with things, will do. Is there an easy way to know what has and has not its own repo ? Where you can do an overall search?

pelwell, 2016-12-09:
[This page](https://github.com/raspberrypi/) is a reasonable place to start, combining a list of our repositories and a search tool.