raspberrypi/linux #1762: Stall on bcm2708_i2c_master_xfer

romanjarosz, 2016-12-12:
Today I noticed that my Raspberry was stuck for 4 hours, from 04:59 to 08:47, than it recovered. From logs it looks like it's some race in i2c. I'm using WiringPi to interact with multiple DS2482-800 and MCP23017 chips. The application is interacting with the chips from multiple threads and is running 24 hours.
The application was running fine from Oct 27 2016 till today, since October I've updated kernel version few times.

Kernel 4.4.26-v7+
[<805b6274>] (__schedule) from [<805b67ec>] (schedule+0x50/0xa8)
[<805b67ec>] (schedule) from [<805b964c>] (schedule_timeout+0x14c/0x270)
[<805b964c>] (schedule_timeout) from [<805b732c>] (wait_for_common+0xbc/0x17c)
[<805b732c>] (wait_for_common) from [<805b742c>] (wait_for_completion_timeout+0x1c/0x20)
[<805b742c>] (wait_for_completion_timeout) from [<7f0ac57c>] (bcm2708_i2c_master_xfer+0x114/0x27c [i2c_bcm2708])
[<7f0ac57c>] (bcm2708_i2c_master_xfer [i2c_bcm2708]) from [<8045ac04>] (__i2c_transfer+0x144/0x50c)
[<8045ac04>] (__i2c_transfer) from [<8045b040>] (i2c_transfer+0x74/0xa0)
[<8045b040>] (i2c_transfer) from [<8045b5c0>] (i2c_smbus_xfer+0x498/0x8ec)
[<8045b5c0>] (i2c_smbus_xfer) from [<7f08f468>] (i2cdev_ioctl_smbus+0x160/0x23c [i2c_dev])
[<7f08f468>] (i2cdev_ioctl_smbus [i2c_dev]) from [<7f08f924>] (i2cdev_ioctl+0xc0/0x270 [i2c_dev])
[<7f08f924>] (i2cdev_ioctl [i2c_dev]) from [<80169b30>] (do_vfs_ioctl+0x424/0x614)
[<80169b30>] (do_vfs_ioctl) from [<80169d64>] (SyS_ioctl+0x44/0x6c)
[<80169d64>] (SyS_ioctl) from [<8000fb40>] (ret_fast_syscall+0x0/0x1c)

[messages.txt](https://github.com/raspberrypi/linux/files/645392/messages.txt)
[kern.log.txt](https://github.com/raspberrypi/linux/files/645393/kern.log.txt)

pelwell, 2016-12-12:
We are just about to move to the 4.8 kernel where we no longer use the downstream bcm2708_i2c driver. You could you try the new kernel - `sudo BRANCH=next rpi-update` - but I think WiringPi may not recognise the board type after the update (we are using more of the upstream code, which causes /proc/cpuinfo to say "Hardware : BCM2835", where wiringPi expects "BCM2708" or "BCM2709"). I've mailed the wiringPi author about this.

romanjarosz, 2016-12-12:
I've updated kernel to 4.8.13-v7+ and WiringPi to 2.36, so far everything looks ok. If anything goes wrong I'll report back.

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65