raspberrypi/linux #1763: Disconnected two wifi cards, interfaces still there, still packets coming in

rodizio1, 2016-12-12:
Had 5 Ralink rt2800 usb wifi cards in monitor mode receiving packets. Disconnected two of them, now the interfaces are still there and there are still packets coming in on those interfaces?

This is what dmesg said after disconnecting the wifi sticks:

```
[  120.797260] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  120.801949] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  120.899006] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  120.911045] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  121.805440] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  121.805999] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  121.905490] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  121.906207] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  122.007107] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  122.112927] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  122.213076] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  122.312989] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  122.413166] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  122.515035] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1314 with error -71
[  122.624650] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1314 with error -71
[  122.729091] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1318 with error -71
[  122.804982] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  122.831344] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1318 with error -71
[  122.905525] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  122.930933] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x131c with error -71
[  123.032947] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x131c with error -71
[  123.133951] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1320 with error -71
[  123.236013] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1320 with error -71
[  123.336400] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1324 with error -71
[  123.436023] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1324 with error -71
[  123.536143] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  123.638850] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  123.799977] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  123.900526] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  124.001123] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  124.102005] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  124.201983] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  124.305815] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1314 with error -71
[  124.405995] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1314 with error -71
[  124.506071] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1318 with error -71
[  124.607373] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1318 with error -71
[  124.707334] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x131c with error -71
[  124.797230] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  124.807199] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x131c with error -71
[  124.898588] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  124.906895] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1320 with error -71
[  125.007669] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1320 with error -71
[  125.109006] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1324 with error -71
[  125.209153] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1324 with error -71
[  125.309096] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  125.420701] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  125.801627] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  125.902454] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  126.797251] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  126.805050] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  126.898010] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  126.905264] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  127.800539] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  127.802177] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  127.900489] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  127.902307] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[  128.001982] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  128.102031] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -71
[  128.203481] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[  128.311276] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1314 with error -71
[  128.411114] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1314 with error -71
[  128.511280] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1318 with error -71
[  128.611011] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1318 with error -71
[  128.712243] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x131c with error -71
[  128.811387] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[... 790 lines cut ...]
[... cut ...]
   19 root      20   0       0      0      0 S   0.0  0.0   0:00.01 ksoftirqd/3                                                                                                                 
   20 root      20   0       0      0      0 S   0.0  0.0   0:00.04 kworker/3:0                                                                                                                 
   21 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 kworker/3:0H                                                                                                                
   22 root      20   0       0      0      0 S   0.0  0.0   0:00.01 kdevtmpfs   
```

rodizio1, 2016-12-12:
Well, after staring at it totally bewildered for some more time (just to repeat: there are packets coming in on interfaces that have been physically disconnected more than 10 minutes ago!), suddenly, it realized that the USB sticks are gone:

```
[ 1486.694066] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x101c with error -71
[ 1486.793849] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1314 with error -71
[ 1486.894321] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1314 with error -71
[ 1486.993992] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1318 with error -71
[ 1487.094250] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1318 with error -71
[ 1487.096971] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1487.194347] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x131c with error -71
[ 1487.197080] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1487.294357] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x131c with error -71
[ 1487.394090] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1320 with error -71
[ 1487.497517] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1320 with error -71
[ 1487.597030] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1324 with error -71
[ 1487.697018] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1324 with error -71
[ 1487.797071] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1487.896961] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1488.092995] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1488.193148] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1489.093016] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1489.101267] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1489.193001] ieee80211 phy0: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1489.200886] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x0438 with error -71
[ 1489.373317] usb 1-1.4: USB disconnect, device number 5
[ 1489.721627] usb 1-1.5: USB disconnect, device number 6
```

P33M, 2016-12-14:
Those aren't packets - the URBs being submitted are returning failure but the driver isn't acting on the fact that the device is no longer there. I would wager that there's some bug in updating the stats counting despite the USB transfer failing.

This is a case of the driver doing a bad thing. On disconnect the RX polling thread isn't getting shut down properly.

rodizio1, 2016-12-15:
The rx application (which fetches incoming packets from the interface via pcap) also still showed incoming packets 10 minutes later.

That application counts packets after it has succesfully identified those packets as "destined to me", with a special fake mac address.

This wasn't just wrong counters I think. It was either packetsbeing stuck in some buffer (for _that_ long? maybe because the load was too high) or those packets were actually coming from other cards and ended up in the data structures of the disconnected cards. Which would be kinda scary. I hope it was just stuck data ...

P33M, 2016-12-19:
Or an alternative explanation - the USB link is flakey in some fashion but enough data is getting through that the device only gets disconnected occasionally.

1. Are you connecting these devices via a powered USB hub or direct to the Pi?
2. If via hub, it needs to be a powered hub with a decent PSU - each of those devices can easily consume 500mA when transmitting.
3. If you have at least some of the devices connected to the Pi directly, what is the output of `vcgencmd get_throttled`? A nonzero value could indicate an undervoltage event - all bets are off in this case.
4. If get_throttled returns 0, can you run only 2 or 3 of the devices connected to the Pi directly without issues?

rodizio1, 2016-12-22:
> the USB link is flakey in some fashion

Yeah, USB is indeed flaky on the Pi. The issue I posted is not the only one, it was just the most bewildering one so far.

To answer your questions:

1. Three sticks connected to a hub, those three sticks are not powered via Pi but directly from a 3 Amp PSU with short proper gauge wires and everything is soldered, no connectors.
The two sticks that I removed were directly connected to the Pi and also powered by it.

2. They do not send any packets at all and consume about 300mA when receiving if I remember correctly

3. Power supply is asolutely stable, no undervoltage. It's the same 3A PSU connected to the Pi GPIO header, also with short thick wires. I have a script running in the background that periodically checks with the vcgencmd get_throttled command if undervoltage or overtemp happened. No problems there.

4. In general, the more USB devices, the more flaky it gets. 
Some examples:

- those Ralink USB wifi cards work pretty well as long as not more than 4 are connected. When I connect five, I sometimes get some "framesize too large" or something error in dmesg, although there are no too large frames being received or transmitted (I am absolutely sure).

- Sometimes, when disconnecting Ralink wifi sticks, (usually only happens when more than three are already connected), the other cards also crash. But the USB stack is still working, ethernet for example is still usable. Sometimes, the whole USB stack or whatever seems to crash, ethernet doesn't work anymore and connecting a USB keyboard also doesn't work.

- With Atheros wifi sticks, it starts getting flaky with more than two sticks. Doesn't matter if these sticks are already connected during boot-up (soldered, and directly powered from the PSU, not from the Pi USB port).  When using three sticks for example, I need to make sure that no other USB traffic is going on while the sticks are being brought up during booting. When an ethernet cable is plugged during boot-up, the whole USB stack crashes and nothing works anymore. When I plug the ethernet cable later, it keeps working. When a USB keyboard is connected together with three Atheros sticks, sometimes the Keyboard behaves erratically, either missing characters or repeating characters. Sometimes the Keyboard just doesn't work at all, all three LEDs light up, and stay lid.

- Four Atheros wifi sticks never work, no matter what, it's completely impossible to get four Atheros sticks working at the same time. It crashes in all kinds of ways, sometimes two adapter "survive" when I plug in the fourth and the Pi stays usable, sometimes the whole thing crashes.

- In general, the Pi seems to be very picky with cables. I know many USB cables are bad quality with too thin wires and bad shielding etc. But: I have several cables that give weird issues (like sudden disconnects or whatever) when being plugged directly between the Pi and USB devices. However, when I plug in any cheap china 5EUR passive USB hub between the Pi and those cheap cables, everything works absolutely stable. The only explanation I have is, that three different cheap crappy passive USB hubs obviously can deal better with "unclean" signals on the data wires (I have also verified that it's not a power problem ...).

- The above issues are not some isolated case on some special combination of hardware of software or something, those problems happen on all Pi's from Pi1 to Pi3 and with all kinds of different kernel versions, Raspbian versions, Firmware versions.

- The same wifi cards. cables, USB hubs (I have a lot ...) etc. work just fine on any other notebook or PC running linux, no problems there.

P33M, 2016-12-22:
Isolating flakiness to just the Pi plus 3-4 peripherals is optimal, because it means I can have a hope of reproducing the issue without speculating about issues arising from PSU/dodgy cables/induced EMI from living next to a steel foundry/phase of the moon.

One thing to try: In /boot/config.txt, add the parameter `usb_mdio=0x7000`. This will reduce the maximum hub depth you can chain to the Pi (to 2 hubs deep) but may improve reliability.

As you can get "flakiness" with just Atheros wifi dongles, do you have dmesg logs from where these dongles are exhibiting "flakiness"? Where can I buy these wifi devices - preferably the exact same model number as the ones in your possession?

rodizio1, 2016-12-23:
Thanks, I'll give that usb_mdio parameter a try.

The Atheros sticks are TP-Link TL-WN722N and Alfa AWUS036NHA with AR9271 chipset, they can be found almost everywhere. 

The Ralink sticks are from CSL and are called "CSL 300 Mbit stick" (with Ralink RT5572 chipset). Not sure if they deliver outside of Germany though:
http://www.ebay.de/itm/300Mbit-WLAN-Stick-mit-Doppel-Antenne-USB-Wireless-Lan-Dongle-2-4GHz-5GHz-/311059044605

You can find similar sticks on Aliexpress:
https://www.aliexpress.com/item/TX-600AN-Ralink-RT5572-300Mbps-dual-band-USB-Wireless-adapter-WIFI-Dongle-Wi-Fi-Receiver-for/32482736817.html

Regarding dmesg logs: Not too much unfortunately, the problem is that often usb keyboard and ethernet don't work anymore, and I cannot see the console because the video is on a layer on front of it. Serial console is also not accessible because the serialport is used for telemetry data. Still need to find some time to setup a testbed with serial console.

Have already opened an issue with the ath9k-htc firmware project, there you can find some logs and USB packet captures.

Shall I enable any kernel debug/tracing features beforehand?

Ruffio, 2017-02-05:
@rodizio1 Can this be closed?

rodizio1, 2017-02-22:
Sorry for the late reply, somehow overlooked your comment. 

The issues are still existent. Still hoping that P33M gets some Ralink and Atheros dongles and looks into it :)

> One thing to try: In /boot/config.txt, add the parameter usb_mdio=0x7000. This will reduce the maximum hub depth you can chain to the Pi (to 2 hubs deep) but may improve reliability.

Have tried this in the meantime, cannot tell a difference.

JamesH65, 2017-05-18:
@P33M Was there any progress on this?

rodizio1, 2017-05-21:
To give an update. There is somebody who has exactly the same issues with Atheros sticks on the Pi, but not on other hardware. See this issue on the ath9k-htc-firmware github page: https://github.com/qca/open-ath9k-htc-firmware/issues/114

This issue is really easy to reproduce. Get 4 Atheros AR9271 based wifi sticks, plug them in. If you're lucky, you get three running, usually you  see all kinds of weird things happening after you plugged the third (also doesn't matter if they are already connected at boot-up). Can be reproduced on all Raspberry Pi versions with all Raspbian versions from (atleast) 2015-12-31 up the the latest.

Another way to make them crash is to put them into promiscous mode, they'll crash in about 1-2 out of 10 times doing that (other people reported the same).

Three Atheros sticks seem to work kind of stable now (that is, after I made sure that nothing else is accessing the usb bus when the sticks are initialized or ifconfig/iwconfig commands are being run, insert a delay between the iwconfig commands and do not put them into promiscous mode ...)

With Ralink sticks you need atleast 6 to make it behave strange and crash.

Mixing Ralink and Atheros works, but only up to two Atheros and two Ralink sticks, anything above that and it crashes and does strange things.

With AR9287 sticks it's even worse, only one works reliably.

P33M, 2017-06-16:
This may be related to the failure mode in #2023 - Atheros dongles specify interrupt endpoints with a period of 1 microframe. For reasons explained in #2023, this causes starvation when servicing other endpoints.

Device disconnects are only detected by the hub, which then raises a port status change flag in its status endpoint (an interrupt endpoint). If the port status is never read, the host doesn't know the device has disappeared.

I would retest this with latest rpi-update and cmdline.txt parameter specified in the linked issue.

JamesH65, 2018-06-28:
@rodizio1 can you test with the latest kernel please, and report back any issues.  

Failing that, this issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

rodizio1, 2018-07-10:
I haven't seen the initial issue (i.e. still packets coming in from nirvana) again.

However, the other issues (see the last post from me in this thread)/still exist.

Absolute maximum is 3 Atheros cards, or 5 Ralink cards, (or 2 Atheros and 2 Ralink), otherwise things get unstable.

CemplexIT, 2018-09-05:
I have a similar issue with 2 RT5370 units and a SimCOM 7100A attached.  Tested with a powered USB hub and without.

[ +0.120031] ieee80211 phy3: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1328 with error -110
[ +0.119963] ieee80211 phy3: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x1328 with error -110

Alexey-Tsarev, 2020-12-03:
The same
```
Linux laser-pi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
Bus 001 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

Dec  2 18:44:57 laser-pi kernel: [1666231.058123] wlancli: deauthenticated from 92:a0:05:92:01:62 (Reason: 1=UNSPECIFIED)
Dec  2 18:44:57 laser-pi wpa_supplicant[483]: wlancli: WPA: Failed to set GTK to the driver (alg=3 keylen=16 keyidx=1)
...
Dec  2 18:48:24 laser-pi kernel: [1666427.720278] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x1700 with error -110
Dec  2 18:48:28 laser-pi wpa_supplicant[483]: wlancli: CTRL-EVENT-BEACON-LOSS.
Dec  2 18:48:28 laser-pi wpa_supplicant[483]: wlancli: CTRL-EVENT-DISCONNECTED bssid=92:a0:05:92:01:62 reason=4 locally_generated=1
```