raspberrypi/linux #1764: RS232 problems in pi3

vay3t, 2016-12-13:
I have the latest version of raspbian (in raspberry pi 3) updated and I have problems with RS232. I have a pl2303 and it works fine. Any idea of the problem?

used commands in pi3:

```bash
sudo su
apt-get update
apt-get dist-upgrade -y
echo "enable_uart=1" >> /boot/config.txt
```

used commands in debian:

```bash
sudo su
screen /dev/ttyUSB0 115200
```

pelwell, 2016-12-13:
What is in your cmdline.txt?

Can you explain how you use the pl2303?

vay3t, 2016-12-15:
i need connect to rpi for serial console. like this tutorial: https://learn.adafruit.com/adafruits-raspberry-pi-lesson-5-using-a-console-cable?view=all but i need use serial port (not usb2ttl)

```
pi@raspberrypi:~ $ cat /boot/cmdline.txt 
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```

vay3t, 2016-12-30:
1) no output
2) yes
3)
```
root@vay3t:~# stty -F /dev/ttyUSB0 -a
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^H; eof = ^D; eol = <undef>;
eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R;
werase = ^W; lnext = ^V; discard = ^O; min = 100; time = 2;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk brkint ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl ixon -ixoff
-iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon iexten -echo echoe echok -echonl -noflsh -xcase -tostop -echoprt
echoctl echoke -flusho -extproc
```

pelwell, 2016-12-30:
That looks OK - in particular, flow control is not enabled. Can you post the output from `dmesg`, ideally shortly after booting?

vay3t, 2016-12-30:
dmesg in rpi or pc?

pelwell, 2016-12-30:
Sorry - on the Pi. I'm trying to see if the correct UART is being enabled and used. You could also run `raspi-gpio` to see if the pins are configured correctly - GPIOs 14 and 15 are the important ones.

vay3t, 2016-12-30:
```
pi@vay3t:~ $ sudo raspi-gpio get 14
GPIO 14: level=1 fsel=2 alt=5 func=TXD1
pi@vay3t:~ $ sudo raspi-gpio get 15
GPIO 15: level=1 fsel=2 alt=5 func=RXD1
```

dmesg rpi > https://gist.githubusercontent.com/vay3t/58a14eba7df5563c0a1de99d048fec8f/raw/94b63fb84936c0ca6f2d5d40e49def8c53068233/rpidmesg.txt

pelwell, 2016-12-30:
The raspi-gpio output shows that UART1 (ttyS0) is mapped to those pins. The dmesg output (console=ttyS0,115200) confirms it. This in itself shouldn't prevent the serial port from working, but it shows that the "dtoverlay=pi3-disable-bt" hasn't worked. What does:
```
sudo vcdbg log msg |& grep -v -E "(HDMI|gpioman)"
```
show?

vay3t, 2016-12-30:
i understand. i need add this option in config.txt?
```
pi@vay3t:~ $ sudo vcdbg log msg |& grep -v -E "(HDMI|gpioman)"
001149.191: *** Restart logging
001150.841: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
001405.938: Loading 'kernel7.img' to 0x8000 size 0x409058
001409.593: No kernel trailer (run mkknlimg to fix) - assuming DT-capable
001413.534: Loading 'bcm2710-rpi-3-b.dtb' to 0x411058 size 0x3e78
001510.074: dtparam: audio=on
001521.311: dtparam: i2c1=on
001531.390: dtparam: i2c_arm=on
001683.017: Device tree loaded to 0x2effbc00 (size 0x4380)
002488.672: vchiq_core: vchiq_init_state: slot_zero = 0xfac80000, is_master = 1
002499.328: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000052
```

pelwell, 2016-12-30:
Sorry, forget about pi3-disable-dt - that was a different thread. If you didn't want to use Bluetooth and you wanted the best system performance then pi3-disable-bt might be useful, but it certainly isn't necessary.

You say your wiring is correct, but it is worth swapping the non-black wires over to be sure. You can force output from the Pi serial port like this:
```
sudo sh -c "echo hello >/dev/serial0"
```

vay3t, 2016-12-30:
when i exec this command in rpi3 `sudo sh -c "echo hello >/dev/serial0"` my screen terminal print > \n

pelwell, 2017-01-17:
Have you made any progress on this? I have the same setup as you, except I've plugged the USB serial adapter back into the Pi, and everything works as expected. Perhaps you can try the same?

Try echoing longer strings - do you get any characters through? If you type things into "screen" do you ever see any output?

vay3t, 2017-01-17:
Unfortunately I still can not make progress. Only USB works

pelwell, 2017-01-17:
Can you remove the "plymouth.ignore-serial-consoles" from your cmdline.txt, then reboot and see it works?

pelwell, 2017-01-18:
Ignore the suggestion to remove plymouth.ignore-serial-consoles - it will cause extra output on the serial port, but it won't solve your problem.

Starting from a clean Raspbian installation:

1. Use raspi-config to enable the serial port (or add `enable_uart=1` to config.txt) and reboot.
2. `sudo apt-get install screen`
3. Connect the three wires of the RS232 adaptor to pins 6, 8 and 10 of the Pi's header, and plug the USB plug into a free port on the Pi.
4. `screen /dev/ttyUSB0 115200`

If this doesn't work for you then either your USB-serial cable is faulty or you haven't connected it correctly. If this does work then there is nothing wrong with the Pi or the cable, so try with your other host. Either way, I don't think this is a kernel issue - if you still require assistance try on the Forums - [Interfacing](https://www.raspberrypi.org/forums/viewforum.php?f=44) or [Advanced Users](https://www.raspberrypi.org/forums/viewforum.php?f=29).

Ruffio, 2017-02-05:
@vay3t have you tried @pelwell solution? Did it work?

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-05-18 by JamesH65

bberak, 2019-11-16:
Hi all,

I think I'm facing a similar problem - can't get any console output via serial connection to my Mac. I've got a Raspberry PI 3 B+. I'm not really a hardware guy, but have tried to follow the discussion above. I'm using Raspian Stretch from June 2018.

My config.txt:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable UART (serial communication)
enable_uart=1
```

cmdline.txt:

```
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=d661f1aa-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
```

Here is the GPIO information, I noticed the different levels, but am not sure what that means.

```
sudo raspi-gpio get 14
GPIO 14: level=0 fsel=2 alt=5 func=TXD1
sudo raspi-gpio get 15
GPIO 15: level=1 fsel=2 alt=5 func=RXD1
```

I've also connected the serial usb cable to one of the usb ports on the Pi itself using `screen /dev/ttyUSB0 115200` and sent jibberish down the line using `sudo sh -c "echo hello >/dev/serial0"` - but the screen terminal just remained blank.

Any help would or pointers would be really appreciated. Please let me know if I can provide more info.