raspberrypi/linux #1766: RC522 not working with RPI 3 model B (2016-11-25-raspbian-jessie)

medyas, 2016-12-14:
Hi, i just got my RPI3 and wanted to make a project with RC522, i installed raspbian the latest version (2016-11-25-raspbian-jessie), i followed alot of blogs and sites, and still have the same problem, it's not showing anything when executing the code. i looked online and in forms and i think it spi problem, cause even after enabling the spi, i found this in a blog : 
`$ dmesg | grep spi
[    5.408904] bcm2708_spi 20204000.spi: master is unqueued, this is deprecated
[    5.659213] bcm2708_spi 20204000.spi: SPI Controller at 0x20204000 (irq 80)`

 but when i do it in my terminal i get nothing, with` lsmod  spi_bcm2835             6678  0 `
and` ls /dev/spi*   /dev/spidev0.0  /dev/spidev0.1`

i did all the updates  and upgrade, also i did firm update, I installed the SPI-Py , python-dev , checked my wires and still got nothing.

Thank you for the help.

pelwell, 2016-12-14:
Sorry for the terse reply - I'm out of the office - but read my post here (
https://github.com/raspberrypi/linux/issues/1547#issuecomment-245220114).

pelwell, 2016-12-30:
Did you get the RC522 working?

pelwell, 2017-01-17:
The latest comments in #1547 confirms again that the instructions there are correct if you follow them very carefully. Feel free to comment there if you are still having difficulties.

closed 2017-01-17 by pelwell

medyas, 2017-01-18:
Hi, sorry for not replying. I have not tested it again but will do next couple of days and will reply with the result. Thank you.