raspberrypi/linux #1767: Can not compile successfully

eeyrw, 2016-12-19:
I just follow the [official cross build instruction](https://www.raspberrypi.org/documentation/linux/kernel/building.md) and met the error like these: 

*** No rule to make target `fs/cachefiles/xattr.o', needed by `fs/cachefiles/cachefiles.o'.  Stop

drivers/char/hw_random/core.c:35:29: fatal error: linux/hw_random.h: No such file or directory
 #include <linux/hw_random.h>
                             ^
compilation terminated.

net/netfilter/xt_DSCP.c:20:37: fatal error: linux/netfilter/xt_DSCP.h: No such file or directory
 #include <linux/netfilter/xt_DSCP.h>
                                     ^

My host linux is ubuntu 14.04 LTS 64bit.

pelwell, 2016-12-19:
Please include the exact commands you ran - use the history command if you still have the shell. We cross compile all the time, so there is probably something about your environment or the commands you ran that caused the error.

eeyrw, 2016-12-19:
I have tried a lot ways which don't make it works, feeling so frustrated, just turn off my virtual machine... So I have retried and update the essential info to the post. Before retrying I ran the command "make mrproper" to clean the working place.

eeyrw, 2016-12-19:
I examine the error:
*** No rule to make target `fs/cachefiles/xattr.o', needed by `fs/cachefiles/cachefiles.o'.  Stop
It looks like missing file... But I've tried extract files many times.....????
[image]

pelwell, 2016-12-19:
Yes, it looks like xattr.c has got lost somehow.

We have observed that zip downloads can be unreliable,  so I would recommend trying to use the git clone method; with the --depth=1 option the download won't be quite so large. I expect that to work as advertised.

eeyrw, 2016-12-19:
It's not the faculty of zip file. The unzip has some bug mentioned in this issue:https://github.com/raspberrypi/linux/issues/34
I change to the 7z and it works fine.

pelwell, 2016-12-19:
Please close the issue if you feel it is resolved.

closed 2016-12-20 by eeyrw