raspberrypi/linux #1769: Exim4 directory permissions

jucajuca, 2016-12-21:
apparently the directory permissions for exim4 are not correct. I got this messsages:

can be fixed easily with chmod 755  for the given directory tree.

Dec 19 16:24:09 fileserver exim[305]: 2016-12-19 16:24:05 1cIznJ-00004v-CN Failed to create spool file /var/spool/exim4/input//1cIznJ-00004v-CN-D: Read-only file system
Dec 19 16:24:09 fileserver exim[305]: 2016-12-19 16:24:05 1cIznJ-00004v-CN Cannot open main log file "/var/log/exim4/mainlog": Read-only file system: euid=114 egid=122
Dec 19 16:24:09 fileserver exim[305]: exim: could not open panic log - aborting: see message(s) above

vk5tu, 2016-12-22:
"Read-only file system" suggests the filesystem was mounted read-only when exim was started. As a result exim could not write to files in the directories you list. A common cause for a filesystem being mounted read-only is that the fillesystem has reported errors. Have a look within the output of `dmesg` for I/O errors and in the system log for messages concerning mounts and re-mounts.

If you fix the filesystem issue so that the filesystem mounts for read-write and allows setuid then exim will work as normal.

The suggested workaround may allow people to read the e-mail of others. You may wish to return the directory permissions back to `drwxr-x--- Debian-exim:Debian-exim`.

This list and bug system is for the Linux kernel running on the RaspberryPi. Debian and Raspbian run their own bug systems. If you feel this issue needs pursuing further then the Debian Bug Tracking System is likely to be the best forum.

Ruffio, 2017-02-05:
@jucajuca can this be closed?

jucajuca, 2017-02-05:
Yes, I think so

- - -
Juan Carlos Carvajal Bermúdez
PhD researcher
Austrian institute of technology
+43 650 4770005
www.ait.ac.at
www. juan-carlos. info

closed 2017-02-05 by popcornmix