raspberrypi/linux #1772: where is irqs.h

mayurcools, 2016-12-27:
where is mach/irqs.h for bcm2708/9

pelwell, 2016-12-27:
All irqs should be allocated via Device Tree now. What are you trying to build?

mayurcools, 2016-12-27:
I already developed a character device driver which uses gpio bit banging along with the timer interrupts, so earlier in my code Inwas using request_irq method and pass IRQ_TIMER1 and IRQ_TIMER3. But now it is throwing error as it is undefined.

Is there any way that I could get around to it

pelwell, 2016-12-27:
Have a look at the i2c-gpio driver - it should be a good example of the current way to handle gpios and interrupts. If that is too high level, try pinctrl-bcm2835 (https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/pinctrl/bcm/pinctrl-bcm2835.c).

Ruffio, 2017-02-05:
@mayurcools can this be closed?

closed 2017-02-08 by mayurcools