raspberrypi/linux #1773: rpi-4.9.y compilation issues

igoropaniuk, 2016-12-29:
Hi there,

I'm encountering compilation issues in `pinctrl-bcm2835` kmod, while trying to build `rpi-4.9.y`.
Previous kernel version (`rpi-4.8.y`) is built without any issues. Am I doing something wrong?
I'm using [linaro-gcc 5.3-2016.05](http://releases.linaro.org/components/toolchain/binaries/5.3-2016.05/aarch64-linux-gnu/) toolchain and default `/arch/arm64/configs/bcmrpi3_defconfig` config.

Output:
```
drivers/pinctrl/bcm/pinctrl-bcm2835.c:106:19: error: field 'gpio_chip' has incomplete type
  struct gpio_chip gpio_chip;
                   ^
drivers/pinctrl/bcm/pinctrl-bcm2835.c: In function 'bcm2835_gpio_direction_input':
drivers/pinctrl/bcm/pinctrl-bcm2835.c:336:42: error: dereferencing pointer to incomplete type 'struct gpio_chip'
  return pinctrl_gpio_direction_input(chip->base + offset);
                                          ^
drivers/pinctrl/bcm/pinctrl-bcm2835.c: In function 'bcm2835_gpio_get':
drivers/pinctrl/bcm/pinctrl-bcm2835.c:341:31: error: implicit declaration of function 'gpiochip_get_data' [-Werror=implicit-function-declaration]
  struct bcm2835_pinctrl *pc = gpiochip_get_data(chip);
                               ^
...
```

And a second question:
Do BCM2837 somehow support ARMv8.1/ARMv8.2 arch features? I've found a commit https://github.com/Electron752/linux/commit/4b23053b1b945bf74e86dde26788b4ab99aa44ef, where all these ARMv8.1/ARMv8.2 features were intentionally enabled (regardless that fact that there is no any information about support by ARM_Cortex-A53, which BCM2837 is based on)
```

+#
+# ARMv8.1 architectural features
+#
+CONFIG_ARM64_HW_AFDBM=y
+CONFIG_ARM64_PAN=y
+CONFIG_ARM64_LSE_ATOMICS=y
+CONFIG_ARM64_VHE=y
+
+#
+# ARMv8.2 architectural features
+#
+CONFIG_ARM64_UAO=y
+CONFIG_ARM64_MODULE_CMODEL_LARGE=y
+CONFIG_RANDOMIZE_BASE=y
```

aBradleyUno, 2016-12-30:
Try adding CONFIG_ARCH_BCM2835=y to arch/arm64/configs/bcmrpi3_defconfig, I've noticed there are some things missing

igoropaniuk, 2016-12-31:
Hi @aBradleyUno ,

Adding CONFIG_ARCH_BCM2835=y fixed my probrem

Thanks

closed 2016-12-31 by igoropaniuk

popcornmix, 2016-12-31:
I've added +``CONFIG_ARCH_BCM2835=y`` to arch/arm64/configs/bcmrpi3_defconfig