raspberrypi/linux #1774: V4L2 driver clips max frame rate to 90fps. V2 sensor can do 120fps

6by9, 2016-12-29:
(Mainly a note to myself)
Due to https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/media/platform/bcm2835/bcm2835-camera.c#L86 the frame rate range is fixed at 1-90 fps but the V2 sensor can do 120fps (just tested and got I420 960x720 with stock settings. 720P drops to 107fps).
Ought to switch between 90 and120 depending on whether a V1 or V2 sensor is found (we already adjust the max resolution).

JamesH65, 2017-09-13:
@6by9 Any further movement?

6by9, 2017-09-13:
Not yet.

compiaffe, 2023-01-17:
@6by9 I see even in version 6 of the kernel the frame rate is still limited to 90 fps: https://github.com/raspberrypi/linux/blob/rpi-6.0.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L68

 You mention that you already adjust the max resolution based on V1 or V2. Could you point to where you are doing that?

6by9, 2023-01-17:
> You mention that you already adjust the max resolution based on V1 or V2. Could you point to where you are doing that?

get_num_cameras (https://github.com/raspberrypi/linux/blob/rpi-6.0.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L1492) called from bcm2835_mmal_probe returns the max resolution of each camera, and that is then stored for future use in clipping the requested resolution.

The legacy stack has been superceded by libcamera which does allow 120fps on a V2 camera, therefore this isn't going to be a priority to fix, hence closing.

closed 2023-01-17 by 6by9