raspberrypi/linux #1775: HiFiBerry Amp driver missing in 4.9

usul27, 2017-01-02:
It seems that the HiFiBerry Amp driver sound/soc_bcm/hifiberry_amp got lost in the 4.9 tree. It is still included in 4.8.
Can you merge the module to 4.9?

popcornmix, 2017-01-02:
It is currently reverted due to upstream API changes stopping it building.
Ping @hifiberry, @HiassofT for suggestions on a fix to get this enabled again.

Error:
```
sound/soc/codecs/tas5713.c:235:2: error: unknown field 'controls' specified in initializer
  .controls = tas5713_snd_controls,
  ^
sound/soc/codecs/tas5713.c:235:2: warning: initialization from incompatible pointer type
sound/soc/codecs/tas5713.c:235:2: warning: (near initialization for 'soc_codec_dev_tas5713.suspend')
sound/soc/codecs/tas5713.c:236:2: error: unknown field 'num_controls' specified in initializer
  .num_controls = ARRAY_SIZE(tas5713_snd_controls),
  ^
sound/soc/codecs/tas5713.c:236:101: warning: initialization makes pointer from integer without a cast
  .num_controls = ARRAY_SIZE(tas5713_snd_controls),
                                                                                                     ^
sound/soc/codecs/tas5713.c:236:101: warning: (near initialization for 'soc_codec_dev_tas5713.resume')
scripts/Makefile.build:293: recipe for target 'sound/soc/codecs/tas5713.o' failed
```
presumably due to this: https://patchwork.kernel.org/patch/9269067/

HiassofT, 2017-01-02:
With the ASoC component refactoring the .control etc fields were moved from snd_soc_codec_driver to snd_soc_component_driver. The codec driver needs to be changed (put .controls etc inside the .component_driver struct), like eg this one: d25f113f969eb9a0ae3a61d3937a0fbfe7ecdbdd

@hifiberry could you a functional test with this change?

HiassofT, 2017-01-02:
With this patch tas5713 compiles again, but I can't test myself as I don't have such a card
```
--- a/sound/soc/codecs/tas5713.c
+++ b/sound/soc/codecs/tas5713.c
@@ -232,8 +232,10 @@ static int tas5713_probe(struct snd_soc_codec *codec)
 static struct snd_soc_codec_driver soc_codec_dev_tas5713 = {
        .probe = tas5713_probe,
        .remove = tas5713_remove,
-       .controls = tas5713_snd_controls,
-       .num_controls = ARRAY_SIZE(tas5713_snd_controls),
+       .component_driver = {
+               .controls = tas5713_snd_controls,
+               .num_controls = ARRAY_SIZE(tas5713_snd_controls),
+       },
```

popcornmix, 2017-01-02:
@usul27 are you able to test @HiassofT's suggested patch?
You should drop https://github.com/raspberrypi/linux/commit/57d16402f799ce6bdda8803d9d9f9cba7c60b189
If you confirm it works I can add it to tree.

hifiberry, 2017-01-02:
I will look into this in the next days.

hifiberry, 2017-01-08:
Tried the first time to compile the kernel directly on the Raspberry Pi ;-) After some hours (had to compile it multiple times), I can confirm now that the patch from @HiassofT works. 

@popcornmix Can you please integrate this into the 4.9 tree ?

pelwell, 2017-01-09:
Done.

ckarrie, 2017-01-09:
Awesome, thank you!

popcornmix, 2017-01-09:
Fix is already in 4.10 tree.
@usul27 okay to close?

usul27, 2017-01-09:
Yes, please close this. Thanks to everyone who worked on this.

closed 2017-01-09 by popcornmix

cjbrigato, 2017-01-09:
- Colin J. βrigato -
  ¥ngénieur Σystème & Δirecteur de Projet
          ✆ 06 43 19 33 49

Closed #1775 <https://github.com/raspberrypi/linux/issues/1775>.

<https://github.com/raspberrypi/linux/issues/1775#event-916540956>, or mute
the thread