raspberrypi/linux #1776: support VIDIOC_CROPCAP and VIDIOC_S_CROP in v4l2 kernel module for bcm2835

zoff99, 2017-01-04:
support VIDIOC_CROPCAP and VIDIOC_S_CROP
in the video4linux2 kernel module of the camera
https://github.com/raspberrypi/linux/tree/rpi-4.4.y/drivers/media/platform/bcm2835

popcornmix, 2017-01-04:
Ping @6by9

zoff99, 2017-01-04:
https://kernel.org/doc/html/latest/media/uapi/v4l/vidioc-g-crop.html

zoff99, 2017-01-04:
if i hardcode my wanted crop values here:
https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/media/platform/bcm2835/bcm2835-camera.c#L1072
and compile the kernel module, will this actually crop? or is more needed?

6by9, 2017-01-04:
This has been kicking around for a while and I've just not had the chance to follow it through.
byterg on the forums had put together a patch (https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=62364&p=779385#p779385) which I had tidied up but never really tested.

I've pushed the change I've had locally for a while up onto my github repo (https://github.com/6by9/linux/, rpi-4.4.y branch), but it's likely to be incorrect. (There is also a second in progress change there that is unrelated but I couldn't be bothered to separate out. It should be innocuous but didn't work for RGB3/BGR3 formats due to an incorrect buffer size).
The weird bit is around what to do if the resolution gets changed and the values to set up with VIDIOC_CROPCAP https://kernel.org/doc/html/latest/media/uapi/v4l/vidioc-cropcap.html#c.v4l2_cropcap

I had a system up using QV4L2 the other day, so I'll try to resurrect the patches and work out what the correct answer is. It may be that we can fix the CROPCAP numbers at some arbitrary value (max resolution?) and then scale, but I don't know if that is the correct answer or not.
It needs to pass v4l-compliance too.

JamesH65, 2017-05-18:
@6by9 I suspect this can be closed....

6by9, 2017-05-18:
Please leave it open.
It should really be titled "support the V4L2 selection API" now due to the cropping API being deprecated.

JamesH65, 2017-05-18:
Changed title...

JamesH65, 2017-09-13:
ping @6by9

6by9, 2017-09-13:
No, no movement.

Serveurperso, 2019-04-10:
+1 digital zoom (crop) on robots is an awesome feature! help ?

Snyp54, 2019-04-15:
+1 for "crop" function :).

6by9, 2019-04-16:
+1's only annoy.
I still haven't managed to get my head around the selection API and how it would apply to the Pi camera configuration. It'll happen when it happens.

Serveurperso, 2019-04-16:
How to support you? there is a way to donate?

pelwell, 2019-04-16:
We are (mostly) all paid employees of Raspberry Pi, and if we aren't working on your issue that's because we are committed to other, higher priority tasks. The only real way to support us is to do the work (or pay somebody else to) and submit a pull request - it sounds like the cropping could be done in the kernel driver, although that might not be the most efficient implementation, so all the necessary information should be available.

6by9, 2023-11-06:
Closing as not planned. The legacy camera stack is deprecated, and cropping is supported by libcamera.

closed 2023-11-06 by 6by9