raspberrypi/linux #1780: firmware-brcm80211 breaks wlan0 on Raspbian

fkbreitl, 2017-01-06:
Package: firmware-brcm80211
Version: 20161130-2 (testing)

After upgrading Raspbian from jessie to testing wlan0 disappears on a Raspberry Pi 3 with a similar error message reported in e.g. Debian Bug #821400 or at https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138629&start=25:
brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

After downgrading to version: 043+rpi5 (stable) wlan0 appears and works again.

043+rpi5 (stable) provides the same version of brcmfmac43430-sdio.bin found at https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm80211/brcm/. 20161130-2 (testing) provides something else that doesn't work on a Raspberry Pi 3.

I wonder if this package could be changed to have wireless working again.

I am using Raspbian GNU/Linux stretch/sid with kernel 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l

I reported this problem to Debian because the Raspbian packet states:

`Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>`

However I was told this might not be correct either. See

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=850380.

popcornmix, 2017-01-06:
This is a packaging question, not a linux kernel issue.
I suspect this firmware should come from the raspberrypi repo.
Best to create an issue here: https://github.com/RPi-Distro/repo where the right dev will see it.

closed 2017-01-06 by popcornmix

fkbreitl, 2017-01-06:
Alright, done!

https://github.com/RPi-Distro/repo/issues/51