raspberrypi/linux #1781: Adding driver M25P80 to bcmrpi_config

Naegolus, 2017-01-07:
Hello,

we (a little open source hardware startup) would like to add an existing driver in the bcmrpi_config.

Motivation:
Our first product is an extension board for the Raspberry Pi Zero and Raspberry Pi A+.
The extension board adds an ethernet interface (driver situation ok) and optionally connects the Raspberry Pi with a FPGA Board which has a M25P80 compatible SPI flash on it. If the M25P80 driver is activated the FPGA can be programmed via the flash tools from the package mtd_utils.
The users of our extension board should be able to install it by just adding a device tree overlay. Unfortunately the needed M25P80 driver isn't included in the official Raspbian image.

Specific changes:
--- a/arch/arm/configs/bcmrpi_defconfig
+++ b/arch/arm/configs/bcmrpi_defconfig
@@ -400,7 +400,10 @@ CONFIG_DMA_CMA=y
 CONFIG_CMA_SIZE_MBYTES=5
 CONFIG_MTD=m
 CONFIG_MTD_BLOCK=m
+CONFIG_MTD_M25P80=m
 CONFIG_MTD_NAND=m
+CONFIG_MTD_SPI_NOR=m
+# CONFIG_MTD_SPI_NOR_USE_4K_SECTORS is not set
 CONFIG_MTD_UBI=m
 CONFIG_OF_CONFIGFS=y
 CONFIG_ZRAM=m

May we activate this?

Joe

pelwell, 2017-01-17:
Enabling those options causes two modules to be built: spi-nor.ko (19kB) and m25p80.ko (13kB). There are no

@popcornmix Are you happy for me to proceed?

popcornmix, 2017-01-17:
Okay.

pelwell, 2017-01-17:
Those options are now enabled in rpi-4.4.y, rpi-4.9.y and rpi-4.10.y, and will be included in future builds.

Naegolus, 2017-01-17:
Thank you very much!

closed 2017-01-17 by Naegolus

meerd, 2017-02-23:
Hello @Naegolus,

Do you have working overlay (dts) file for m25p80?

Thanks.

Naegolus, 2017-02-23:
Hello @meerd,

Sure. Here: https://github.com/Naegolus/rr_base/blob/master/raspberry_pi_zero/dts/overlays/dspc-de0_nano-overlay.dts

Best regards

meerd, 2017-02-23:
Hello @Naegolus! 

Thanks for your reply. I have w25q128 connected to my board (based on compute module) and I have been trying to integrate it with raspberry pi. Electrically, w25q128 is OK but for some reason the kernel does not recognize the it. Please see my overlay file. Can you please review it?  

`
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&gpio>;
		__overlay__ {
			spi2_pins: spi2_pins {
				brcm,pins = <40 41 42>;
				brcm,function = <3>; /* alt4 */
			};

			spi2_cs_pins: spi2_cs_pins {
				brcm,pins = <43>;
				brcm,function = <1>; /* out */
			};
		};
	};

	fragment@1 {
		target = <&spi2>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
			pinctrl-0 = <&spi2_pins &spi2_cs_pins>;
			cs-gpios = <&gpio 43 1>;
			status = "okay";

			flash0: w25q128@0 {
				compatible = "w25q128, m25p128-nonjedec", "spi-nor";
				reg = <0>; /* CE0 */
				#address-cells = <1>;
				#size-cells = <0>;
				spi-max-frequency = <12000000>;
				status = "okay";

				partition@0 {
					label = "data";
					reg = <0x0 0x8000000>;
				}; 
			};
		};
	};
};
`

meerd, 2017-02-24:
After many unsuccessful attempts, it is working now! I am aiming to add this to the kernel repository after a few improvements. For now, the example needs to be used with spi2-1cs overlay.

dtoverlay=spi2-1cs,cs0_spidev=disabled            
dtoverlay=w25q128

`/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709", "brcm,bcm2710";

	fragment@0 {
		target = <&spi2>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			flash: m25p80@0 {
				#address-cells = <1>;
				#size-cells = <1>;
				compatible = "w25q128","spi-nor";
                                reg = <0 0x800000>;
                                linux,modalias = "m25p80", "w25q128";
				spi-max-frequency = <1000000>;
				m25p,fast-read;

				partition@0 {
					label = "data";
					reg = <0x0 0x8000000>;
				}; 
			};
		};
	};
};`

Naegolus, 2017-02-24:
Hello @meerd,

I usually don't review code but today I'm in the mood ...
For debugging you can use the following:
1. dmesg ... Displays kernel messages
2. vcdbg log msg ... Displays device tree messages
3. /proc/device-tree ... Here you see the actual device-tree the system uses. Navigate to your spi controller and see if it contains the devices and settings you expected
4. Documentation:
4.1 The kernel docs are very helpful here: https://www.kernel.org/doc/Documentation/
4.2 Also the raspberry pi docs: https://www.raspberrypi.org/documentation/configuration/device-tree.md
I guess you already checked this.

Maybe you had a conflict with a spidev on cs0. On the raspberry pi it seems that you have a "raw" spi device called spidev. If you use cs0 you first have to disable the spidev on this chip select.
You can do this inside the overlay see https://github.com/Naegolus/rr_base/blob/master/raspberry_pi_zero/dts/overlays/dspc-base-overlay.dts for example or as in your case in the /boot/config.txt

So your first overlays seems to be fine but you could add this to your spi2 controller:
spidev@0{
	status = "disabled";
};
Maybe this works too ...
In my m25p80 overlay I didn't have to do this because I used cs2.

jayslovak, 2017-07-30:
Sorry for bringing this thread up from the dead, but it just got me excited to see that someone has already worked on attaching w25q80 class of devices to raspberry pi.

@meerd Were you able to finalize the DTS and is it ready for mainlining?

Thanks!