raspberrypi/linux #1784: kernel 4.9: please enable TCP: BBR  congestion control algorithm

holywen, 2017-01-09:
tcp: BBR congestion control algorithm
Please enable it as it can increase the network speed much.

About BBR : 
http://queue.acm.org/detail.cfm?id=3022184
https://lwn.net/Articles/701177/

from the kernel patch description:

This patch series implements a new TCP congestion control algorithm:
BBR (Bottleneck Bandwidth and RTT). A paper with a detailed
description of BBR will be published in ACM Queue, September-October
2016, as "BBR: Congestion-Based Congestion Control". BBR is widely
deployed in production at Google.

The patch series starts with a set of supporting infrastructure
changes, including a few that extend the congestion control
framework. The last patch adds BBR as a TCP congestion control
module. Please see individual patches for the details.

TW-YingjieOuyang, 2017-01-12:
Totally agree, it will be super useful!

popcornmix, 2017-01-16:
Options added to 4.9 tree and in latest ``BRANCH=next rpi-update`` kernel.
Please report back if you are able to get measurable benefits from using this option.

amosnet, 2017-01-17:
Hi,
Just would like to say thank you to Raspberry Pi contributors.

Long story short,

I'm running a socks5 proxy on raspberry pi 3 between two network with very high latency and packet loss rate.
When with old cubic, youtube lags even on 144P. 
1080P is now not a problem with BBR.

I think there are definitely obvious measurable benefits for me.

zgzh, 2017-01-21:
Hi，
Very grateful to upgrade BBR after the speed of the network has undergone a qualitative change in the equipment, the same line environment, the original YouTube only 360P, can now 1440P, the effect is obvious, thanks again.

BenChud, 2017-01-24:
According to the documentation, BBR should not be used without the fair queue scheduler as it can result in a higher packet loss rate.

I don't think CONFIG_NET_SCH_FQ is currently set as I cannot enable in the stock kernel.

tc qdisc show dev eth0
qdisc pfifo_fast 0: root refcnt 2 bands 3 priomap  1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

tc qdisc add dev eth0 root fq maxrate 0.1gbit
RTNETLINK answers: No such file or directory

As per commit notes:

NOTE: BBR *must* be used with the fq qdisc ("man tc-fq") with pacing
enabled, since pacing is integral to the BBR design and
implementation. BBR without pacing would not function properly, and
may incur unnecessary high packet loss rates.

Given the above it would make sense to also set CONFIG_NET_SCH_FQ=m?

popcornmix, 2017-01-25:
I've added  CONFIG_NET_SCH_FQ module to 4.9 and 4.10 trees.

popcornmix, 2017-01-25:
Latest BRANCH=next rpi-update firmware has fair queue scheduler module included.

BenChud, 2017-01-25:
Thanks - can confirm I've been able to enable now

popcornmix, 2017-01-26:
@holywen Everything good? Okay to close?

holywen, 2017-01-27:
Yes, Thanks and  please  close it.

closed 2017-01-27 by popcornmix

free-the-internet, 2022-12-24:
Kernel 5.15.61-v8+
BBR is not available.

popcornmix, 2022-12-25:
`CONFIG_TCP_CONG_BBR` is enabled:
https://github.com/raspberrypi/linux/blob/3f4092766eaf692ed79b69f59c98dbe38e557fe7/arch/arm/configs/bcm2711_defconfig#LL105C17-L105C20

and this is a resolved issue.
If you have a problem, please create a new issue, and explain exactly what you have tried and how it failed.