raspberrypi/linux #1785: undable to add device tree overlays larger than 4KiB

cjk32, 2017-01-09:
I am unable to install a device tree overlay larger than 4KiB, as exemplified:

pi@raspberrypi:~/linux $ uname -a
Linux raspberrypi 4.4.39-v7+ #17 SMP Mon Jan 9 19:18:52 UTC 2017 armv7l GNU/Linux
pi@raspberrypi:~/linux $ ls -al /boot/overlays/w1-gpio
w1-gpio.dtbo         w1-gpio-pullup.dtbo  
pi@raspberrypi:~/linux $ ls -al /boot/overlays/w1-gpio
w1-gpio.dtbo         w1-gpio-pullup.dtbo  
pi@raspberrypi:~/linux $ ls -al /boot/overlays/w1-gpio.dtbo 
-rwxr-xr-x 1 root root 5769 Jan  9 19:49 /boot/overlays/w1-gpio.dtbo
pi@raspberrypi:~/linux $ sudo dtoverlay -v /boot/overlays/w1-gpio.dtbo 
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: Wrote 5769 bytes to '/tmp/.dtoverlays/0_w1-gpio.dtbo'
DTOVERLAY[debug]: Wrote 5769 bytes to '/sys/kernel/config/device-tree/overlays/0_w1-gpio/dtbo'
* Failed to apply overlay '0_w1-gpio' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post

[ 1802.983928] unflatten: error -11 processing FDT
[ 1802.983936] unflatten: error -11 processing FDT
[ 1802.984645] unflatten: error -11 processing FDT
[ 1802.984653] unflatten: error -11 processing FDT

Adding some tracing to the kernel, we find that by the time the blob reaches cfs_overlay_item_dtbo_write, it's become,

 o 4096b of valid data
 o 4096b of zeros
 o 1673b of valid data

i.e. we've gained an extra 4KiB in the middle of the blob.

The reason seems simple:

static ssize_t
configfs_write_bin_file(struct file *file, const char __user *buf,
			size_t count, loff_t *ppos)

does,

        len = simple_write_to_buffer(buffer->bin_buffer,
                        buffer->bin_buffer_size, ppos, buf, count);
        if (len > 0)
                *ppos += len;

which is wrong: simple_write_to_buffer() advances ppos itself.  Thus, the above code is double counting bytes, and every hunk that reaches configfs_write_bin_file() will end up followed by an equally sized hunk of zeroes.

The obvious fix of,

diff --git a/fs/configfs/file.c b/fs/configfs/file.c
index 3687187..6e322f2 100644
--- a/fs/configfs/file.c
+++ b/fs/configfs/file.c
@@ -357,8 +357,6 @@ configfs_write_bin_file(struct file *file, const char __user *buf,
 
        len = simple_write_to_buffer(buffer->bin_buffer,
                        buffer->bin_buffer_size, ppos, buf, count);
-       if (len > 0)
-               *ppos += len;
 out:
        mutex_unlock(&buffer->mutex);
        return len;

solves the problem for me.

Electron752, 2017-01-15:
That's cool.  Perhaps you should/could submit this as a fix to the upstream kernel.  The process is a bit of a pain, but it might be worth it.

Check out the document at Documentation/SubmittingPatches for an outline of the process.

Anybody can submit changes to the upstream kernel, but you have to send it to a large e-mail list for peer review and it ultimately needs to be applied by the maintainer.

popcornmix, 2017-01-15:
Yes, @Electron752 is correct, the bug is in the upstream code, so it's best to get it fixed there.
Assuming they are happy with the fix we can cherry-pick here.

pelwell, 2017-01-15:
The bug was already fixed as of kernel 4.7 - see https://github.com/raspberrypi/linux/commit/f8608985f851c917b3884b692d8e326b0210d34e. I've cherry-picked it back to rpi-4.4.y.

popcornmix, 2017-01-15:
Try with latest rpi-update kernel.

popcornmix, 2017-01-30:
@cjk32 is this still issue? If resolved then please close.

Ruffio, 2017-02-05:
@cjk32 can this be closed?

popcornmix, 2017-02-06:
The fix is present so I'll close.
@cjk32 feel free to reopen if the issue is not resolved.

closed 2017-02-06 by popcornmix