raspberrypi/linux #1789: Problem updating clone of rpi linux repo

MrEngman, 2017-01-12:


pelwell, 2017-01-12:
I can. Unfortunately we broke one of our own rules and force-pushed the rpi-4.4.y branch in order to remove some rogue commits. The future branches - 4.8, 4.9, 4.10 - are regularly rebased and force-pushed to keep the Linux history clean, but once they become the branch from which mainline releases are made we stop that and use merges to pull in subsequent Linux changes.

If you find yourself in that situation, the first thing to do is to abort the current rebase - `git rebase --abort` - to allow you to assess the situation before making any changes. If you have local commits then you'll want to rebase them on top of the new upstream head, otherwise you can just hard-reset your local tracking branch to the upstream head.

For these examples, `rpi-4.4.y` is the branch tracking `origin/rpi-4.4.y`, and `dev` contains some local commits on top of `rpi-4.4.y`.

To rebase the `dev` branch onto the updated upstream branch:
```
git rebase --onto origin/rpi-4.4.y rpi-4.4.y dev
```
This translates to "take the commits between rpi-4.4.y and dev and rebase them on top of origin/rpi-4.4.y, updating dev to the new head".

To update the tracking branch:
```
git checkout rpi-4.4.y
git reset --hard origin/rpi-4.4.y
```
I hope this helps, and apologies for the inconvenience.

pelwell, 2017-01-12:
It occurs to me that:
```
git checkout rpi-4.4.y
git reset --hard origin/rpi-4.4.y
```
can be replaced with:
```
git rebase --onto origin/rpi-4.4.y rpi-4.4.y rpi-4.4.y
```
which can be further simplified to:
```
git rebase --onto origin/rpi-4.4.y rpi-4.4.y
```
As well as being slightly shorter, this may be quicker since it skips the checkout of the old rpi-4.4.y branch.

MrEngman, 2017-01-13:
Looks like I managed to sort things out using:

```
git checkout rpi-4.4.y
git reset --hard origin/rpi-4.4.y
```

Thanks for your help.

Electron752, 2017-01-15:
@MrEngman 

If you have a solution, perhaps it would make sense to close the issue?

popcornmix, 2017-01-15:
I'll close.
BTW @MrEngman if your script runs a ``git pull`` then you might consider using ``git pull --rebase``.
I believe that will always in cases where git pull would have worked, and will also work in cases like this.

closed 2017-01-15 by popcornmix