raspberrypi/linux #1793: Please add CONFIG_DM_CACHE=m

ChewbaccaKL, 2017-01-14:
Please consider enabling dm-cache.ko (CONFIG_DM_CACHE=m). Using raspberry pi with a fast internal ssd card can speed up using the external disk connected with a slow USB2.

Thanks

Electron752, 2017-01-15:
Just to add my two cents here, I've found that a good USB attached flash drive will often out perform a typical camera card flash card on the RPI.

popcornmix, 2017-01-15:
@ChewbaccaKL Do you have any numbers with and without this enabled?
E.g. boot time, or time to launch browser and open web page?

I think I'm okay with this. @pelwell ?

pelwell, 2017-01-15:
The option is marked as experimental, and it seems to rely on the selection of "policies" to tune the behaviour. Is this something we want to enable for all users without a wider trial?

popcornmix, 2017-01-15:
@pelwell 
The module has no dependencies on options we don't already use.
It is enabled (=m) on Ubuntu.
I imagine building the module will have no effect on a user who doesn't manually configure it.

Electron752, 2017-01-15:
Sounds like the only real downside to accepting this change is the increased build time and the potential for needing to deal with more build breaks.  And I don't think one more module is going to make much difference.

pelwell, 2017-01-15:
I withdraw my concern.

popcornmix, 2017-01-15:
Should be in latest rpi-update kernel.

ChewbaccaKL, 2017-01-16:
@popcornmix
I tried on encrypted 2TB hdd (encryption AES) 5MB/s with local cache (encrypted using xor with random key 13MB/s) and “felt” the improvement. The encrypted devices without caching were about 3MB/s.

ChewbaccaKL, 2017-01-16:
Thanks a lot. It really saves me a lot of work with recompiling dm-cache every kernel release.

closed 2017-01-30 by popcornmix