raspberrypi/linux #1797: Suggest to enable accelerated AES modules on arm64/v4.11 defconfig

ardbiesheuvel, 2017-01-15:
Just a head's up to anyone interested that v4.11 will contain some AES improvements for arm64, which will benefit anyone doing VPN or disk encryption on their Pi3s

CONFIG_AES_ARM64
scalar AES that is 10-15% faster, but more importantly, uses only 1/4 of the lookup tables compared to generic AES, so it is much easier on the D-cache (i.e., it does not interfere as much with other tasks executing concurrently)

CONFIG_AES_ARM64_BS
NEON based AES in ECB/CBC/CTR/XTS modes, about 2x faster than generic AES, and more secure than scalar AES since it does not use any lookup tables at all (i.e., cache timing attacks are impossible)

Example output from tcrypt mode=500 for aes-generic, aes-arm64 and aes-neon-bs, respectively

testing speed of async xts(aes) (xts(ecb(aes-generic))) encryption
test 0 (256 bit key, 16 byte blocks): 453148 operations in 1 seconds (7250368 bytes)
test 1 (256 bit key, 64 byte blocks): 269528 operations in 1 seconds (17249792 bytes)
test 2 (256 bit key, 256 byte blocks): 97046 operations in 1 seconds (24843776 bytes)
test 3 (256 bit key, 1024 byte blocks): 29279 operations in 1 seconds (29981696 bytes)
test 4 (256 bit key, 8192 byte blocks): 3840 operations in 1 seconds (31457280 bytes)
test 5 (384 bit key, 16 byte blocks): 399585 operations in 1 seconds (6393360 bytes)
test 6 (384 bit key, 64 byte blocks): 234249 operations in 1 seconds (14991936 bytes)
test 7 (384 bit key, 256 byte blocks): 83065 operations in 1 seconds (21264640 bytes)
test 8 (384 bit key, 1024 byte blocks): 24735 operations in 1 seconds (25328640 bytes)
test 9 (384 bit key, 8192 byte blocks): 3253 operations in 1 seconds (26648576 bytes)
test 10 (512 bit key, 16 byte blocks): 371454 operations in 1 seconds (5943264 bytes)
test 11 (512 bit key, 64 byte blocks): 210489 operations in 1 seconds (13471296 bytes)
test 12 (512 bit key, 256 byte blocks): 73345 operations in 1 seconds (18776320 bytes)
test 13 (512 bit key, 1024 byte blocks): 21419 operations in 1 seconds (21933056 bytes)
test 14 (512 bit key, 8192 byte blocks): 2798 operations in 1 seconds (22921216 bytes)

testing speed of async xts(aes) (xts(ecb(aes-arm64))) encryption
test 0 (256 bit key, 16 byte blocks): 497409 operations in 1 seconds (7958544 bytes)
test 1 (256 bit key, 64 byte blocks): 293040 operations in 1 seconds (18754560 bytes)
test 2 (256 bit key, 256 byte blocks): 104635 operations in 1 seconds (26786560 bytes)
test 3 (256 bit key, 1024 byte blocks): 31314 operations in 1 seconds (32065536 bytes)
test 4 (256 bit key, 8192 byte blocks): 4106 operations in 1 seconds (33636352 bytes)
test 5 (384 bit key, 16 byte blocks): 463638 operations in 1 seconds (7418208 bytes)
test 6 (384 bit key, 64 byte blocks): 263244 operations in 1 seconds (16847616 bytes)
test 7 (384 bit key, 256 byte blocks): 92024 operations in 1 seconds (23558144 bytes)
test 8 (384 bit key, 1024 byte blocks): 27111 operations in 1 seconds (27761664 bytes)
test 9 (384 bit key, 8192 byte blocks): 3530 operations in 1 seconds (28917760 bytes)
test 10 (512 bit key, 16 byte blocks): 429851 operations in 1 seconds (6877616 bytes)
test 11 (512 bit key, 64 byte blocks): 239053 operations in 1 seconds (15299392 bytes)
test 12 (512 bit key, 256 byte blocks): 81809 operations in 1 seconds (20943104 bytes)
test 13 (512 bit key, 1024 byte blocks): 23710 operations in 1 seconds (24279040 bytes)
test 14 (512 bit key, 8192 byte blocks): 3092 operations in 1 seconds (25329664 bytes)

testing speed of async xts(aes) (xts-aes-neonbs) encryption
test 0 (256 bit key, 16 byte blocks): 342882 operations in 1 seconds (5486112 bytes)
test 1 (256 bit key, 64 byte blocks): 340216 operations in 1 seconds (21773824 bytes)
test 2 (256 bit key, 256 byte blocks): 200157 operations in 1 seconds (51240192 bytes)
test 3 (256 bit key, 1024 byte blocks): 58559 operations in 1 seconds (59964416 bytes)
test 4 (256 bit key, 8192 byte blocks): 7544 operations in 1 seconds (61800448 bytes)
test 5 (384 bit key, 16 byte blocks): 298730 operations in 1 seconds (4779680 bytes)
test 6 (384 bit key, 64 byte blocks): 295504 operations in 1 seconds (18912256 bytes)
test 7 (384 bit key, 256 byte blocks): 172183 operations in 1 seconds (44078848 bytes)
test 8 (384 bit key, 1024 byte blocks): 49970 operations in 1 seconds (51169280 bytes)
test 9 (384 bit key, 8192 byte blocks): 6356 operations in 1 seconds (52068352 bytes)
test 10 (512 bit key, 16 byte blocks): 263140 operations in 1 seconds (4210240 bytes)
test 11 (512 bit key, 64 byte blocks): 262062 operations in 1 seconds (16771968 bytes)
test 12 (512 bit key, 256 byte blocks): 150960 operations in 1 seconds (38645760 bytes)
test 13 (512 bit key, 1024 byte blocks): 43404 operations in 1 seconds (44445696 bytes)
test 14 (512 bit key, 8192 byte blocks): 5597 operations in 1 seconds (45850624 bytes)

Electron752, 2017-01-15:
Cool.  Sounds like ARM64 is slowly gaining some value add.

Ruffio, 2017-02-05:
@ardbiesheuvel must something be enabled pr what is the issue? If there is no issue, then this should be closed IMHO

ardbiesheuvel, 2017-02-05:
On 5 February 2017 at 10:43, Rasmus Christiansen <notifications@github.com>
wrote:

> @ardbiesheuvel <https://github.com/ardbiesheuvel> must something be
> enabled pr what is the issue? If there is no issue, then this should be
> closed IMHO
>
>
​Simply a matter of adding the Kconfig definitions to the RPi3 defconfig.
​

JamesH65, 2017-05-18:
@popcornmix Worth adding to the 4.11 branch?

popcornmix, 2017-05-18:
We don't directly support or test the arm64 kernel, so I'm not keen to add options untested.
If someone wants to PR a tested change to ``arch/arm64/configs/bcmrpi3_defconfig`` and no one objects then we'll accept it.

closed 2017-05-18 by popcornmix

ShapeShifter499, 2018-04-27:
Why was this closed?    I'd like to see any improvement in encryption