raspberrypi/linux #1798: Fixing the default build configuration mess

Electron752, 2017-01-15:
As ARM64 is catching up, the difference in the default build configuration from ARM32 is getting smaller.  Since maintaining these configurations appear to be somewhat time consuming, I've been toying with the idea of dynamically generating the default build configuration.  After looking through the build system for awhile now, I can see that some of the platforms such as MIPS already do this so the kernel has a few examples on how this can be done.

The downstream tree uses 3 build configurations(bcmrpi_defconfig, bcm2709_defconfig, and bcmrpi3_defconfig).  I can count on both hands the number of differences between the configs. 

What I'm thinking is this. Instead of storing the full default config for arm64, perhaps it would be better to store in the tree a .patch files with the delta from bcm2709_defconfig?   When the builder runs "make bcmrpi3_defconfig", the Makefile can apply the diff/delta to bcm2709_defconfig to automatically generate bcmrpi3_defconfig then run that through the build system to generate .config.  This could be logically extended to make bcm2709_defconfig also be a .patch from bcmrpi_defconfig. An alternative to diff and patch would be to factor out a common file and cat it with the architecture specific version.  This would be simpler, but less flexible.

The question I have is:
1. Is this idea something that makes sense, or I'm I completely out in left field here?
2. If I implement this, is this something that is likely to be accepted/merged into the tree?

pelwell, 2017-01-15:
The users of the default configurations vastly outnumber the maintainers of them, so some pain for the maintainers is preferable to even one extra step for all users. I have a script that takes named configurations and adds or removes options, so keeping the two (or three) in step is easy enough. We could share a script to make maintenance easier, but the end result should be standard configurations.

Electron752, 2017-01-15:
Sharing the script would be cool.  

What I don't understand is where the extra build step is for the user/builder.   The user would still simply type "make bcmrpi3_defconfig" or "make bcm2709_defconfig" and all the magic could happen transparently.  In fact, they wouldn't even need to know that these are not hardcoded files.

Electron752, 2017-01-15:
Also look at it this way, if by some long shot ARM64 gains traction just consider the reduction in the need for public pull requests that need to be reviewed/merged/propagated.  A small delta file is much easier to review and it probably wouldn't even change much.

pelwell, 2017-01-15:
Yes, the actions could be triggered automatically, but why add magic to Makefiles to burn (admittedly trivial amounts of) CPU on all build machines when it doesn't really benefit anyone to do so?

I like the fact that the config files are simple text files that can be searched, diffed and have their changes tracked in the git history.

Electron752, 2017-01-15:
OK, I understand better now.  Just two more questions, and I will leave this alone.

1. Aren't CPU cycles cheaper then human cycles and less error prone?
2. How is this different from the defconfig being run through the kconfig system to produce .config?  Why isn't .config stored in the git history?

pelwell, 2017-01-15:
1. Yes, which is why I'm in favour of using tools to automate maintenance tasks.
2. It is exactly like the defconfig -> .config transformation, but with an extra level of meta. .config isn't stored in the git history because it is the output from the Kconfig process, which is complex but useful and documented. It reduces the churn and errors in saved configurations by separating user choices and hardware requirements from internal rules and dependencies. Your aim is similarly to reduce churn and avoid errors - I understand that - but if we expect other people to submit pull requests then any change of input representation has to be carefully documented. I can imagine the temptation to add a macro facility...

@popcornmix You've stayed quiet so far - what's your take on this?

Electron752, 2017-01-15:
@pelwell the getting other people to help is a good point.   The build process is already complex and if it's made too complex, then it makes it too difficult for people to get started.

popcornmix, 2017-01-15:
I agree with @pelwell 
The format of files in arch/arm/configs is straightforward and standard (all non-default .config settings) and it is fairly easy to link to them and compare them.
I don't think we want add another layer that devs/users need to understand.

I've not seen what is present upstream. If there is a straightforward, documented scheme for having config files depend on other config files, then I could be persuaded, but I don't think we want to go down some non-standard route where we have to continually explain to devs/users how it works.

Electron752, 2017-01-15:
Good points.  Perhaps it's better for things to be the way they are for now.

I'll close out the issue.

closed 2017-01-15 by Electron752