raspberrypi/linux #1799: bcm2835-i2s regression from asymmetric sample formats change

HiassofT, 2017-01-16:
PR #1783 introduced a regression: when both a playback and record stream are opened at the same time several registers for the second stream aren't set up and the second stream fails.

eg use aplay | arecord to reproduce.

At the very beginning bcm2835_i2s_hw_params checks if any stream has been started and if yes hw_params bails out immediately:
```
	/*
	 * If a stream is already enabled,
	 * the registers are already set properly.
	 */
	regmap_read(dev->i2s_regmap, BCM2835_I2S_CS_A_REG, &csreg);

	if (csreg & (BCM2835_I2S_TXON | BCM2835_I2S_RXON))
		return 0;
```

This leads to eg BCM2835_I2S_TXC_A_REG/BCM2835_I2S_RXC_A_REG, BCM2835_I2S_MODE_A_REG not being initialized.

A proper fix would need quite some rework (only set up clocks once in hw_params, do separate clear_fifos for RX/TX etc) and also a lot of testing because I2S reconfiguration while another stream is  running has never been tested before.

@gtrainavicius could you have a look at this?

@popcornmix @pelwell  I suggest it's better to revert #1783 for now

pelwell, 2017-01-16:
Reverted.

closed 2017-01-16 by HiassofT

gtrainavicius, 2017-01-16:
@pelwell @HiassofT Thank you, I will take a look when I have time and submit a new pull request.