raspberrypi/linux #1800: printf  problems with negative numbers after latest update

kenzon, 2017-01-16:
A few days ago I updated my respeberry pi 2 with normal updates.

Now     printf '%.1f\n' -3.2
gives errors (= invalid number) regarding negative numbers, BEFORE update there was no problems.

PI INFO:
 uname -a
Linux 01 4.4.32-v7+ #924 SMP Tue Nov 15 18:11:28 GMT 2016 armv7l GNU/Linux

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

XECDesign, 2017-01-16:
Not seeing this myself and it doesn't sound like a linux kernel issue.
```
pi@serge-pi ~ $ sudo apt-get update && sudo apt-get dist-upgrade
Get:1 http://archive.raspberrypi.org jessie InRelease [22.9 kB]
Get:2 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]              
Hit http://archive.raspberrypi.org jessie/main Sources                                              
Get:3 http://mirrordirector.raspbian.org jessie/main Sources [7,746 kB]                             
Hit http://archive.raspberrypi.org jessie/ui Sources
Hit http://archive.raspberrypi.org jessie/staging Sources                                                               
Hit http://archive.raspberrypi.org jessie/untested Sources                                                              
Get:4 http://archive.raspberrypi.org jessie/main armhf Packages [136 kB]                                                
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                      
Hit http://archive.raspberrypi.org jessie/staging armhf Packages
Hit http://archive.raspberrypi.org jessie/untested armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB                                                                               
Ign http://archive.raspberrypi.org jessie/main Translation-en                                             
Ign http://archive.raspberrypi.org jessie/staging Translation-en_GB
Ign http://archive.raspberrypi.org jessie/staging Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://archive.raspberrypi.org jessie/untested Translation-en_GB
Ign http://archive.raspberrypi.org jessie/untested Translation-en
Get:5 http://mirrordirector.raspbian.org jessie/contrib Sources [64.5 kB]
Get:6 http://mirrordirector.raspbian.org jessie/non-free Sources [113 kB]
Get:7 http://mirrordirector.raspbian.org jessie/rpi Sources [1,132 B]
Get:8 http://mirrordirector.raspbian.org jessie/main armhf Packages [8,981 kB]
Get:9 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:10 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]
Get:11 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                                                                                                                                                                         
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB                                                                                                                                                                         
Ign http://mirrordirector.raspbian.org jessie/main Translation-en                                                                                                                                                                            
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB                                                                                                                                                                     
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en                                                                                                                                                                        
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB                                                                                                                                                                          
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en                                                                                                                                                                             
Fetched 17.2 MB in 36s (474 kB/s)                                                                                                                                                                                                            
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@serge-pi ~ $ cat test.c 
#include <stdio.h>

int main()
{
	printf("%.1f\n",-3.2);
}
pi@serge-pi ~ $ gcc test.c -o test1
pi@serge-pi ~ $ ./test1
-3.2
pi@serge-pi ~ $
```

pelwell, 2017-01-16:
What language are you writing in? Regardless of the answer, posting a minimal example program will save time.

[ I'm currently wondering if your '-' is really a minus sign ]

popcornmix, 2017-01-16:
I believe it is just bash. But it works for me.
```
pi@domnfs:~ $  printf '%.1f\n' -3.2
-3.2
```
An incorrect minus would give that error:
```
pi@domnfs:~ $ printf '%.1f\n' _3.2
-bash: printf: _3.2: invalid number
```

XECDesign, 2017-01-16:
Yup, some wordpress templates, for example, like to replace symbols with ones which look identical.
```
pi@serge-pi ~ $ printf "%.1f\n" ‐3.2
-bash: printf: ‐3.2: invalid number
0.0
pi@serge-pi ~ $ printf "%.1f\n" -3.2
-3.2
```

kenzon, 2017-01-16:
done sudo apt-get update && sudo apt-get dist-upgrade

Did not help on the reported respeberry pi.
'-' minus sign hase been entered ny numerical minus - just for test.

But tried on an other pi and strange enoug this Works:
~ $ printf '%.1f\n' -3.2
-3.2

THIS PI INFO:
 uname -a
Linux de-Thurah-web-02 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       Jessie

To XECDesign 
Yes - thats exactly what I'm expericing I'm not using wordpress.

I will investigate symbol problems on the box - THANS for all the anwsers :-)

popcornmix, 2017-01-16:
Can you run
```
pi@domnfs:~ $ echo -n "-" | od -An -tuC
  45
```
(typing in the minus, rather than copying pasting it)

pelwell, 2017-01-16:
You beat me to it - mine was:
```
perl -e 'print ord("-"),"\n"'
```

kenzon, 2017-01-16:
Thanks for all the help - I found the "bag guy" = language problems   ???

Both 
$ echo -n "-" | od -An -tuC 
AND
$ perl -e 'print ord("-"),"\n"'
RETURNED 45

On the box with the problem I run sudo raspi-config and removed "da_DK.UTF-8 UTF-8 " now only  "LANG=en_GB.UTF-8" remains

It required a reboot before all was in place, so now

:/home/pi# printf '%.1f\n' -3.2
-3.2

THANS for all the anwsers,  Kenzon      :-)

kenzon, 2017-01-16:
You can close the issue, Kenzon

closed 2017-01-16 by kenzon