raspberrypi/linux #1801: ARM64: What's missing at this point

Electron752, 2017-01-16:
This is mostly a note to self and anyone else interesting in working on ARM64/AARCH64.

Here is a list of what I know is still missing for arm64 support on the RPI3 as of the current downstream 4.10 version.

1. VCHIQ is missing a compatibility interface for the ioctls for calls from 32 bit applications.   A patch was submitted upstream for this but for various reasons it wasn't completed.   This should probably be fixed in upstream.

2. VMCS Shared Memory Driver - This is a small driver and the remaining issues appears to be mostly related to CPU cache flushing.

3. VideoCore CMA - Same issues as #2 and is relatively easy to fix.   The only issue if if it's worth fixing this as vc_cma is not supported as far as I can tell.

4. VIDEO_BCM2835_MMAL: This is the camera interface driver.  I'm not sure if this is only used by the V1 camera, or used by both V1 and V2.  If it is only used by V1, then it can probably be ignored.  The VIDEO_BCM2835 driver compiles without any issues.

In theory, fixing #1 and #2 should allow out of box Raspbian to work with accelerated 3D and the hardware video codec license and should give a very functional base for running Raspbian. Essentially to just be able to "Drop In" a 64 bit kernel on a standard install.

I'm hoping someone with more knowledge of the kernel can chime in as to the importance of #2-#4.  I'm not sure how much actually uses the shared memory driver.

6by9, 2017-01-16:
2 - I'm hoping to give VC Shared Memory some love soon to teach it about dmabuf. It's a very useful tool to improve transfer performance from VC to ARM and vice versa and both processors then use the same memory block instead of copying. Yes there is some cache flushing/invalidation required on locking and unlocking to ensure correct behaviour.

3 - vc_cma. Not used AFAIK. It was a way of allowing VC to request chunks of CMA from the kernel on demand. It never worked particularly well.

4 - V4L2. Used for both camera modules as it is just another MMAL client implementation but within the kernel and then presenting it in through the V4L2 API. pelwell already picked up on a nasty in MMAL that it stores a client context pointer in the message sent to the GPU, and that is only 32bit. https://github.com/raspberrypi/firmware/issues/550#issuecomment-190842043
Needs a bit of working around, and also hits the userland implementation too :-(

Electron752, 2017-01-17:
Cool.

I'll probably try submitting a new patch for the VCHIQ ioctl issue to upstream in the near future.   I'm thinking of using the method of writing a layer on top of the existing ioctls rather then ripping out old code.   Even though much of VCHIQ isn't that easy to deal with, I think the on top of layer approach is more likely to be accepted upstream and raise less security concerns.

I might buy the camera soon and make an attempt at fixing the V4L2 driver as well, but to me that's less important then finishing VCHIQ.

6by9, 2017-01-17:
The tricky bit with the "on-top" method that I've seen before is that the existing code is expecting user pointers, so you end up having to copy_from_user the 32 bit __user pointer into a 64 bit __user pointer that the kernel has to allocate. It can get a bit messy, but perhaps I've missed the simple aproach.
I haven't looked at VCHIQ to see whether that will affect it significantly, but I suspect so as it is passing a fair number of pointers around.

V4L2 also wants some love to use the videobuf2_contig allocator instead of vmalloc, but that is contingent on having VCSM being able to import external allocations (eg dmabuf, see item 2). That will also need a little thought for backward compatibility as the current downstream kernel here has very little CMA memory allocated. 
All on the to-do list, but assistance welcome :-)

Electron752, 2017-01-17:
The later kernels have some internal functions for allocating the converted structures on the 32 bit userland stack.  The way when the "real"/64 bit version gets called, it still gets a __user pointer just like it would expect.   This appears to be the approach taken by the video DRM subsystem.  It was originally suggested that that the "on top" solution was a better solution, but at that time I thought the VCHIQ code was so bad that it would be worth trying to cleanup.

You bet, I'm certainly willing to help improve VCSM and V4L2 as well.

6by9, 2017-01-17:
> The later kernels have some internal functions for allocating the converted structures on the 32 bit userland stack.

I'm aware of ```compat_alloc_user_space```, but it seems to be such a horrid set of hoops to jump through to convert from 32bit userspace to a second userspace allocation, to then convert to kernel.
A refactor to have implementations for each ioctl function that take kernel 64bit pointers, and then the 64 bit ioctl call deals with copying to/from a 64 bit userspace, and the compat 32 bit ioctl deals with doing the same from a 32 bit userspace and converting to 64 bit kernelspace. It's all work though.

I'll try to draw up some issues to track the proposed work on VCSM and V4L2. Half the VCSM stuff is firmware side, but the rest is kernel.
Eric has already covered some of the V4L2 stuff in https://github.com/anholt/linux/issues/12 and has the contig allocator working, but it is still making VCHI copy all the data. That needs VCSM / VideoCore to be able to import dmabufs and correctly release the reference to them when finally done.

Electron752, 2017-01-17:
I'm interested to see the information on VCSM/V4L2 when available.

I'm curious how you think the whole ioctl mess in VCHIQ should be cleaned up?  I still have the patch that I submitted that was a rather extensive cleanup that broke up the ioctl mess into helper functions and did not rely on compat_alloc_user_space.   The thinking was that the patch was too much/too big for people to review and I think breaking it up into a chain of 20 commits that change it a bit at a time yet keeps things working on each commit was way too much work.   I'm certainly open to tweaking the original patch and resubmitting it.

6by9, 2017-01-17:
Assuming you posted it to the mailing list I'll try to find your patch and have a look. A quick pointer with the date, or a direct link to the archive, would speed the searching up...

It sounds like you did exactly what I was expecting to do. A helper function for each ioctl's implementation, and ```vchiq_ioctl``` becoming any relevant copy_from_user / copy_to_user calls. The compatibility ioctl handler is then the same thing but with the 32 bit versions of any ioctl structures.
Breaking it up into a chain of commits will be a little tedious, but predominantly obvious as just moving blocks of code around so shouldn't cause too much grief in reviewing.

Electron752, 2017-01-17:
Here is the patch:
http://lists.infradead.org/pipermail/linux-rpi-kernel/2016-November/004736.html

It's a very large patch.  Perhaps it went a bit too far.

6by9, 2017-01-17:
Thanks.
That's pretty unreadable as a patch!

The examples I'm thinking of are all in vendor specific patches, on an older kernel, and almost all down in V4L2 land, so I can't really reference them. However, just to be clearer, my thoughts were something like:
```
static vchiq_do_create_service(instance, VCHIQ_CREATE_SERVICE_T args)
{
  // All the stuff that is currently in the ioctl handler
  // except the copy_from_user / copy_to_user calls.
}

static long vchiq_ioctl(struct file *file, unsigned int cmd, unsigned long arg)
{
  switch(cmd) {
	case VCHIQ_IOC_CREATE_SERVICE: {
		VCHIQ_CREATE_SERVICE_T args;
		USER_SERVICE_T *user_service = NULL;
		void *userdata;
		int srvstate;

		if (copy_from_user
			 (&args, (const void __user *)arg,
			  sizeof(args)) != 0) {
			ret = -EFAULT;
			break;
		}
		handle = vchiq_do_create_service(instance, ...);

		if (copy_to_user((void __user *)
				&(((VCHIQ_CREATE_SERVICE_T __user *)
					arg)->handle),
				(const void *)&handle,
				sizeof(handle)) != 0) {
				ret = -EFAULT;
				vchiq_remove_service(service->handle);
		}
}

static long vchiq_ioctl32(struct file *file, unsigned int cmd, unsigned long arg)
{
  switch(cmd) {
	case VCHIQ_IOC_CREATE_SERVICE32: {
		VCHIQ_CREATE_SERVICE32_T args32;
		VCHIQ_CREATE_SERVICE_T args;
		USER_SERVICE_T *user_service = NULL;
		void *userdata;
		int srvstate;

		if (copy_from_user
			 (&args32, (const void __user *)arg32,
			  sizeof(args32)) != 0) {
			ret = -EFAULT;
			break;
		}

		// Copy all values from args32 to args. Special handling for any pointers in the structs.

		handle = vchiq_do_create_service(instance, &args);

		if (copy_to_user((void __user *)
				&(((VCHIQ_CREATE_SERVICE32_T __user *)
					arg32)->handle),
				(const void *)&handle,
				sizeof(handle)) != 0) {
				ret = -EFAULT;
				vchiq_remove_service(service->handle);
		}
	}
}
...
static const struct file_operations
vchiq_fops = {
	.owner = THIS_MODULE,
	.unlocked_ioctl = vchiq_ioctl,
#ifdef CONFIG_COMPAT
	.compat_ioctl = vchiq_ioctl32,
#endif
	.open = vchiq_open,
	.release = vchiq_release,
	.read = vchiq_read
};
```
Rinse and repeat for all the other IOCTLs. When it comes to it there may be some that can be pruned out as unnecessary on Pi and deprecate them.
Tedious to do, and lots of patches, but probably worthwhile overall. Hopefully it works as well. And I have no ideas on Phil's view of that proposal!

Electron752, 2017-01-17:
I agree that would be easier to review and get approved and is probably the path I'll take.  Although I think the end result won't be nearly as clean or maintainable in the long term as what I have in the patch.

That's probably the next RPI related things I'll be doing, but since I'm a volunteer here I can't give a time for when I will do it:)

6by9, 2017-01-17:
I'm a volunteer too and have observed just how painful getting stuff merged upstream can be. That's part of the reason why I normally work downstream here...

6by9, 2017-01-19:
Issues created as requested. #1806 and #1807.

6by9, 2017-01-19:
And I see you're having "fun" on the mailing list trying to get reviewers to look at your VCHI changes. Playing upstream almost seems too painful to be worth it. Either that or I need thicker skin and more patience.

pelwell, 2017-01-19:
Yes, it is painful to watch.

Electron752, 2017-01-20:
Yes, I might have burned a few bridges there.  Hopefully not too many.  I think I better let the dust settle a bit.

The whole concept of having a very low bar for getting new drivers into staging, yet bully or nit-pick on issues for changes that attempt to improve things doesn't sound to me like a good process.

@pelwell - I'm going to try to continue to get the ioctl changes in upstream but assuming I can't for whatever reason... Any possibility of getting the changes checked into downstream as a temp solution?

notro, 2017-01-20:
@Electron752
If you want to be a kernel developer, this can be seen as a top notch crash course by one of Linus' lieutenants (Greg) and one of the best kernel bug hunters (Dan, [smatch](https://lwn.net/Articles/691882/)). I don't know how many dozen patches they review every day, and yet they spend time reviewing newbie patches.

If you just want to add some code, then yes it's painful.

My advice is to just accept their judgement as long as the code works. Keep sending new versions. Even seasoned developers can end up with many revisions before their work is accepted.

Electron752, 2017-01-22:
@notro 
That's good advice.  Greg and Dan both have to be careful about allowing garbage to get into the linux kernel so they are certainly justified.  I just need to not take it too personal when they point out issues with submitted patches.

On a somewhat unrelated topic, I sent out new patches this morning for 32 compat for vchiq.  For fun, I tried applying the submitted patches to the downstream 4.10 kernel here on github and I used it to boot an out of box Raspbian with an arm64 kernel.   

One of things I noticed is working now is minecraft_pi which doesn't work without 32 bit VCHIQ support.  It seems to be rather functional, and if I understand correctly minecraft_pi uses some hardware acceleration.  I also was able to get glxgears to work at 40 fps which is much slower then what can be obtained with the vc4 driver but it's still interesting.   Note this is with firmware acceleration, not the vc4 kms driver. 

Raspbian's video plugin still can't play YouTube videos and are course Mathematica doesn't work either due to copy protection and os detection issues.

closed 2017-02-03 by Electron752