raspberrypi/linux #1802: Any circumstances in which gpio-poweroff overlay partially works?

MrSurly, 2017-01-17:
I've posted this over at the [Adafruit forums](https://forums.adafruit.com/viewtopic.php?f=47&t=109257) where the gpio-poweroff overlay only partially works.

Using the Adafruit 4.4.27-v7+ kernel that's available for the [PiTFT Plus](https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi).  This enables framebuffer output to the PiTFT via SPI.

While using the Adafruit kernel, the gpio-poweroff overlay partially works.  The line goes high while the device is on, but fails to go low on poweroff.

Using the stock Raspbian Jessie kernel (4.4.34-v7+), the gpio-poweroff overlay works correctly, with the same hardware.

Here's the relevant section from `/boot/config.txt`, which is the same for both.

`dtoverlay=gpio-poweroff,gpiopin=6,active_low="y"`

Both kernels have `CONFIG_POWER_RESET=y` and `CONFIG_POWER_RESET_GPIO=y`.

I have tried using other available GPIO pins, with the same results.

pelwell, 2017-01-18:
Apart from the (harmless but unnecessary) quotes in `active_low="y"` that all looks fine.
I'm not sure what you are expecting us to do - the gpio-poweroff code and overlay in the Adafruit kernel appear to be unmodified from the standard RPi versions, so something else must have broken it.

If I were debugging this I would first add some pr_err() calls to significant points in the gpio-poweroff driver and rebuild the kernel, then use a serial port to capture the kernel output (make sure there is no 'quiet' in cmdline.txt and include 'loglevel=8').

MrSurly, 2017-01-18:
> I'm not sure what you are expecting us to do - the gpio-poweroff code and overlay in the Adafruit kernel appear to be unmodified from the standard RPi versions, so something else must have broken it.

Was hoping something along the lines of "oh yeah, that version is broken," though searching issues didn't yield an answer.  Your confirmation of "should work" is very helpful on it's own.  Thank you.

MrSurly, 2017-01-18:
Turns out that the `rpi_power_switch` [module](https://github.com/notro/fbtft_tools/blob/master/rpi_power_switch/rpi_power_switch.c) (written by Adafruit) prevents this from working correctly.  I haven't dug into it too deeply, but I assume this module intercepts the shutdown. I'll raise an issue with Adafruit, even if it's just updating the docs to say "this breaks `gpio-poweroff`.

Thanks for your time.

closed 2017-01-18 by MrSurly

pelwell, 2017-01-18:
Yes - `pm_power_off` is a global pointer to a single power-off handling function, and both modules are competing for it.

MrSurly, 2017-01-18:
Understood, thank you.  I am (at best) a very amateur kernel hacker.