raspberrypi/linux #1811: System freezes (task blocked) and display echo 0 > /proc/sys/kernel/hung_task_timeout_secs

Mister-X-, 2017-01-22:
Using Kernel 4.4.39-v7_Re4son-Kali-Pi-TFT+ (which is based on this kernel with adafruit patches for TFT display and 802.11 injection patches),
I fairly often run into issue where the whole system becomes completely unusable, the same way it would be with a kernel panic (network connection die, cannot use keyboard/mouse, etc). I can often (but not always) trigger it by running ```svn up``` on a repo I'm working on (or run ```airmon-ng```) and I can see the following on the screen (blocked task names vary, though):

[image]

I tried different SD cards but that doesn't seem to be the issue as it happens even with brand new microSD cards.
It can happen 2 minutes after I started working on it or 10 minutes later, it's pretty much unpredicatable when it will happen.

popcornmix, 2017-01-30:
We can't do much to help with custom kernels. Does the problem occur with a default raspbian install?
Are you overclocking?
How is the power supply? What does ``vcgencmd get_throttled`` report?

Mister-X-, 2017-02-04:
I haven't tried Raspbian and vcgencmd is not available on the system.

I don't think the patches affect any of the scheduling at all (driver for TFT screen and injection patch that affect the wireless drivers).

Anyway, I'm not overclocking, I even have heatsinks on the chips and the powersupply is Canakit 2.5A 5V.

asbubam, 2017-02-10:
@Mister-X-  Would you try vm.dirty_background_ratio, vm.dirty_ratio for reduce IO stress like this?

https://www.reddit.com/r/linux/comments/3h7w8f/better_linux_disk_caching_performance_with/

Mister-X-, 2017-02-11:
The values from one of the last posts (with local.conf) seems to do the trick. I've been having a svn up loop for a little while and it's still going strong.

rodizio1, 2017-02-22:
> (or run airmon-ng) [...]

I have similar issues with wifi cards. One thing I found is that Atheros cards don't like promiscous mode on the Pi, it'll crash in about 2 from 10 tries. Also, bringing up or down wifi cards or setting them to monitor mode only works flawless if there is nothing else going on on the USB bus. Ethernet cable plugged in and pump requesting an IP address at the same time while bringing up usb wifi interfaces cause USB crashes often.

The vm_dirty values you changed are not really a solution, just a workaround. OSes shouldn't freeze completely just because there is some disk access.

closed 2017-05-18 by JamesH65