raspberrypi/linux #1819: feature request - add slcan kernel module

GBert, 2017-01-27:
Hi,

slcan is a CAN driver for SLCAN API based boards. These adapters are easy to build and cheap - [like this](https://github.com/GBert/misc/tree/master/stm32-slcan). 
Would you be so kind to add slcan to the standard Raspbian distribution ? The slcan kernel module only consumes a few kbytes (roughly 15kByte) disc space.

Thx in advance
Gerd

6by9, 2017-01-27:
Is it just ```CONFIG_CAN_SLCAN=m``` that you want, or other options?

GBert, 2017-01-27:
` CONFIG_CAN_SLCAN=m` -> yes, that's it. And distribute the slcan.ko with the Raspbian relase, of course ;-)

popcornmix, 2017-01-30:
This has been added and is in latest rpi-update kernel.

Ruffio, 2017-02-05:
@GBert  so this issue can be closed now?

GBert, 2017-02-05:
Many Thx !

GBert, 2017-02-24:
slcan is missing again:
pi@raspberrypi:/lib/modules/4.9.11-v7+ $ find ./ -name slcan.ko
pi@raspberrypi:/lib/modules/4.9.11-v7+ $

pelwell, 2017-02-24:
That commit got dropped in the move to 4.9, but it's back in now and will be in the next build.

GBert, 2017-03-01:
Thx - it's back !

closed 2017-03-01 by GBert

AlisterAmo, 2017-10-13:
4.4.21 still missing slcan module