raspberrypi/linux #1820: kernel header always missing / can't compile modules

zoff99, 2017-01-31:
this happend on almost any version of raspian
now again with 4.4.44-v7+

also the modules (via apt-get) are old.
what can i do? 

```
sudo apt-get install v4l2loopback-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  v4l2loopback-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/31.2 kB of archives.
After this operation, 115 kB of additional disk space will be used.
Selecting previously unselected package v4l2loopback-dkms.
(Reading database ... 187015 files and directories currently installed.)
Preparing to unpack .../v4l2loopback-dkms_0.8.0-4_all.deb ...
Unpacking v4l2loopback-dkms (0.8.0-4) ...
Setting up v4l2loopback-dkms (0.8.0-4) ...
Loading new v4l2loopback-0.8.0 DKMS files...
```
First Installation: checking all kernels...
dpkg: warning: version '*-*' has bad syntax: version number does not start with digit
It is likely that 4.4.44-v7+ belongs to a chroot's host
Module build for the currently running kernel was skipped since the
kernel source for this kernel does not seem to be installed.

popcornmix, 2017-01-31:
This isn't a kernel issue. I suspect it's a distribution issue. Possibly due to missing the kernel headers package.
https://github.com/RPi-Distro/repo may be a better choice to report this, but I suspect that the answer is dkms is not supported. Ping @XECDesign who I believe has experimented with dkms.

zoff99, 2017-01-31:
not an issue with dkms. apt-get install can not install the headers for the current kernel most of the time

popcornmix, 2017-01-31:
I'll close this in favour of https://github.com/RPi-Distro/repo/issues/56

closed 2017-01-31 by popcornmix