raspberrypi/linux #1827: New Xin Mo device ID

stuntstein, 2017-02-02:
There is a new Xin Mo usb controller which has the same issue as fixed in 
https://github.com/raspberrypi/linux/commit/cb2c9e3f92480a292670e2cc261723ce4de8059e

I have created a patch which is working and is uploaded to pastbin: http://pastebin.com/4BMidcJV

Here is my description on retropie:
https://retropie.org.uk/forum/topic/7832/fix-up-left-joystick-direction-for-tht-2p-arcade-controller-with-device-id-0x75e1

Would it be possible to add the changes to the rpi repo?

Thanks.

popcornmix, 2017-02-06:
The patch is to upstream linux code, rather than anything Pi specific.
As such the patch should go upstream. See:
https://kernel.org/doc/html/latest/process/submitting-patches.html

Assuming the patch is accepted upstream we can cherry pick it here to get the changes sooner.

stuntstein, 2017-02-09:
Thanks for the info. Patch sent to Linux team.

P33M, 2017-05-04:
Fixed upstream https://github.com/torvalds/linux/commit/9257821c5a1dc57ef3a37f7cbcebaf548395c964

JamesH65, 2017-05-18:
@popcornmix Can now be cherry picked.

JamesH65, 2017-09-13:
Please update to the latest kernel which contains a fix for this issue.

closed 2017-09-13 by JamesH65