raspberrypi/linux #1828: Applying RT patches to 4.9.y

vanfanel, 2017-02-04:
Good morning,

It seems that while you jump from odd to odd version number (4.9.y is now on 4.9.7!) the rt patches go from even to even (it's now 4.9.6, from previous 4.9.4!).
So it's impossible to get the same "y" minor version number on both to apply the patch.
What's the suggested solution, please?

popcornmix, 2017-02-04:
Clone the rpi-4.9.y branch of the repo.
You can revert pi kernel tree one minor version of upstream linux:
```
git rebase -i --autosquash --onto v4.9.6 stable/linux-4.9.y rpi-4.9.y
```
Then you can apply whatever patches that come from 4.9.6.

But really these are git questions for the forum - nothing to do with this repo.

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65