raspberrypi/linux #1829: Add Mic Bias to linux/sound/soc/codecs/wm8753.c

charlton91, 2017-02-06:
Currently there is no option in this driver to enable the microphone biasing circuit for the WM8753.

The following should be added to as an item in the struct wm8753_snd_controls in order to enable the mic bias circuit.

`SOC_SINGLE("Mic Bias", WM8753_PWR1, 5, 1, 0),`

popcornmix, 2017-02-06:
This looks like an upstream kernel change. As such it should be submitted upstream:
https://kernel.org/doc/html/latest/process/submitting-patches.html

Assuming it is accepted upstream we can cherry-pick it.

HiassofT, 2017-02-06:
micbias is usually controlled via DAPM, and the WM8753 already contains the necessary dapm widgets - you only have to define the routes in your machine driver - i.e. connect the MICBIAS widget to your input widget.

See for example the dapm routes in my Cirrus card driver in PR #1810

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65