raspberrypi/linux #1836: bcm2835 v4l2 missing timestamps only in H.264

duvrai, 2017-02-10:
When configured to capture H.264 video from the Raspberry Camera Module v2.1 through the video4linux2 driver, ffmpeg complains that "Timestamps are unset", causing sync issues when combined with audio coming from an other device. As you can see from the ffmpeg output, "This is deprecated and will stop working in the future. Fix your code to set the timestamps properly"

Timestamps are correct when capturing MJPEG instead of H.264. I first tried with the default kernel (4.4.38-v7+) and after upgrading to bleeding edge (4.4.47-v7+) the problem persists. Here is some timestamp debug data I got from `ffmpeg -debug_ts -vcodec h264 -i /dev/video0 -c:v copy -t 0.1 timestamps.mkv`:

```
ffmpeg version N-83414-ge477f09 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --enable-gpl --enable-nonfree --enable-mmal --enable-omx-rpi --bindir=/usr/local/bin
  libavutil      55. 46.100 / 55. 46.100
  libavcodec     57. 75.100 / 57. 75.100
  libavformat    57. 66.101 / 57. 66.101
  libavdevice    57.  2.100 / 57.  2.100
  libavfilter     6. 72.100 /  6. 72.100
  libswscale      4.  3.101 /  4.  3.101
  libswresample   2.  4.100 /  2.  4.100
  libpostproc    54.  2.100 / 54.  2.100
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 5211.202638, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1024x768, 30 fps, 30 tbr, 1000k tbn, 2000k tbc
Output #0, matroska, to 'timestamps.mkv':
  Metadata:
    encoder         : Lavf57.66.101
    Stream #0:0: Video: h264 (High) (H264 / 0x34363248), yuv420p(progressive), 1024x768, q=2-31, 30 fps, 30 tbr, 1k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-5211202638 off_time:-5211.2
muxer <- type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:0 pkt_dts_time:0 size:12488
[matroska @ 0x2158b00] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:5211202638 pkt_pts_time:5211.2 pkt_dts:5211202638 pkt_dts_time:5211.2 off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-5211202638 off_time:-5211.2
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:2351
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:5211235965 pkt_pts_time:5211.24 pkt_dts:5211235965 pkt_dts_time:5211.24 off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:33327 pkt_pts_time:0.033327 pkt_dts:33327 pkt_dts_time:0.033327 off:-5211202638 off_time:-5211.2
muxer <- type:video pkt_pts:33 pkt_pts_time:0.033 pkt_dts:33 pkt_dts_time:0.033 size:2199
demuxer -> ist_index:0 type:video next_dts:99993 next_dts_time:0.099993 next_pts:99993 next_pts_time:0.099993 pkt_pts:5211269292 pkt_pts_time:5211.27 pkt_dts:5211269292 pkt_dts_time:5211.27 off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:66654 pkt_pts_time:0.066654 pkt_dts:66654 pkt_dts_time:0.066654 off:-5211202638 off_time:-5211.2
muxer <- type:video pkt_pts:67 pkt_pts_time:0.067 pkt_dts:67 pkt_dts_time:0.067 size:5292
demuxer -> ist_index:0 type:video next_dts:133320 next_dts_time:0.13332 next_pts:133320 next_pts_time:0.13332 pkt_pts:5211302619 pkt_pts_time:5211.3 pkt_dts:5211302619 pkt_dts_time:5211.3 off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:99981 pkt_pts_time:0.099981 pkt_dts:99981 pkt_dts_time:0.099981 off:-5211202638 off_time:-5211.2
muxer <- type:video pkt_pts:100 pkt_pts_time:0.1 pkt_dts:100 pkt_dts_time:0.1 size:7874
demuxer -> ist_index:0 type:video next_dts:166647 next_dts_time:0.166647 next_pts:166647 next_pts_time:0.166647 pkt_pts:5211335946 pkt_pts_time:5211.34 pkt_dts:5211335946 pkt_dts_time:5211.34 off:-5211202638 off_time:-5211.2
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:133308 pkt_pts_time:0.133308 pkt_dts:133308 pkt_dts_time:0.133308 off:-5211202638 off_time:-5211.2
frame=    5 fps=0.0 q=-1.0 Lsize=      30kB time=00:00:00.10 bitrate=2451.6kbits/s speed=53.6x    
video:29kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 2.476493%
```

In MJPEG I get correct timestamps (and they seem to sync quite well when I add USB audio). From `ffmpeg -debug_ts -vcodec mjpeg -i /dev/video0 -c:v copy -t 0.1 timestamps.mkv`:

```
[...]
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:6161714914 pkt_pts_time:6161.71 pkt_dts:6161714914 pkt_dts_time:6161.71 off:-6161714914 off_time:-6161.71
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-6161714914 off_time:-6161.71
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:53961
demuxer -> ist_index:0 type:video next_dts:33333 next_dts_time:0.033333 next_pts:33333 next_pts_time:0.033333 pkt_pts:6161748242 pkt_pts_time:6161.75 pkt_dts:6161748242 pkt_dts_time:6161.75 off:-6161714914 off_time:-6161.71
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:33328 pkt_pts_time:0.033328 pkt_dts:33328 pkt_dts_time:0.033328 off:-6161714914 off_time:-6161.71
muxer <- type:video pkt_pts:33 pkt_pts_time:0.033 pkt_dts:33 pkt_dts_time:0.033 size:44384
demuxer -> ist_index:0 type:video next_dts:66661 next_dts_time:0.066661 next_pts:66661 next_pts_time:0.066661 pkt_pts:6161781569 pkt_pts_time:6161.78 pkt_dts:6161781569 pkt_dts_time:6161.78 off:-6161714914 off_time:-6161.71
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:66655 pkt_pts_time:0.066655 pkt_dts:66655 pkt_dts_time:0.066655 off:-6161714914 off_time:-6161.71
muxer <- type:video pkt_pts:67 pkt_pts_time:0.067 pkt_dts:67 pkt_dts_time:0.067 size:44177
demuxer -> ist_index:0 type:video next_dts:99988 next_dts_time:0.099988 next_pts:99988 next_pts_time:0.099988 pkt_pts:6161814896 pkt_pts_time:6161.81 pkt_dts:6161814896 pkt_dts_time:6161.81 off:-6161714914 off_time:-6161.71
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:99982 pkt_pts_time:0.099982 pkt_dts:99982 pkt_dts_time:0.099982 off:-6161714914 off_time:-6161.71
muxer <- type:video pkt_pts:100 pkt_pts_time:0.1 pkt_dts:100 pkt_dts_time:0.1 size:44061
demuxer -> ist_index:0 type:video next_dts:133315 next_dts_time:0.133315 next_pts:133315 next_pts_time:0.133315 pkt_pts:6161848224 pkt_pts_time:6161.85 pkt_dts:6161848224 pkt_dts_time:6161.85 off:-6161714914 off_time:-6161.71
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:133310 pkt_pts_time:0.13331 pkt_dts:133310 pkt_dts_time:0.13331 off:-6161714914 off_time:-6161.71
frame=    4 fps=0.0 q=-1.0 Lsize=     183kB time=00:00:00.10 bitrate=14843.9kbits/s speed=0.738x    
video:182kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.440019%
```

[... 19 lines cut ...]
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)
```
After `rpi-update`
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.47-v7+ #961 SMP Sun Feb 5 20:17:00 GMT 2017 armv7l GNU/Linux

pi@raspberrypi:~ $ modinfo bcm2835-v4l2
filename:       /lib/modules/4.4.47-v7+/kernel/drivers/media/platform/bcm2835/bcm2835-v4l2.ko
version:        0.0.2
license:        GPL
author:         Vincent Sanders
description:    Broadcom 2835 MMAL video capture
srcversion:     A7D6DBEB2D67BE83A6FC776
depends:        videobuf2-v4l2,videodev,videobuf2-vmalloc,videobuf2-core,v4l2-common
intree:         Y
vermagic:       4.4.47-v7+ SMP mod_unload modversions ARMv7 
parm:           debug:int
parm:           bcm2835_v4l2_debug:Debug level 0-2
parm:           video_nr:videoX start numbers, -1 is autodetect (array of int)
parm:           max_video_width:Threshold for video mode (int)
parm:           max_video_height:Threshold for video mode (int)
parm:           gst_v4l2src_is_broken:If non-zero, enable workaround for Gstreamer (int)
```

duvrai, 2017-02-10:
I don't know what upsets ffmpeg. I just did `modprobe bcm2835-v4l2 debug=2` to compare

h264 bcm2835-v4l2 debug:
```
[15724.022393] bcm2835-v4l2: Start time 15727297908 size 8
[15724.024336] bcm2835-v4l2: buffer_cb: status:0, buf:af631e00, length:27, flags 36, pts 0
[15724.026022] bcm2835-v4l2: buffer_prepare: dev:b52e4800
[15724.026036] bcm2835-v4l2: buffer_queue: dev:b52e4800 buf:af631e00
[15724.063994] bcm2835-v4l2: buffer_cb: status:0, buf:af631000, length:2108, flags 12, pts 15727299653
[15724.064006] bcm2835-v4l2: Convert start time 15723.859973 and 15727297908 with offset 15727299653 to 15723.861718
[15724.064051] bcm2835-v4l2: buffer_prepare: dev:b52e4800
[15724.064059] bcm2835-v4l2: buffer_queue: dev:b52e4800 buf:af631000
[15724.096496] bcm2835-v4l2: buffer_cb: status:0, buf:af630200, length:567, flags 4, pts 15727332981
[15724.096507] bcm2835-v4l2: Convert start time 15723.859973 and 15727297908 with offset 15727332981 to 15723.895046
```

ffmpeg says for the *first* frame: `pkt_pts:NOPTS` and `pkt_dts:NOPTS` but `off:-15723895046` seems to be the *second* frame from bcm2835-v4l2 debug above.
```
Press [q] to stop, [?] for help
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-15723895046 off_time:-15723.9
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-15723895046 off_time:-15723.9
muxer <- type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:0 pkt_dts_time:0 size:2135
[matroska @ 0x24ac330] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:15723895046 pkt_pts_time:15723.9 pkt_dts:15723895046 pkt_dts_time:15723.9 off:-15723895046 off_time:-15723.9
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-15723895046 off_time:-15723.9
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:567
```

bcm2835-v4l2 debug with mjpeg:
```
[15468.630002] bcm2835-v4l2: Start time 15471904865 size 8
[15468.633330] bcm2835-v4l2: buffer_cb: status:0, buf:  (null), length:0, flags 0, pts 0
[15468.673252] bcm2835-v4l2: buffer_cb: status:0, buf:af663800, length:15246, flags 12, pts 15471911855
[15468.673265] bcm2835-v4l2: Convert start time 15468.469767 and 15471904865 with offset 15471911855 to 15468.476757
[15468.677568] bcm2835-v4l2: buffer_prepare: dev:b52e4800
[15468.677582] bcm2835-v4l2: buffer_queue: dev:b52e4800 buf:af663800
[15468.705275] bcm2835-v4l2: buffer_cb: status:0, buf:af662000, length:16427, flags 12, pts 15471945182
[15468.705287] bcm2835-v4l2: Convert start time 15468.469767 and 15471904865 with offset 15471945182 to 15468.510084
```
now ffmpeg gets actual values `pkt_pts:15468476757` and `pkt_dts:15468476757` corresponding to the first frame in bcm2835-v4l2 debug output:
```
Press [q] to stop, [?] for help
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:15468476757 pkt_pts_time:15468.5 pkt_dts:15468476757 pkt_dts_time:15468.5 off:-15468476757 off_time:-15468.5
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-15468476757 off_time:-15468.5
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:15246
demuxer -> ist_index:0 type:video next_dts:33333 next_dts_time:0.033333 next_pts:33333 next_pts_time:0.033333 pkt_pts:15468510084 pkt_pts_time:15468.5 pkt_dts:15468510084 pkt_dts_time:15468.5 off:-15468476757 off_time:-15468.5
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:33327 pkt_pts_time:0.033327 pkt_dts:33327 pkt_dts_time:0.033327 off:-15468476757 off_time:-15468.5
muxer <- type:video pkt_pts:33 pkt_pts_time:0.033 pkt_dts:33 pkt_dts_time:0.033 size:16427
```

duvrai, 2017-02-10:
In short: timestamps are missing/corrupted for the first frame, subsequent frames are ok.

6by9, 2017-02-10:
It looks like there is an issue on both sides.
I've fixed the V4L2 driver to return the start time for the header bytes, and ffmpeg 3.1.1 that I'm running still complains. What version are you running as you appear to have an odd hash as your version number (N-83414-ge477f09). Self built? 
```
Output #0, matroska, to 'timestamps.mkv':
  Metadata:
    encoder         : Lavf57.41.100
    Stream #0:0, 0, 1/1000: Video: h264, 1 reference frame (H264 / 0x34363248), yuv420p(left), 1024x768 (0x0), 0/1, q=2-31, -5 kb/s, 30 fps, 30 tbr, 1k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-124185976 off_time:-124.186
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-124185976 off_time:-124.186
muxer <- type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:0 pkt_dts_time:0 size:19951
[matroska @ 0x59a530] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[matroska @ 0x59a530] Writing block at offset 633, size 19951, pts 0, dts 0, duration 0, keyframe 1
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:124185976 pkt_pts_time:124.186 pkt_dts:124185976 pkt_dts_time:124.186 off:-124185976 off_time:-124.186
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-124185976 off_time:-124.186
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:2382
[matroska @ 0x59a530] Writing block at offset 20592, size 2382, pts 0, dts 0, duration 0, keyframe 0
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:124219291 pkt_pts_time:124.219 pkt_dts:124219291 pkt_dts_time:124.219 off:-124185976 off_time:-124.186
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:33315 pkt_pts_time:0.033315 pkt_dts:33315 pkt_dts_time:0.033315 off:-124185976 off_time:-124.186
muxer <- type:video pkt_pts:33 pkt_pts_time:0.033 pkt_dts:33 pkt_dts_time:0.033 size:3834
[matroska @ 0x59a530] Writing block at offset 22981, size 3834, pts 33, dts 33, duration 0, keyframe 0
demuxer -> ist_index:0 type:video next_dts:99981 next_dts_time:0.099981 next_pts:99981 next_pts_time:0.099981 pkt_pts:124252608 pkt_pts_time:124.253 pkt_dts:124252608 pkt_dts_time:124.253 off:-124185976 off_time:-124.186
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:66632 pkt_pts_time:0.066632 pkt_dts:66632 pkt_dts_time:0.066632 off:-124185976 off_time:-124.186
```
vs extended kernel log
```
[  124.391059] bcm2835-v4l2: Start time 129167509 size 8
[  124.393326] bcm2835-v4l2: buffer_cb: status:0, buf:ac7ffe00, length:27, flags 36, pts 0
[  124.393330] bcm2835-v4l2: Buffer time set as start timestamp
[  124.393334] bcm2835-v4l2: Buffer has ts 124150932000
[  124.395451] bcm2835-v4l2: buffer_prepare: dev:af22b800
[  124.395460] bcm2835-v4l2: buffer_queue: dev:af22b800 buf:ac7ffe00
[  124.433329] bcm2835-v4l2: buffer_cb: status:0, buf:ac7ffc00, length:19924, flags 12, pts 129169236
[  124.433338] bcm2835-v4l2: Convert start time 124.150932 and 129167509 with offset 129169236 to 124.152659
[  124.433341] bcm2835-v4l2: Buffer has ts 124152659000
[  124.433482] bcm2835-v4l2: buffer_prepare: dev:af22b800
[  124.433489] bcm2835-v4l2: buffer_queue: dev:af22b800 buf:ac7ffc00
[  124.464079] bcm2835-v4l2: buffer_cb: status:0, buf:ac7ffa00, length:2382, flags 4, pts 129202553
[  124.464085] bcm2835-v4l2: Convert start time 124.150932 and 129167509 with offset 129202553 to 124.185976
[  124.464088] bcm2835-v4l2: Buffer has ts 124185976000
[  124.464580] bcm2835-v4l2: buffer_prepare: dev:af22b800
[  124.464587] bcm2835-v4l2: buffer_queue: dev:af22b800 buf:ac7ffa00
[  124.499212] bcm2835-v4l2: buffer_cb: status:0, buf:ac7ff800, length:3834, flags 4, pts 129235868
[  124.499225] bcm2835-v4l2: Convert start time 124.150932 and 129167509 with offset 129235868 to 124.219291
[  124.499228] bcm2835-v4l2: Buffer has ts 124219291000
[  124.502323] bcm2835-v4l2: buffer_prepare: dev:af22b800
[  124.502331] bcm2835-v4l2: buffer_queue: dev:af22b800 buf:ac7ff800
[  124.531287] bcm2835-v4l2: buffer_cb: status:0, buf:ac7ff600, length:4895, flags 4, pts 129269185
[  124.531295] bcm2835-v4l2: Convert start time 124.150932 and 129167509 with offset 129269185 to 124.252608
[  124.531299] bcm2835-v4l2: Buffer has ts 124252608000
```
All those kernel log timestamps look valid to me. The interesting bit is that the kernel reports delivering buffers of size 27, 19924, 2382, 3834. ffmpeg is reporting frames of 19951, 2382, 3834. It has combined the header bytes into the first frame, and I suspect it has dropped the timestamp in the process.

http://lists.ffmpeg.org/pipermail/ffmpeg-devel-irc/2016-March/003424.html from 13:38:02 CET onwards appears to indicate that there is an issue in ffmpeg if handling raw H264 streams.

I'll look into making sure that the V4L2 timestamps are correct (mostly done already), but I'm not going to go debugging FFmpeg in any detail.

duvrai, 2017-02-10:
Thanks @6by9 for all the work and the quick response! I'll try your modifications when they are ready and file a bug with ffmpeg if needed.

I built my ffmpeg from the master git branch because I couldn't find a binary distribution. (Compiling times on the pi 3 are now bearable with `make -j4`)

I also tried libav's avconv, but couldn't get it to work at all with v4l2.

6by9, 2017-02-10:
I'll dig out my diffs for V4L2 and push them to my fork of the kernel branch if you fancy rebuilding them. I have been working against the 4.9 tree instead of 4.4, but that shouldn't make a big difference.

As it happens I pulled down and kicked off a build of ffmpeg before I left the office. Hopefully it'll build successfully and be waiting for me on Monday.

6by9, 2017-02-13:
Sorry, I'm going to have to say this appears to be FFMPEG mishandling header bytes from V4L2 devices.
With my rebuilt version from top of tree + some extra logging, the V4L2 of FFMPEG is seeing
```
Probing video4linux2,v4l2 score:99 size:0
[video4linux2,v4l2 @ 0x1828220] fd:3 capabilities:85200005
[video4linux2,v4l2 @ 0x1828220] Current input_channel: 0, input_name: Camera 0, input_std: 0
[video4linux2,v4l2 @ 0x1828220] Querying the device for the current frame size
[video4linux2,v4l2 @ 0x1828220] Setting frame size to 1024x768
[video4linux2,v4l2 @ 0x1828220] timestamp 240223992983, length 27, flags 00002001
[video4linux2,v4l2 @ 0x1828220] timestamp 240223993105, length 19407, flags 00002009
[video4linux2,v4l2 @ 0x1828220] timestamp 240224026419, length 1815, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 7, nal_ref_idc: 1
[h264 @ 0x18292f0] nal_unit_type: 8, nal_ref_idc: 1
[h264 @ 0x18292f0] nal_unit_type: 5, nal_ref_idc: 1
[h264 @ 0x18292f0] Reinit context to 1024x768, pix_fmt: yuv420p
[video4linux2,v4l2 @ 0x1828220] timestamp 240224059734, length 924, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224093049, length 4076, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224126366, length 6912, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224159680, length 8324, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224192996, length 12513, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224226311, length 15426, flags 00002001
[h264 @ 0x18292f0] nal_unit_type: 1, nal_ref_idc: 1
[video4linux2,v4l2 @ 0x1828220] timestamp 240224259626, length 17226, flags 00002001
[video4linux2,v4l2 @ 0x1828220] All info found
[video4linux2,v4l2 @ 0x1828220] stream 0: start_time: 240224.026 duration: -9223372036854.775
[video4linux2,v4l2 @ 0x1828220] format: start_time: 240224.026 duration: -9223372036854.775 bitrate=0 kb/s
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 240224.026419, bitrate: N/A
    Stream #0:0, 8, 1/1000000: Video: h264 (High), 1 reference frame, yuv420p(progressive, left), 1024x768, 0/1, 30 fps, 30 tbr, 1000k tbn, 2000k tbc
Successfully opened the file.
Parsing a group of options: output url timestamps.mkv.
Applying option c:v (codec name) with argument copy.
Applying option t (record or transcode "duration" seconds of audio/video) with argument 0.5.
Successfully parsed a group of options.
Opening an output file: timestamps.mkv.
[file @ 0x184c360] Setting default whitelist 'file,crypto'
Successfully opened the file.
Output #0, matroska, to 'timestamps.mkv':
  Metadata:
    encoder         : Lavf57.66.101
    Stream #0:0, 0, 1/1000: Video: h264 (High), 1 reference frame (H264 / 0x34363248), yuv420p(progressive, left), 1024x768 (0x0), 0/1, q=2-31, 30 fps, 30 tbr, 1k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
demuxer -> ist_index:0 type:video next_dts:NOPTS next_dts_time:NOPTS next_pts:NOPTS next_pts_time:NOPTS pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-240224026419 off_time:-240224
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:NOPTS pkt_dts_time:NOPTS off:-240224026419 off_time:-240224
muxer <- type:video pkt_pts:NOPTS pkt_pts_time:NOPTS pkt_dts:0 pkt_dts_time:0 size:19434
[matroska @ 0x187f990] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[matroska @ 0x187f990] s->oformat->flags 00030440, st->disposition 00000000, pts 9223372036854775808, dts 0
[matroska @ 0x187f990] get_metadata_duration returned: 0
[matroska @ 0x187f990] Write early duration from recording time = 500
[matroska @ 0x187f990] Writing block at offset 9, size 19434, pts 0, dts 0, duration 0, keyframe 1
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:240224026419 pkt_pts_time:240224 pkt_dts:240224026419 pkt_dts_time:240224 off:-240224026419 off_time:-240224
demuxer+ffmpeg -> ist_index:0 type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 off:-240224026419 off_time:-240224
muxer <- type:video pkt_pts:0 pkt_pts_time:0 pkt_dts:0 pkt_dts_time:0 size:1815
[matroska @ 0x187f990] s->oformat->flags 00030440, st->disposition 00000000, pts 0, dts 0
[matroska @ 0x187f990] Writing block at offset 19451, size 1815, pts 0, dts 0, duration 0, keyframe 0
demuxer -> ist_index:0 type:video next_dts:66666 next_dts_time:0.066666 next_pts:66666 next_pts_time:0.066666 pkt_pts:240224059734 pkt_pts_time:240224 pkt_dts:240224059734 pkt_dts_time:240224 off:-240224026419 off_time:-240224
```
(I've just added the code to correctly signal key frames through V4L2 in case FFMPEG was needing to see those, and that is why the second buffer has flags 0x2009)
Those buffers coming from V4L2 are correct and have valid timestamps.

```
[video4linux2,v4l2 @ 0x1828220] stream 0: start_time: 240224.026 duration: -9223372036854.775
[video4linux2,v4l2 @ 0x1828220] format: start_time: 240224.026 duration: -9223372036854.775 
```
[... cut ...]
Then again something has taken the header bytes and I-frame and combined them into a single buffer with pts=AV_NOPTS_VALUE (which is also what triggers the warning), but there's too much code for me to trawl through to try and find where. Hopefully the FFMPEG developers would be able to identify that pretty quickly.

I'll push my driver updates (all kernel side), but then I'm going to have to park this waiting on more information.
There are only a couple of V4L2 cameras that I know of that can produce H264. The Pi is one, and the Logitech C920/930 being the other, so this is quite probably a fairly unusual case within FFMPEG.

6by9, 2017-02-13:
Updates pushed to https://github.com/6by9/linux rpi-4.9.y branch.
The timestamp code changed between 4.4 and 4.9, so backporting is a bit of a faff.

duvrai, 2017-02-13:
Ok I'll try 4.9

I noticed that the timestamps generated by the bcm2835 driver are boot-time based, while the other devices all use UNIX time. Is that also a bug?

6by9, 2017-02-13:
> I noticed that the timestamps generated by the bcm2835 driver are boot-time based, while the other devices all use UNIX time. Is that also a bug?

Nope. All our buffers have the V4L2_BUF_FLAG_TIMESTAMP MONOTONIC flag set, which means they are with respect to clock_gettime(CLOCK_MONOTONIC). 
```man clock_gettime```
> CLOCK_MONOTONIC
> Clock that cannot be set and represents monotonic time since some unspecified  starting  point.
This  clock  is  not  affected  by  discontinuous jumps in the system time (e.g., if the system
administrator manually changes the clock), but is affected by the incremental adjustments  per‐
formed by adjtime(3) and NTP.

The unspecified starting point happens to be boot time.

JamesH65, 2017-05-18:
@6by9 Are we done with this one?

JamesH65, 2017-05-23:
@6by9 does the linked commit fix this issue?

6by9, 2017-05-23:
I don't know.
I haven't tested with FFMPEG, and I can't remember whether I was fully happy with the results when using inline headers. 
It'd be helpful if @duvrai could retest, otherwise I guess it gets added to my to-do list.

JamesH65, 2017-09-13:
ping @6by9 @duvrai. Any news on this, or ready for closing?

6by9, 2017-09-13:
Nothing more from me. Waiting for feedback, but as nothing for 6 months probably a candidate for closing.

JamesH65, 2017-12-04:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-12-04 by JamesH65

zhoub, 2018-06-02:
Sorry I still have this issue from RaspiVid 1.3.12 with latest 3 Model B+, with Camera V2.

The ffmpeg shows
```
Elodia:raspi-build Bo$ ffmpeg -r 30 -i ~/Downloads/Hand-720p.h264 -vcodec copy Hand-720p.mp4
ffmpeg version 4.0 Copyright (c) 2000-2018 the FFmpeg developers
  built with Apple LLVM version 9.1.0 (clang-902.0.39.1)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.0 --enable-shared --enable-pthreads --enable-version3 --enable-hardcoded-tables --enable-avresample --cc=clang --host-cflags= --host-ldflags= --enable-gpl --enable-libmp3lame --enable-libx264 --enable-libxvid --enable-opencl --enable-videotoolbox --disable-lzma
  libavutil      56. 14.100 / 56. 14.100
  libavcodec     58. 18.100 / 58. 18.100
  libavformat    58. 12.100 / 58. 12.100
  libavdevice    58.  3.100 / 58.  3.100
  libavfilter     7. 16.100 /  7. 16.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  1.100 /  5.  1.100
  libswresample   3.  1.100 /  3.  1.100
  libpostproc    55.  1.100 / 55.  1.100
Input #0, h264, from '/Users/Bo/Downloads/Hand-720p.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 1280x720, 25 fps, 25 tbr, 1200k tbn, 50 tbc
Output #0, mp4, to 'Hand-720p.mp4':
  Metadata:
    encoder         : Lavf58.12.100
    Stream #0:0: Video: h264 (High) (avc1 / 0x31637661), yuv420p(progressive), 1280x720, q=2-31, 25 fps, 25 tbr, 15360 tbn, 30 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
[mp4 @ 0x7f8daf09c800] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
```

6by9, 2018-07-02:
@zhoub You appear to be trying to process the H264 elementary stream presumably from Raspivid, not using the V4L2 driver. Your issue is therefore not the same as being discussed on this issue and is a new issue.
In this case it appears also to be an issue with FFmpeg and not the kernel, therefore this isn't the correct place to report it.

If you're only worrying about the log message
```
[mp4 @ 0x7f8daf09c800] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
```
then it is only a warning and shouldn't stop the stream from being muxed correctly.