raspberrypi/linux #1842: Crash when using two USB audio interfaces plugged to the same hub: usb_interface_failed (-110)

jpetazzo, 2017-02-11:
Executive summary: this is a really weird bug that happens only in one specific combination: when plugging two USB audio interfaces to the same (powered) USB hub. Alas, that combination happens to be the one that _should work the best in theory_, and the problem is hard to workaround (when it happens, the Pi loses network connectivity and gets unresponsive). I tried to switch many things (power supplies, kernels, hardware...), to no avail.

What I do: if I try to use two USB audio interfaces at the same time (I'm using Pulseaudio's `combine` module to make a multi-room hifi system by the way, and it's going to be _awesome_ once I find a workaround for this issue! 👍 ), my Raspberry Pi 3 crashes immediately _but only if both USB audio interfaces are plugged to the same hub_. The hub is powered, and according to my multimeter, it's not "backfeeding" power to the Pi.

If I enable kernel messages on the console, I see (after the USB device detection):
```
Transfer to device 6 endpoint 0x2 frame 793 failed - FIQ timed out. Data may have been lost.
usb 1-1.5.3: 2:1: usb_set_interface failed (-110)
```
The last line is repeated every 5-10 seconds, then I get a watchdog timeout for eth0, then the `usb_set_interface failed` message a bunch of more times. Then I powercycle the Pi out of despair and I try again.

I tried with a different Pi (model 3 as well), just moving the SDCARD to the other Pi: same problem.

I tried with a different Pi _and_ a different SDCARD: same problem.

I tried with different USB audio interfaces (I have ~10 of them, all different). I tried ~10 different combinations, and all of them led to the same problem: instant crash when I plug the 2nd interface and Pulseaudio tries to open it.

I tried with different kernels (4.4.34-930, 4.4.38-938, +another more recent one that I got with rpi-update -- will have to lookup the exact version if necessary, since I forgot to write it down and re-imaged the SDCARD since then!)

I tried with a "good" power supply (the one that comes with the Vilros kit, rated for 2.5A) as well as with the "charging ports" of the powered hub as well as with a few "big" USB charging hubs – no difference at all.

I have a USB multimeter, that I attached to the Pi inputs and outputs; and it did not register any significant voltage fluctuation when plugging or opening the audio interfaces.

Now, hold on to your systemd, things get _really weird_. 

I tried the following things:
- connect both USB audio interfaces directly to the Pi (works)
- connect one USB audio interface directly to the Pi, the other one to an external powered hub (works)
- connect both USB audio interfaces to two different external powered hubs (works)
- connect both USB audio interfaces to the external powered hub in my monitor (works)
- connect both USB audio interfaces to a specific external powered hub (sometimes works, sometimes crashes, depending on which ports I use)

I investigated the last scenario, and found out that this 7-ports hub was, in fact, 2 4-ports hubs chained together. The test results become:
- connect both USB audio interfaces to ports belonging to different "sub-hubs" (works)
- connect both USB audio interfaces to ports belonging to the same "sub-hub" (crash)

Conclusion: _the problem happens only when both USB audio interfaces are connected to the same hub (in the USB topology sense), and only with that specific USB hub._

(I've seen some weird stuff in the past, but this is A++ bug right?)

Note that the problem happens only when _opening_ the audio interfaces. Plugging the interfaces is fine; but as soon as Pulseaudio opens them, the Pi becomes irresponsive (no Ethernet, no keyboard; didn't try WiFi).

I tried to get proper kernel logs and traces with [netconsole], but unfortunately, the Pi's ethernet interface doesn't seem to support polling. So I attached screen captures if that can be of any help.

I can provide usbmon traces if necessary (everything appears to stop after a volley of `Z` urbs, but I'm no USB wizard so I have no idea what I'm doing when I read these things!)

USB bus topology, with both audio interfaces plugged into my monitor's hub, itself plugged into the external powered hub:
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M (Pi's hub)
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx
        |__ Port 5: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M (External hub, "parent sub-hub")
            |__ Port 1: Dev 11, If 0, Class=Hub, Driver=hub/4p, 480M (Monitor's hub)
                |__ Port 1: Dev 13, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 1: Dev 13, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 1: Dev 13, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 1: Dev 13, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 2: Dev 14, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 2: Dev 14, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 2: Dev 14, If 2, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 2: Dev 14, If 3, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 4: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
                |__ Port 4: Dev 12, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
            |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M (External hub, "child sub-hub")
```

`lsusb` for the same topology:
```
Bus 001 Device 005: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 012: ID 17f6:0905 Unicomp, Inc
Bus 001 Device 014: ID 0d8c:0014 C-Media Electronics, Inc.
Bus 001 Device 013: ID 0d8c:0020 C-Media Electronics, Inc.
Bus 001 Device 011: ID 0424:2134 Standard Microsystems Corp.
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Vendor/product IDs involved:
- audio interfaces: `0d8c:0014`, `0d8c:0020`, and others
- monitor's hub: `0424:2134`
- external hub: `1a40:0101`

Other details:
- plugging one audio interfaces to a non-powered hub doesn't work (the Pi doesn't crash, but the USB device is not initialized properly, apparently because it claims more power than the USB port is willing to provide)
- plugging multiple audio interfaces directly to the Pi works, but in the long run, audio quality deteriorates (the more interfaces I add, the faster it deteriorates). I suspected a power issue: perhaps over time, the main polyfuse does a "slow blow" and lets less and less current in? (Again, I tried with a bunch of different power supplies, no difference). That's how I ended up trying to power my USB audio interfaces with an external hub ... And that's how I fell in this rabbit hole. Except instead of rabbits there are snakes and while I'm not fluent in snakeish, I think they're currently debating whether I would taste good with ketchup. _Pls snd halp! Thx._

Screenshots:
[image]
[image]

[netconsole]: https://www.kernel.org/doc/Documentation/networking/netconsole.txt

P33M, 2017-02-12:
Your hub has a single transaction translator. This is a known limitation when using full-speed devices in these hubs.

See: 4. Single-TT USB hubs in the site documentation:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md

jpetazzo, 2017-02-12:
_Your raspberry-fu is very impressive._ 🙇 

OK, I don't know how I missed that. After looking for more information on TT, I found [this KB nugget](http://www.cypress.com/knowledge-base-article/how-check-whether-usb-20-hub-multi-tt-or-single-tt-kba93279) which led me to look at the `bDeviceProtocol` field of my USB hubs, and _yes indeed_, in the `lsusb -v` output, it shows clearly `Single TT` (for my external hub) vs `Multi TT` (for my monitor hub). The [Wikipedia USB page](https://en.wikipedia.org/wiki/USB_hub#Transaction_translator) also has a bunch of extra info. I feel more educated 🤓 . Thank you for solving this mystery case so fast!

For the sake of the experiment, I tried again on a Linux PC; and indeed, when I plug at least 3 interfaces to the same (single TT) hub, I get a bunch of `Not enough bandwidth for new device state.` and `usb_set_interface failed (-28)` kernel messages. That makes sense! (But the machine doesn't crash.)

Do you know if the issue is:
- specific to the USB controller in the Pi?
- specific to the OTG mode (and affecting other devices like phones, tablets, etc) ?
- other?

Since crashes are rather frustrating and unexpected, do you think it's worth trying to come up with e.g. at least a kernel warning when detecting multiple full-speed devices on a single TT hub? Or this is considered a too specific corner case?

Do you want me to do a bunch of testing involving e.g. keyboards and mouses?

(For now I'm connecting one sound interface per hub and will be looking for a Multi-TT hub. Thanks again for your help!)

P33M, 2017-02-13:
There's two things that could be done to improve performance.

Currently dwc_otg's scheduling algorithm is too optimistic. It assumes all of a downstream TT's bandwidth budget can be made use of, when in fact hardware limitations mean that only ~45% can be used by periodic transfers. Tightening up the scheduling would prevent submitting transfers to devices that would result in error messages.

The "everything falls in a big heap" problem is most likely a dwc_otg bug. Do you have a minimal case (alsa config/set of commands) that would allow me to replicate your issue?

jpetazzo, 2017-02-19:
O mighty slayer of bugs! I found a (relatively simple) way to reproduce the issue.

Setup:
- Raspberry Pi 3
- Ethernet connection
- Two USB audio interfaces connected to a Single TT Hub

Repro steps:
1. Boot with a fresh Raspbian Lite image (I used 2017-01-11). The only thing I did to prepare the image was to touch `/boot/ssh` to do everything over SSH.
2. Log in with the `pi` user.
3. Install `sox` because it's the simplest thing I found to generate test tones:
   ```
   sudo apt-get install sox -y
   ```
4. Play a synth test tone over all 3 interfaces at the same time:
   ```
   for N in 0 1 2; do AUDIODEV=hw:$N,0 play -q -r 44100 -n synth & done
   ```
5. The Pi crashes almost instantly 😢 

Useful commands:
- To stop the playback of the test tone, just do `kill %1 %2 %3; wait`
- If it doesn't crash but you don't hear the test tones, make sure that volumes are set properly; I use the following snippet to max all volumes:
   ```
   for CARD in 0 1 2; do
     for CONTROL in PCM Speaker; do
       amixer -c $CARD sset $CONTROL 100%
     done
   done
   ```

Now, I have a few extra observations.
- The bug _doesn't seem to happen_ when setting the bitrate to 48000 instead of 44100. That's particularly annoying, because my MPD setup defaults to 44100, but sox defaults to 48000 (which is why it took me so long to find an easy way to reproduce the problem!)
- If I put the USB interfaces on different hubs, the system doesn't crash, but the sound is heavily garbled. Playing on one interface at a time works fine, but two interfaces at a time yields light distortion, and three interfaces yields heavy distortion.

Since I _still_ get distortion when putting the interfaces on different hubs, I'm going to try to setup MPD to 48kHz (and try to find a good Multi TT hub, too).

Thanks again for your help!

jpetazzo, 2017-02-21:
Following up with the results of my test matrix:

Bit rate | Interfaces connected to ... | Result
--------|---------------------------|---------
44100 | Pi | sound eventually* becomes garbled
48000 | Pi | ?
44100 | Single TT hub | immediate crash
48000 | Single TT hub | sound is garbled
44100 | 1 interface per Single TT hub | sound eventually* becomes garbled
48000 | 1 interface per Single TT hub | OK

*_Eventually_ means that sometimes sound issue will appear immediately; sometimes it will take a few hours (or a day or two) before the problem appears.

_I will update this table when I get more results._

jpetazzo, 2017-03-07:
Switching the whole audio pipeline to 48 KHz solved the problem (it has been stable for days on, without audio distortion).

I will repeat my tests once in a while (when new kernels are released) and report back; but feel free to close this issue if you think it doesn't need to stay open.

Thanks again!

P33M, 2017-03-07:
Can you post a full lsusb -v with the sound devices connected, for completeness?

jpetazzo, 2017-03-30:
Here it is. Sorry for the delay. (`lsusb -t` and `lsusb -v`)

[lsusb.t.txt](https://github.com/raspberrypi/linux/files/880709/lsusb.t.txt)
[lsusb.v.txt](https://github.com/raspberrypi/linux/files/880710/lsusb.v.txt)

P33M, 2017-05-03:
I can replicate the issue. It's more serious than not being able to talk to the downstream audio devices; pinging on the USB network device also stops working.

P33M, 2017-05-03:
On the bus analyzer, two very odd (and bad) things happen.

Because we have two sound devices active, we queue two isochronous packets to the same hub address and port (single TT hub). We don't start the isochronous transfers from the driver, we let the FIQ code (enabled by fiq_fsm_enable) handle sequencing as outbound transfers benefit better from our limited ability to pipeline transfer phases in the FIQ.

After the USB core finishes the first transfer, the FIQ searches for and starts the next transfer to the same TT. This transfer completes successfully and looks valid, but then I get a flurry of bogus transfers with bogus lengths in the same microframe (which shouldn't be possible).

The state machine is doing something wrong when pipelining isochronous OUT transfers that span only one packet. You don't see this behaviour at 48000Hz because the nominal transfer size is 192 bytes for stereo (sent every 1ms) which always spans 2 packets vs 1 packet for 44100Hz.

Worse than the FIQ software bug, it appears that after several iterations of this behaviour it kills the bus entirely. No further transfers ever take place despite transfers being queued by software independently of the FIQ.

P33M, 2017-05-04:
I have a fix for the garbled sound and breakage of the bus. The initial state wasn't being set up properly when the FIQ started the next isochronous OUT transfer. This had the side effect of the FIQ re-enabling the host channel with incorrect data lengths and presumably this can lead to core lockups.

FIQ timeouts are an unavoidable consequence of high interrupt loading of the Pi, it's less likely to happen on Pi2 and Pi3 as the CPUs perform critical sections quicker but interrupt latencies >300uS run the risk of dropping data on the floor.

closed 2017-05-04 by popcornmix

popcornmix, 2017-05-04:
Latest rpi-update kernel has the fix from @P33M

vio-f, 2022-04-28:
Interestingly I have similar issues (same error message and wireless connection restarted) on Ubuntu 21.04 when connecting a single USB audio device directly to a laptop.