raspberrypi/linux #1844: Issue buildung kernel module 4d-hats on branch rpi-4.9.y

PeterPablo, 2017-02-13:
Dear community,

during the last months I successfully built the [4d-hats kernel module](https://bitbucket.org/4DPi/4d-hats) for the kernel branch rpi-4.4.y. Since the switch to 4.9 is immanent, I wanted to give this a go, though unfortunately am failing at my novice kernel building skills.

The issue is related to #1771 -- the removal of `platform.h` leads to e.g.:
```
drivers/video/4d-hats/pwmbl.c: In function ‘init_dma_pwm’:
drivers/video/4d-hats/pwmbl.c:321:36: error: ‘GPIO_BASE’ undeclared (first use in this function)
     gpio_reg = (uint32_t *)ioremap(GPIO_BASE, GPIO_LEN);
                                    ^
drivers/video/4d-hats/pwmbl.c:321:36: note: each undeclared identifier is reported only once for each function it appears in
drivers/video/4d-hats/pwmbl.c:326:36: error: ‘DMA_BASE’ undeclared (first use in this function)
     dma_reg  = (uint32_t *)ioremap(DMA_BASE,  DMA_LEN) + dma_chan * (DMA_CHAN_SIZE / sizeof(dma_reg));
                                    ^
drivers/video/4d-hats/pwmbl.c:59:20: error: ‘BCM2708_PERI_BASE’ undeclared (first use in this function)
 #define CLK_BASE  (BCM2708_PERI_BASE + 0x101000)
                    ^
drivers/video/4d-hats/pwmbl.c:329:36: note: in expansion of macro ‘CLK_BASE’
     clk_reg  = (uint32_t *)ioremap(CLK_BASE,  CLK_LEN);
                                    ^
drivers/video/4d-hats/pwmbl.c:298:9: warning: unused variable ‘ret’ [-Wunused-variable]
     int ret;
         ^
scripts/Makefile.build:299: recipe for target 'drivers/video/4d-hats/pwmbl.o' failed
```

A more or less recent [commit](https://bitbucket.org/4DPi/4d-hats/commits/860ce5faac2d1949da366f1e6a4306662942731b?at=master) seems to have added compatability with kernel 4.4.y, introducing the dependency on `platform.h`.

The manufacturers forum unfortunately currently does not allow registration of new users. In case my question here is inapropriate, please feel free to close.

Thank you!

PeterPablo, 2017-02-13:
Is a patch like https://github.com/raspberrypi/linux/commit/9f2b0c04a3f545769d72c9bcf858bf1608db147a possibly the correct way forward?

pelwell, 2017-02-13:
Your problem isn't a lack of skill, it's having to build a driver from the pre-Device Tree dark ages.

That patch removes the fallback to hard-coded addresses, but in the case of the 4d-hats driver it makes no attempt to query the resources (the resources are populated from Device Tree) so somebody will need to add the appropriate code.

The real solution though is to let the GPIO device driver manage the gpios, and the DMA driver configure the DMA channel, restricting the backlight driver to the manipulating the PWM interface. The original driver from which this was forked has moved on - see https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/video/backlight/gpio_backlight.c

JamesH65, 2017-05-18:
Closing as this issue is unrelated to Pi kernel.

closed 2017-05-18 by JamesH65

PeterPablo, 2017-05-18:
May I ask a brief question: if the driver were to be included, should it be
directly upstreamed or would inclusion into this kernel repository also be
a valid option?
In case I try to tackle this myself, I could open a PR against rpi-4.9.y.

pelwell, 2017-05-18:
If it is a Pi-specific driver then we will consider merging it downstream - there are plenty of HAT soundcards with downstream drivers.

PeterPablo, 2017-07-31:
An update:
I opened a [thread](http://forum.4dsystems.com.au/forum/forum-aa/4d-systems-hardware/raspberry-pi-display-modules-and-shields/59074-compiling-a-kernel-for-branch-rpi-4-9-y) in the discussion forums of 4D SYSTEM. Trying to solve the issue on my own, an internet search led me to [this issue report](https://github.com/RetroPie/RetroPie-Setup/issues/1963) that contains a hacky [patch](https://github.com/psyke83/gamecon-gpio-rpi/commit/76ff1220e9ec590bdae2fb5056fa2a9cf66d20da) by @psyke83. Applying this patch to [4d-hats.c](https://bitbucket.org/4DPi/4d-hats/src/8a2234f96190fcad79eb44d314a7409c878c3cfc/4d-hats.c?at=master&fileviewer=file-view-default#4d-hats.c-50) makes this module compile.
I am not yet at a functional kernel, especially as the touchscreen driver [4dpi_touch.c](https://bitbucket.org/4DPi/4d-hats/src/8a2234f96190fcad79eb44d314a7409c878c3cfc/4dpi_touch.c?at=master) seems to require additional work.

My intention is to give an update and link to the external forum thread, where discussion may go one. In any case I would also be glad to read comments here.

PeterPablo, 2018-01-15:
Dear community,
4DPi provided sources ([reference](http://forum.4dsystems.com.au/forum/forum-aa/4d-systems-hardware/raspberry-pi-display-modules-and-shields/58507-4dpi-24-hat-kernel-for-debian-stretch-and-or-buster?p=61403#post61403)) and consequently I was able to compile my own kernel ` 4.9.76-v7+`.
Kind regards
  Peter