raspberrypi/linux #1847: rpi-4.9.x branch missing objects

mranostay, 2017-02-14:
The meta-raspberrypi layer kernel recipe for 4.9.4 points to the git commit "8b0be887f834e7eccf0de3edf630003880779a5f" which doesn't exist.. I have a question are the branches being rebased and force pushed? If so, that is going to always break the existing recipes..

Link: http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/commit/?h=morty&id=5b5c4d9334ab2342ae5f9fb53ed56053dda87be6

popcornmix, 2017-02-15:
Yes. rpi-4.4.y is considered stable and upstream commits are merged.
Newer branches (rpi-4.9.y and rpi-4.10.y) are rebased onto upstream kernel.

If you want to use an unstable kernel tree, I suggest you either just use the HEAD as a reference, or you fork the linux tree and link to your fork.

What you are doing should be safe with stable (rpi-4.4.y) tree.

kraj, 2017-02-16:
@popcornmix usually it be good if the branched which get rebased can be put under a different namespace may be and have a 4.9.y release branch which is not rebased. That will send a right message to community who just follows kernel conventions where public branches are not rebased at all 

4.9 is a LTS kernel so I am assuming that it will be declared stable for rpi soon too ?

popcornmix, 2017-02-16:
Yes, 4.9 will be the stable kernel tree in the near future.
At that point it will no longer be rebased.

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65