raspberrypi/linux #1850: `make uImage` needs ZRELADDR in rpi-4.9.y branch

betafive, 2017-02-18:
In the rpi-4.4.y branch, ZRELADDR is set by zreladdr-y in arch/arm/boot/mach-bcm2709/Makefile.boot (using bcm2709 as an example). In the rpi-4.9.y branch, ZRELADDR is left unset as no Makefile.boot file is included from a mach-* directory.

The effect is that on the rpi-4.4.y branch, the following works:

    make bcm2709_defconfig
    make uImage

Whereas on the rpi-4.9.y branch, I need to do the following:

    make bcm2709_defconfig
    make uImage LOADADDR=0x00008000

Is this change expected? If so, is there an appropriate place to document this?

If it's not expected, should I send a patch to fix this? As CONFIG_ARCH_MULTIPLATFORM=y is set by default, MACHINE is left unset during the build and no mach-* directory is used, so if I add back Makefile.boot I'd also have to unset CONFIG_ARCH_MULTIPLATFORM in the defconfigs. Or is there another possible solution I'm missing?

Many thanks,
Paul

popcornmix, 2017-02-19:
It is a change due to using the upstream kernel which defines ARCH_MULTIPLATFORM which selects AUTO_ZRELADDR.
@notro may have an idea about the options available to you, but using LOADADDR seems like a reasonable solution to me.

betafive, 2017-02-19:
Ok, LOADADDR is easy enough. Is there anywhere we can document this though so others don't have to figure it out themselves by looking back at arch/arm/boot/mach-bcm2709/Makefile.boot in the rpi-4.4.y branch?

betafive, 2017-03-17:
Ping on this. Is the new requirement to set LOADADDR when building a uImage documented anywhere?

pelwell, 2017-03-17:
I found this post (https://lists.denx.de/pipermail/u-boot/2015-May/214749.html) referring to the problem, and since this page was the first hit on my search term this issue is effectively the documentation.

pelwell, 2017-03-17:
Feel free to submit a PR to the [building](https://github.com/raspberrypi/documentation/blob/master/linux/kernel/building.md) page, but note that uImage isn't a recommended option.

betafive, 2017-03-17:
Thanks @pelwell. If uImage isn't recommended I'll give zImage a try with u-boot. After that I'll look at submitting a PR to the documentation.

closed 2017-03-17 by betafive

pelwell, 2017-03-17:
uImage may be the best target for use with u-boot. Although we are happy for people to use U-boot, and we've worked with @swarren to improve the kernel stubs, we don't support U-boot ourselves, so a better option might be to update the build instructions on a U-boot site and add a link to our building page.

betafive, 2017-03-17:
Ok cool, I'll see if I can find some build instructions for Raspberry Pi on their site.

swarren, 2017-03-17:
Even with U-Boot, I personally recommend not using uImage. U-Boot is perfectly capable of loading/executing standard kernel zImage/Image files, and as such I recommend avoiding wrapping those files in U-Boot-specific wrappers such as uImage. Aside from simplifying the build process and avoiding bootloader-specific image formats, avoiding uImage (and hence using zImage/Image directly) also allows you to load the kernel image directly to the address it should be placed at (typically $kernel_addr_r in many modern U-Boot builds), and jumping straight to it. With uImage, you need to load the kernel image to some different location and U-Boot does a pointless memcpy to the correct location when jumping to it.